zigpy/zigpy #984: asyncio.exceptions.CancelledError when sending message

agatah2333, 2022-05-03:
Hello, I am trying to use Zigby as an independent library to operate my IoT. And I successfully operated my smart bulb and smart plug and the main code like this. 

<meta charset="utf-8"><b style="font-weight:normal;" id="docs-internal-guid-00da0cc5-7fff-8fe2-f21b-e03bd66bd07f"><div dir="ltr" style="margin-left:0pt;" align="left">

Sengled | Smart Bulb | b0:ce:18:14:00:1b:06:49 | 0xac66
-- | -- | -- | --
LUMI | Smart plug | 00:15:8d:00:02:a9:a5:73 | 0xf343

</div></b>

``` python
try:
        logging.debug("start device command")
        dev= zigbeeApp.get_device(nwk= 0xa6cc).endpoints[1]
        res = await dev.on_off.on()
        res = await dev.on_off.off()
        res= await zigbeeApp.get_device(nwk= 0xf343).endpoints[1].on_off.on()
        res= await zigbeeApp.get_device(nwk= 0xf343).endpoints[1].on_off.off()
        logging.debug("done device command")
except Exception as e:
        print(e)
```
But after these operations, I got an error:
And I noticed after sending zcl command frame
```
2022-05-03 14:47:04,143 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 4, 0, <DeconzAddressEndpoint address_mode=2 address=0xA6CC endpoint=1>, 260, 6, 1, b'\x01\x03\x01', 2, 0)
```
Another message will be sending to the device:
```
2022-05-03 14:47:04,247 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 23 under 7 request id, data: b'18170b0a00'
```
I wonder :
1. why need to send this message? Does a frame come from zcl or zdo? Cause I could not find any special meaning of 0x18?
2. why did this error happen? 

Thank you so much for you valuable time!

[image]

And the log file of these operations are as follow:
```
2022-05-03 14:47:04,140 DEBUG (MainThread) [root] start device command
2022-05-03 14:47:04,142 DEBUG (MainThread) [zigpy.device] [0xa6cc] Extending timeout for 0x03 request
2022-05-03 14:47:04,142 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 3 under 4 request id, data: b'010301'
2022-05-03 14:47:04,143 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 4, 0, <DeconzAddressEndpoint address_mode=2 address=0xA6CC endpoint=1>, 260, 6, 1, b'\x01\x03\x01', 2, 0)
2022-05-03 14:47:04,143 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121a0019001200040002cca601040106000103000103010200
2022-05-03 14:47:04,146 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121a00090002002204
2022-05-03 14:47:04,147 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 4]
2022-05-03 14:47:04,155 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1b000700a600
2022-05-03 14:47:04,156 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2022-05-03 14:47:04,156 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-03 14:47:04,157 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x041b0007000000
2022-05-03 14:47:04,159 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x041b0013000c00220402cca601010000000000
2022-05-03 14:47:04,160 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 4: 00
2022-05-03 14:47:04,161 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x04 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xa6cc endpoint=1>, status: 0x00
2022-05-03 14:47:04,211 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1c000700aa00
2022-05-03 14:47:04,212 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-05-03 14:47:04,213 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, 1)
2022-05-03 14:47:04,214 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x171c000800010001
2022-05-03 14:47:04,217 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x171c0023001c00220200000102cca60104010600050008030b010000afff53e12300e6
2022-05-03 14:47:04,218 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [28, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0x0000>, 1, <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa6cc>, 1, 260, 6, b'\x08\x03\x0b\x01\x00', 0, 175, 255, 83, 225, 35, 0, -26]
2022-05-03 14:47:04,220 DEBUG (MainThread) [zigpy.zcl] [0xa6cc:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Default_Response>
2022-05-03 14:47:04,222 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xa6cc>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'08030b0100'
2022-05-03 14:47:04,223 DEBUG (MainThread) [zigpy.device] [0xa6cc] Extending timeout for 0x05 request
2022-05-03 14:47:04,224 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 5 under 6 request id, data: b'010500'
2022-05-03 14:47:04,224 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 6, 0, <DeconzAddressEndpoint address_mode=2 address=0xA6CC endpoint=1>, 260, 6, 1, b'\x01\x05\x00', 2, 0)
2022-05-03 14:47:04,225 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x121d0019001200060002cca601040106000103000105000200
2022-05-03 14:47:04,227 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e1d000700aa00
2022-05-03 14:47:04,228 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_INDICATION|2: 170>, 0]
2022-05-03 14:47:04,230 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x121d00090002002a06
[... 96 lines cut ...]
2022-05-03 14:47:05,376 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=ADDRESS_MODE.NWK address=0xf343>, ep: 1, profile: 0x0104, cluster_id: 0x0006, data: b'180a0b0000'
2022-05-03 14:47:05,377 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2022-05-03 14:47:05,377 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x042a0007000000
2022-05-03 14:47:05,379 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x042a0013000c00220b0243f301010000000000
2022-05-03 14:47:05,380 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 11: 00
2022-05-03 14:47:05,381 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x0b 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0xf343 endpoint=1>, status: 0x00
2022-05-03 14:47:05,381 DEBUG (Thread-1) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7fb4a56a80>) completed
2022-05-03 14:47:05,382 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (20, 12, 0, <DeconzAddressEndpoint address_mode=2 address=0xA6CC endpoint=1>, 260, 6, 1, b'\x18\x18\x0b\n\x00', 2, 0)
2022-05-03 14:47:05,383 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x122b001b0014000c0002cca6010401060001050018180b0a000200
2022-05-03 14:47:05,384 DEBUG (MainThread) [root] done device command
2022-05-03 14:47:05,386 DEBUG (Thread-1) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fb4a56a80>, 'INSERT OR REPLACE INTO attributes_cache_v7 VALUES (?, ?, ?, ?, ?)', (00:15:8d:00:02:a9:a5:73, 1, 6, 61440, 117440520))
2022-05-03 14:47:05,389 DEBUG (Thread-1) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fb4a56a80>, 'INSERT OR REPLACE INTO attributes_cache_v7 VALUES (?, ?, ?, ?, ?)', (00:15:8d:00:02:a9:a5:73, 1, 6, 61440, 117440520)) completed
2022-05-03 14:47:05,470 ERROR (MainThread) [zigpy.zcl] [0xa6cc:1:0x0006] Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy/device.py", line 265, in request
    result, msg = await self._application.request(
  File "/home/pi/.local/lib/python3.9/site-packages/zigpy_deconz/zigbee/application.py", line 257, in request
    await self._api.aps_data_request(
  File "/home/pi/.local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 517, in aps_data_request
    return await self._command(
  File "/home/pi/.local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 305, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 468, in wait_for
    await waiter
asyncio.exceptions.CancelledError
```

puddly, 2022-05-03:
Zigpy is sending a default response to your bulb's attribute report and the request seems to fail:

```Python
2022-05-03 14:47:04,245 DEBUG (MainThread) [zigpy.zcl] [0xa6cc:1:0x0006] Attribute report received: on_off=1
```

Can you try updating your Conbee to the most recent firmware?

agatah2333, 2022-05-03:
Sure! Thanks for your explanation!

agatah2333, 2022-05-05:
I update my conbee two firmware, and the error never shows again! But the message starting with "0x18" still shows in the log. I read the code you mentioned but am still confused. Is it the correct event for sending those messages?

puddly, 2022-05-05:
> But the message starting with "0x18" still shows in the log.

Which one are you referring to?

agatah2333, 2022-05-05:
This message in the log:
```
2022-05-03 14:47:04,247 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 23 under 7 request id, data: b'18170b0a00'
```

Hedda, 2022-05-10:
> Hello, I am trying to use Zigby as an independent library to operate my IoT.

@agatah2333 Maybe will not help directly here but for troubleshooting when developing it would probably be a good idea for you to also get a Texas Instrument CC26x2/CC13x2 and also a Silicon Labs EFR32MG1x/EFR32MG2x Zigbee Coordinator for the reason that they all run different Zigbee stacks so might show different symptoms and messages with some devices or scenarios.

agatah2333, 2022-05-12:
Thanks! I will check with that required hardware. Do you prefer the bellow (Silicon lab dongle) rather than deconz (conbee ii)?

puddly, 2022-05-13:
I'm seeing something strange in your log:

```
2022-05-03 14:47:04,220 DEBUG (MainThread) [zigpy.zcl] [0xa6cc:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=3 command_id=Command.Default_Response>
```

Zigpy logs sent commands (including default responses) more verbosely as of 0.44.0, which was released a little over six weeks ago:

```
2022-05-13 06:10:41.0838 DEBUG zigpy.zcl:378 [0x4C7C:1:0x0008] Attribute report received: current_level=1
2022-05-13 06:10:41.0840 DEBUG zigpy.zcl:326 [0x4C7C:1:0x0008] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=104, command_id=<GeneralCommand.Default_Response: 11>, *is_reply=True)
2022-05-13 06:10:41.0841 DEBUG zigpy.zcl:327 [0x4C7C:1:0x0008] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
```

You should update your libraries. The logs should make more sense then.

Hedda, 2022-05-13:
> Thanks! I will check with that required hardware. Do you prefer the bellow (Silicon lab dongle) rather than deconz (conbee ii)?

Not sure if prefered is the right term, however as I understand, puddly is using Texas Instruments CC26x2/CC13x2 based Zigbee Coordinator in his personal home "main production" Zigbee network, and Adminiuga is using EFR32MG1x/EFR32MG2x based Zigbee Coordinator in his personal home "main production" Zigbee network, and both of those development platforms for Zigbee are currently the two top Zigbee industry leaders, and they have been so so for a long time. With the runner-up's competition TI and Silabs being NXP and Nordic Semiconductor (nither of which have zigpy radio libraires as of yet). 

Also just how I see it, I believe that while deconz (conbee/raspbee) is very popular as Zigbee Coordinator adapter producs for the DIY home automation end-user market, my guess is that there are very few independent developers using them for development, and that is because Dresden Electronik who makes deconz/conbee/raspbe) does not sell or produce Zigbee modules/chip for third-party manufacturers or provide Zigbee stack software development kits and documentation for external professionals such as Texas Instruments and Silicon Labs do because it is not the core business of Dresden Electronik to do so. 

My guess is that the same will almost for certain also go with other developers out there that work professionally with Zigbee, they will all mainly focus on developing with Texas Instruments, Silicon Labs, NXP, and/or Nordic Semi hardware and Zigbee stacks.

Hedda, 2022-05-13:
> I am trying to use Zigby as an independent library to operate my IoT. And I successfully operated my smart bulb and smart plug

@agatah2333 off-topic but I think that it would be very cool and useful to many others if you as a developer of an a custom IoT solution that will depend on zigpy could also post a new "discussion" that provides constructive feedback and input on how the zigpy library and the [new zigpy-cli command-line interface for zigpy](https://github.com/zigpy/zigpy-znp) could be improved to as an independent library for other custom IoT projects -> https://github.com/zigpy/zigpy/discussions (specific problems and bugs you stumble upon are of course normally still best posted as seperate new issues under respective repository, e.i. https://github.com/zigpy/zigpy/issues and https://github.com/zigpy/zigpy-cli etc.).

At least I'm hoping that understanding how other independent projects could use zigpy could eventually indirectly help improve all projects that are dependent on the zigpy libraires. That is at least something similar to what has happened with the latest reincarnation of the https://github.com/zigbeefordomoticz/Domoticz-Zigbee project began as an open dialogue here -> https://github.com/zigpy/zigpy/discussions/865

Even more off-topic, by the way, I also understand that the zha-websocket-server (a.k.a. "ZHAWS") that is currently in early development https://github.com/zigpy/zha-websocket-server (as well as the matching Home Assistant addon for it https://github.com/zigpy/zhaws-addon for reference) could maybe also make integrations of Home Assistant's ZHA into other applications possible as a modular consistent application with a WebSocket API(?).

agatah2333, 2022-05-18:
Thanks! will do

closed 2022-05-18 by agatah2333