zigpy/zigpy #987: Sonoff Zigbee OTA provider for Sonoff branded Zigbee devices by ITead

Hedda, 2022-05-24:
_Originally posted by @Daniel-zhan-itead in https://github.com/zigpy/zha-device-handlers/issues/1519#issuecomment-1132783043_

> We have raised the pull request for the OTA update image on github https://github.com/Koenkk/zigbee-OTA/pulls.
> Once it is approved, you can download the image and update the ZBMINI-L devices.

@Daniel-zhan-itead Great to see that [ITead has publicly published your first OTA file for a Sonoff branded Zigbee device on GiHub](https://github.com/Koenkk/zigbee-OTA/tree/master/images/Sonoff) now.

Follow up question to that is if you at ITead can spare some development time at extending the OTA provider for it in the zigpy library?

As you might know, the zigpy library is now used as a dependecy by the many popular home automation software and having Sonoff Zigbee OTA provider support could potentially enable easier OTA upgrades for end-users of those in the future, and that today include; [ZHA integration](https://www.home-assistant.io/integrations/zha) for Home Assistant, [Zigbee plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz) ([Domoticz-Zigbee](https://github.com/zigbeefordomoticz/Domoticz-Zigbee) project), and [Zigbee plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/).

Please see code https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py in https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

@Daniel-zhan-itead Tough before deciding to add that Koenkk's repository as URL link as source for Sonoff Zigbee OTA provider, can you at ITead please consider creating a new separate dedicated repository called "Sonoff_Zigbee_OTA" or similar under ITead's own official organization on GitHub and use that as your primary publishing location for all officially released Zigbee OTA files for Sonoff branded Zigbee devices?

https://github.com/itead

(That new repository would for example be called **_hxxps://github.com/itead/Sonoff_Zigbee_OTA_** or something like that).

Preferably also copying the concept of `index.json` with metadata on all files like Koenkk repository has to make scripting easier:

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

And of course `README.md` file which among other things could contain some basic information on how to update those:

https://github.com/Koenkk/zigbee-OTA/blob/master/README.md

Perhaps also as keeping some kind of `CHANGELOG.md` with changelog release history for each device firmware version:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

For ex. `zbmini-l_v1.1.1_release_notes.md` and so on with release notes for each version released + compounded changelog.

That way it will be a single trusted and official primary source of official Zigbee OTA files released by ITead, and that OTA provider source repository with your Zigbee OTA files could not only be used by zigpy as it could also be reused by other Zigbee implementations independently (both other open-source implementations as well as closed source and proprietary implementations).

PS: By the way, please also refer to the somewhat related discussion regarding here Koenkk zigbee-OTA repository here -> https://github.com/zigpy/zigpy/issues/535

Daniel-zhan-itead, 2022-05-25:
Hi Hedda,

Thanks for your suggestion.

We released the ZBMINI-L firmware update image for users is a temporary solution as many users need to fix the issue ASAP.

Our team is also developing the "auto"  OTA upgrade with zigpy as you advise, but it takes time to develop and the expected release timeline is in July. The OTA update image will be located on our server (example: https://zigbee-ota.sonoff.tech/releases/xxx) so users can update the Zigbee device via such OTA firmware image source URL. thus, we did not plan to create a new separate repository in ITEAD github.

Hedda, 2022-08-04:
> Our team is also developing the "auto" OTA upgrade with zigpy as you advise, but it takes time to develop and the expected release timeline is in July. The OTA update image will be located on our [...]

@Daniel-zhan-itead Any news on the mentioned Sonoff Zigbee OTA update server from ITead?

Daniel-zhan-itead, 2022-08-19:
> @Daniel-zhan-itead Any news on the mentioned Sonoff Zigbee OTA update server from ITead?

@Hedda The OTA update capability for Zigbee device is still under development as we met some issues when we are testing. Our team is investigating and fixing the issue. Once it can work robustly, we will release it to the public.

Daniel-zhan-itead, 2022-08-30:
@Hedda The Zigbee device auto OTA function for the open source platform is ready, our engineer has submitted a pull request on [zigpy](https://github.com/zigpy/zigpy) as the below link shown

https://github.com/zigpy/zigpy/pull/1039

Hedda, 2022-09-05:
> @Hedda The Zigbee device auto OTA function for the open source platform is ready, our engineer has submitted a pull request on [zigpy](https://github.com/zigpy/zigpy) as the below link shown
> 
> #1039

@Daniel-zhan-itead  Noticed that @lisongjun12 closed that and instead submitted new pull request https://github.com/zigpy/zigpy/pull/1040

So the SONOFF Zigbee device OTA firmware server is now available https://zigbee-ota.sonoff.tech/releases/upgrade.json ?

Daniel-zhan-itead, 2022-09-05:
> @Daniel-zhan-itead Noticed that @lisongjun12 closed that and instead submitted new pull request #1040
>
> So the SONOFF Zigbee device OTA firmware server is now available [...]

Yup, I asked him to update the title of the pull request, so he closed the first one.

closed 2022-09-30 by puddly

Hedda, 2022-10-03:
@Daniel-zhan-itead @lisongjun12 Note that while zigpy release 0.51.0 now has a merged OTA provider for Sonoff devices you still need to remember to submit a matching pull request for the `next` branch of home-assistant.io repository to document ZHA's Sonoff OTA provider config option once the ZHA developers have bumped that as zigpy dependency in Home Assistant core repository (which is not yet), as otherwise Sonoff OTA support will not be shows in the documentation for the ZHA integration:

https://github.com/zigpy/zigpy/releases/tag/0.51.0

https://github.com/home-assistant/core/pulls?q=is%3Apr+zigpy

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

https://github.com/home-assistant/home-assistant.io/blob/next/source/_integrations/zha.markdown

PS: See example PR -> https://github.com/home-assistant/home-assistant.io/pull/22955

Daniel-zhan-itead, 2022-10-13:
> @Daniel-zhan-itead @lisongjun12 Note that while zigpy release 0.51.0 now has a merged OTA provider for Sonoff devices you still need to remember to submit a matching pull request for the `next` branch of home-assistant.io repository to document ZHA's Sonoff OTA provider config option once the ZHA developers have bumped that as zigpy dependency in Home Assistant core repository (which is not yet), as otherwise Sonoff OTA support will not be shows in the documentation for the ZHA integration:
> 
> https://github.com/zigpy/zigpy/releases/tag/0.51.0
> 
> https://github.com/home-assistant/core/pulls?q=is%3Apr+zigpy
> 
> https://www.home-assistant.io/integrations/zha#ota-firmware-updates
> 
> https://github.com/home-assistant/home-assistant.io/blob/next/source/_integrations/zha.markdown
> 
> PS: See example PR -> [home-assistant/home-assistant.io#22955](https://github.com/home-assistant/home-assistant.io/pull/22955)

Thanks for your reminder, @lisongjun12 will submit a PR to update the documentation for zha#ota-firmware-updates

Hedda, 2022-10-15:
@Daniel-zhan-itead @lisongjun12 since this is now already availavle in tje current code of Home Assistant core you can now target their `current` branch instead of thier `next` branch for the ZHA documentation https://github.com/home-assistant/home-assistant.io/blob/current/source/_integrations/zha.markdown

Hedda, 2023-04-04:
@Daniel-zhan-itead @lisongjun12 FYI, puddly is working on standard for "Generic OTA providers" + a new JSON format schema:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

  "**_This allows for new providers to be added to zigpy without extra code. This also allows for manufacturers to distribute test feed URLs to customers without deploying them globally._**"

This feature will be introduced via this new pull request that is still a work-in-progress. Please see ->  https://github.com/zigpy/zigpy/pull/1165