zigpy/zigpy #988: Ubisys OTA provider for Ubisys branded Zigbee devices

Hedda, 2022-05-28:
Posting this as dedicated feature request here as someone posted this question in Home Assistant community forum after asking Ubisys:

https://community.home-assistant.io/t/zha-ubisys-ota-updates-firmware-is-available-on-their-website/290430

Ubisys support sent this reply to users  question about Zigbee OTA support in ZHA and other open source Zigbee implementations:

> **_We do in fact make the newst firmware available to the general public on our website, so a proper OTA server (e.g. Raspberry… ) can be used to distribute our update files easily._** 
>  
> **_Here´s the link: [http://www.ubisys.de/en/support/firmware/ 11](http://www.ubisys.de/en/support/firmware/)_**
> 
> **_As far as I´m aware of we neither have we ourselves approached, nor have we been approached by ZHA, regarding 3rd party OTA files being supported, sorry!_** 
> 
> **_You could get in touch with them as well, maybe that´ll create a ticket on their side - initiating first contact with us :wink:_**
> 
> **_I´ll also mention this in our next team meeting!_**

Hedda, 2022-08-04:
FYI, I got this reply from Ubisys when I asked about a Ubisys Zigbee OTA provider in zigpy for Home Assistant's ZHA integration:

> "**_Hi,_**
> 
> **_Currently we do not have any development resources available, but already have agreed to take care of this topic._**
> 
> **_However, it is difficult to give a timeline, but we will try to work on it in the short term and keep you posted._**
> 
> **_Kind regards_**
> 
> **_Ubisys Support Team_**"

Hedda, 2023-02-15:
FYI, puddly has just now added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce a skeleton JSON index from a given set of OTA files, and that way make sure manufacturers OTA files are correctly-structured so that they will be compatible with zigpy out-of-the-box. See the README.md for that "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool:

https://github.com/zigpy/zigpy-cli/blob/dev/README.md#ota

PS: puddly mentioned when added this that that feature for generating OTA index files via zigpy-cli was done in preparation for some OTA rewrites in the main zippy library that he is planning to do.

Hedda, 2023-02-27:
FYI, puddly is now working on supporting "Generic OTA providers" for zigpy via this new JSON format schema:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

  "**_This allows for new providers to be added to zigpy without extra code. This also allows for manufacturers to distribute test feed URLs to customers without deploying them globally._**"

This feature will be introduced via this new pull request that is still a work-in-progress ->  https://github.com/zigpy/zigpy/pull/1165

Hedda, 2023-08-16:
> FYI, puddly is now working on supporting "Generic OTA providers" for zigpy via this new JSON format schema:
>
> https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers
>
> "**_This allows [...]

FYI, “**_Manufacturer OTA Firmware Update Support_**” (also referred to as “**_Generic OTA providers_**”) shold now be abilable for use in ZHA as that was added with zigpy 0.56.3 release and that was very recently bumped in Home Assistant core version 2023.8.1(?)

https://github.com/zigpy/zigpy/pull/1165

This new feature in zigpy (and ZHA) should allow for new OTA providers to be added to zigpy (and ZHA) without extra code, which in turn also allows both manufacturers (and end-users) to distribute test feed URLs to customers without deploying them globally.

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

PS: Again, puddly also added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce/generate a skeleton index from a given set of OTA files, and that way makes sure manufacturers OTA files are correctly-structured so that they will be compatible with this new JSON format for zigpy OTA providers out-of-the-box. Check out the README.md for the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" (zigpy CLI tool) -> [https://github.com/zigpy/zigpy-cli/blob/dev/README.md#generate-ota-index-files](https://github.com/zigpy/zigpy-cli/blob/dev/README.md#generate-ota-index-files)

Hedda, 2024-02-18:
More relevant now with https://github.com/zigpy/zigpy/pull/1340

puddly, 2024-02-18:
https://github.com/zigpy/zigpy/issues/1060#issuecomment-1951415531

closed 2024-02-18 by puddly