zigpy/zigpy #992: dresden elektronik Zigbee OTA provider in Home Assistant ZHA integration code?

Hedda, 2022-06-07:
@manup Wondering if you would please consider contributing download code for a dresden elektronik Zigbee OTA provider to zigpy?

http://deconz.dresden-elektronik.de/otau/

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/OTA-Image-Types---Firmware-versions#dresden-elektronik

| Product Code(s) | Image type | Product(s)  | Observed FW Version(s) | OTAU via DeCONZ |
|-----------------|------------|-------------|------------------------|-----------------|
| FLS-PP3         | 0x0000     | FLS-PP lp   | [0x201000F5](http://deconz.dresden-elektronik.de/otau/1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee)             | ✅              |
| Lighting Switch  | 0x0005     | Lighting Switch | [0x0202001C](http://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |
| Scene Switch    | 0x0005     | Scene Switch | [0x0202001C](http://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |

Home Assistant's ZHA integration is a Zigbee gateway implementation (which depends on the "zigpy" library) is one the most popular integrations in Home Assistant as per statistics provided by the few procentage of their users who opt-in to submit usage analytics (as analytics is optional and disabled by default in Home Assistant):

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

https://analytics.home-assistant.io/#integrations

Having more Zigbee OTA providers included with Home Assistant's ZHA integration is a common feature request from many users in the Home Assistant community , so would like to request that you to submit a small piece of code to allow the zigpy library to download Zigbee OTA images (Over-The-Air firmware updates) for your own devices from your public Zigbee OTA provider as source:

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py

PS: Please note that other than in the ZHA integration for Home Assistant the same zigpy library is also used as a dependecy by the official Zigpy plugin for Jeedom as well as the Zigbee plugin for Domoticz, see:

https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/

and

https://www.domoticz.com/wiki/ZigbeeForDomoticz

manup, 2022-06-17:
Hi @Hedda it would be good to have an unified approach like the index.json, I can provide this on the server for our OTA files. This mimics the Ikea OTA handling I think. In future it might be worth to check the Matter OTA handling which is distributed but  a bit over engineered.

For the Python OTA provider perhaps someone else can jump in, I'm currently a bit short on time :/

Hedda, 2022-06-17:
> it would be good to have an unified approach like the index.json, I can provide this on the server for our OTA files. This mimics the Ikea OTA handling I think.

That would be great! FYI, it also looks like Inovelli's firmware.json file as well taken inspiration from Koenkk's index.json or IKEA?

https://files.inovelli.com/firmware/firmware.json

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

http://fw.ota.homesmart.ikea.net/feed/version_info.json

http://fw.test.ota.homesmart.ikea.net/feed/version_info.json

Hedda, 2022-06-17:
> For the Python OTA provider perhaps someone else can jump in, I'm currently a bit short on time :/

Perhaps could try motivating existing zigpy developers by asking dresden elektronik's bosses to ship samples of those devices? ;)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/OTA-Image-Types---Firmware-versions#dresden-elektronik
| Product Code(s) | Image type | Product(s)  | Observed FW Version(s) | OTAU via DeCONZ |
|-----------------|------------|-------------|------------------------|-----------------|
| [FLS-PP3 ](https://zigbee.blakadder.com/dresden_elektronik_FLS-PP.html)        | 0x0000     | FLS-PP3, FLS-PP-IP   | [0x201000F5](http://deconz.dresden-elektronik.de/otau/1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee)             | ✅              |
| [Lighting Switch](https://zigbee.blakadder.com/dresden_elektronik_BN-600087.html)  | 0x0005     | Lighting Switch | [0x0202001C](http://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |
| [Scene Switch](https://zigbee.blakadder.com/dresden_elektronik_BN-600085.html)    | 0x0005     | Scene Switch | [0x0202001C](http://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |

Hedda, 2022-08-04:
> Hi @Hedda it would be good to have an unified approach like the index.json, I can provide this on the server for our OTA files. This mimics the Ikea OTA handling I think. In future it might be worth [...]

@manup Have you given any more thought to providing some kind of index file(s) dresden elektronik server for Zigbee OTA files? 

FYI, in related news Home Assistant's Z-Wave (Z-Wave JS) developers exposed OTA updates for Z-Wave devices in UI from Home Assistant 2022.7 release, and they also created a new Firmware Update Service (web service with an API) to power semi-automatic firmware updates in Z-Wave JS, 

https://github.com/zwave-js/firmware-updates

https://github.com/zwave-js/firmware-updates/blob/main/README.md

The firmware update service for Z-Wave JS

https://firmware.zwave-js.io

They talk more about it in this video and blog post:

https://youtu.be/HnA_pZQyZAs?t=2908

https://www.home-assistant.io/blog/2022/07/06/release-20227/#update-z-wave-devices-directly-from-home-assistant

## **_Z-Wave JS Firmware Update Service_**

**_This is the web service powering semi-automatic firmware updates in Z-Wave JS._**

### **_How it works_**

**_Firmware upgrades are defined in the files in the firmware directory. This information is parsed by the web service and used to respond to requests._**

**_Z-Wave JS will query the web service on demand, providing the device identification of the device to be updated. If an update is available, the web service will respond with the necessary information to display to the user._**

**_When desired, Z-Wave JS will download the update(s) from the provided URL, verify their integrity and install them_**

## **_Update Z-Wave devices directly from Home Assistant_**

**_All software has bugs, including the software on your Z-Wave devices. But how to update those? Have no fear! As of today, we have a solution for this!_**

**_Thanks to the hard work and collaboration between [@AlCalzone](https://github.com/AlCalzone) from Z-Wave JS and [@raman325](https://github.com/raman325) from Home Assistant, you can now install updates on your Z-Wave devices directly from the Home Assistant interface!_**

**_Screenshot showing the new upload firmware item on the device page of a Z-Wave device page_**

[image]

**_On the device page of your Z-Wave device, there is now a menu item allowing you to install Z-Wave firmware updates onto your device manually._**

**_Get a software update for your Z-Wave device from the manufacturer, and start an update in Home Assistant, which allows you to upload the update file you got from the manufacturer. The rest is pure magic!_**

**_Screenshot showing the a Z-Wave device that has firmware installation in progress_**

[image]

**_During update installation, you’ll be presented with the update’s progress._**

**_It is that easy_** 💪

PS: They also made a tool is meant to help generate the integrity hash for that Z-Wave JS firmware update service:

https://github.com/zwave-js/firmware-integrity

Hedda, 2022-09-27:
@manup noticed that Dresden Elektronik have now released a new "Phoscon Kobold" Zigbee 3.0 dimmer switch module device:

https://phoscon.de/en/kobold

Looks like it is so far only sold in Europe(?) but its hardware specification lits AC 100-240V at 50/60Hz so should work globally?

Any chance you consider donating a few of those if there are perhaps one or more Python developer(s) contributing to zigpy and/or zha that might be willing to look at dresden elektronik Zigbee OTA provider for dresden elektronik made/branded Zigbee devices?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/OTA-Image-Types---Firmware-versions#dresden-elektronik

<html>
<body>
<h2 style="box-sizing: border-box; margin-top: 24px; margin-bottom: 16px; font-size: 1.5em; font-weight: 600; line-height: 1.25; padding-bottom: 0.3em; border-bottom: 1px solid var(--color-border-muted); color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">dresden elektronik</h2><p style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Manufacturer code:<span> </span><strong style="box-sizing: border-box; font-weight: 600;">0x1135</strong></p>

Product Code(s) | Image type | Product(s) | Observed FW Version(s) | OTAU via DeCONZ
-- | -- | -- | -- | --
Phoscon Kobold | 0x0100 | Phoscon Kobold | 0xF0900029 | ✅

</body>
</html>

https://deconz.dresden-elektronik.de/otau/1135-0100-F0900029-Kobold.zigbee

[image]

Hedda, 2022-10-18:
@manup a related question is if dresden elektronik is aware of Home Assistant's "**_Works with Home Assistant_**" partner program with its matching "**_Works via Zigbee with Home Assistant_**" compatibility badges and if dresden elektronik will consider joining it?

https://www.home-assistant.io/blog/2022/07/12/partner-program/

https://partner.home-assistant.io/

[image]

FYI, just a few days ago they announced that another company that makes Zigbee devices has joined their partner program,

https://www.home-assistant.io/blog/2022/10/13/third-reality-partner/

I think it is related to zigpy as it may imply a commitment from companies to work with OTA updates and ZHA Device Handlers.

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling

https://github.com/zigpy/zha-device-handlers

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

Hedda, 2023-02-15:
@manup FYI, puddly has just now added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce a skeleton JSON index from a given set of OTA files, and that way make sure manufacturers OTA files are correctly-structured so that they will be compatible with zigpy out-of-the-box. See the README.md for that "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool:

https://github.com/zigpy/zigpy-cli/blob/dev/README.md#ota

PS: puddly mentioned when added this that that feature for generating OTA index files via zigpy-cli was done in preparation for some OTA rewrites in the main zippy library that he is planning to do.

Hedda, 2023-02-27:
FYI, puddly is now working on supporting "Generic OTA providers" for zigpy via this new JSON format schema:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

  "**_This allows for new providers to be added to zigpy without extra code. This also allows for manufacturers to distribute test feed URLs to customers without deploying them globally._**"

This feature will be introduced via this new pull request that is still a work-in-progress ->  https://github.com/zigpy/zigpy/pull/1165

Hedda, 2023-08-16:
FYI, “**_Manufacturer OTA Firmware Update Support_**” (also referred to as “**_Generic OTA providers_**”) shold now be abilable for use in ZHA as that was added with zigpy 0.56.3 release and that was very recently bumped in Home Assistant core version 2023.8.1(?)

https://github.com/zigpy/zigpy/pull/1165

This new feature in zigpy (and ZHA) should allow for new OTA providers to be added to zigpy (and ZHA) without extra code, which in turn also allows both manufacturers (and end-users) to distribute test feed URLs to customers without deploying them globally.

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

PS: Again, puddly also added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce/generate a skeleton index from a given set of OTA files, and that way makes sure manufacturers OTA files are correctly-structured so that they will be compatible with this new JSON format for zigpy OTA providers out-of-the-box. Check out the README.md for the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" (zigpy CLI tool) -> [https://github.com/zigpy/zigpy-cli/blob/dev/README.md#generate-ota-index-files](https://github.com/zigpy/zigpy-cli/blob/dev/README.md#generate-ota-index-files)

Hedda, 2023-11-23:
FYI, dresden elektronik now also provide OTAU images for their new Phoscon Hive, Phoscon Kobold, and FLS-A lp wireless ballast:

https://www.dresden-elektronik.com/product/Hive.html

https://www.dresden-elektronik.com/product/kobold.html

https://www.dresden-elektronik.com/product/fls-a-lp-wireless-ballast.html

| Product Code(s) | Image type | Product(s)  | Observed FW Version(s) | OTAU via DeCONZ |
|-----------------|------------|-------------|------------------------|-----------------|
|Phoscon Hive   | 0x0101     | Phoscon Hive| [0x0020001B](https://deconz.dresden-elektronik.de/otau/1135-0101-0020001B-Hive.zigbee) | ✅              |
|Phoscon Kobold   | 0x0100     | Phoscon Kobold | [0x1000002A](https://deconz.dresden-elektronik.de/otau/1135-0100-1000002A-Kobold.zigbee) | ✅              |
| FLS-PP3         | 0x0000     | FLS-PP lp   | [0x201000F5](https://deconz.dresden-elektronik.de/otau/1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee)             | ✅              |
| FLS-A lp         | 0x0004     | FLS-A lp   | [0x201000E9](https://deconz.dresden-elektronik.de/otau/1135-0004-201000E9-FLS-A2_MD.zigbee)             | ✅              |
| Lighting Switch  | 0x0005     | Lighting Switch | [0x0202001C](https://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |
| Scene Switch    | 0x0005     | Scene Switch | [0x0202001C](https://deconz.dresden-elektronik.de/otau/1135-0005-0202001C-JUTA.zigbee)             | ✅              |

Hedda, 2024-11-07:
> > Any chance you consider donating a few of those if there are perhaps one or more Python developer(s) contributing to zigpy and/or zha that might be willing to look at dresden elektronik Zigbee OTA provider for dresden elektronik made/branded Zigbee devices?
> 
> Yes I think that should be possible, I'll ask the team.

@manup did you ask about contributing to the zigpy and zha projects? Home Assistant has grown A LOT in the last few years!

https://analytics.home-assistant.io

FYI, "Works with Home Assistant" partner program rules been updated requiring each specific device get certified by Nabu Casa:

https://www.home-assistant.io/blog/2024/08/08/works-with-home-assistant-becomes-part-ohf/

In related news, Home Assistant and official projects like zigpy and zha been transfered to non-profit Open Home Foundation:

https://www.openhomefoundation.org/about/

PS: Home Assistant was just now announced as the top open-source project with contributions from the most amount of people:

https://github.blog/news-insights/octoverse/octoverse-2024/

Hedda, 2025-10-20:
> Any chance you consider donating a few of those if there are perhaps one or more Python developer(s) contributing to zigpy and/or zha that might be willing to look at dresden elektronik Zigbee OTA [...]

@manup @Haerteleric Bump the request for native Zigbee OTA "provider" support for Dresden Elektronik devices in zigpy and zha:

* https://www.home-assistant.io/integrations/zha#ota-updates-of-zigbee-device-firmware

* https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

* https://github.com/zigpy/zigpy/wiki/OTA-Configuration

* https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/

TheJulianJES, 2026-01-16:
The zigpy-ota `stable` channel is active by default with Home Assistant Core 2026.1.0 and later. No config required.
You only need extra YAML configuration when you want to switch to the `beta` or `dev` channel. We'll expose an option to enable the `beta` zigpy-ota channel via the UI in the next major HA release. Most users should just stay on `stable` though.

Any images merged in that repo will automatically show up when using ZHA after we create a (stable) release on GitHub.
Submissions can easily be made using the [issue form](https://github.com/zigpy/zigpy-ota/issues/new/choose), no PR or JSON required. So far, we want manufacturers to submit images **including changelog/release notes** (shown in HA). We'd appreciate if Dresden Elektronik does this, of course 😄 

Here's a more extensive introductory comment I wrote for ThirdReality: https://github.com/zigpy/zigpy/discussions/1724#discussioncomment-15410926

The [zigpy-ota README](https://github.com/zigpy/zigpy-ota#readme) also contains links at the top where you can browse the firmware available in each channel with the metadata, all in a human-readable markdown file.