zigpy/zigpy #996: ZHA sends events only by one key (qbkg03lm)

Shurik85, 2022-06-12:
ZHA sends events only by one key (qbkg03lm), the second key is simply ignored and not fixed in zha_event

Hedda, 2022-06-17:
Any high-level issues with Zigbee devices such as this really belong in the repository for "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” instead of here in zigpy repository which is only for low-level Zigbee issues:

https://github.com/zigpy/zha-device-handlers/issues

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) provides some information about how/why quirks are used and points people to that project:

https://www.home-assistant.io/integrations/zha/#zha-exception-and-deviation-handling

https://www.home-assistant.io/integrations/zha/#reporting-issues

https://www.home-assistant.io/integrations/zha/#debug-logging

###  _ZHA EXCEPTION AND DEVIATION HANDLING_

_The ZHA implementation in Home Assistant relies on a library called “[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers)” to resolve issues with Zigbee devices that do not fully conform with the Zigbee standards. The few devices that deviate from the Zigbee specifications set by the [Zigbee Alliance](https://zigbeealliance.org/) may therefore require proper bug reports with debug logs from users to assistant the developers in writing custom [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/blob/dev/README.md) for all of a device functions to work properly with the ZHA integration._

_Such a custom “ZHA Device Handler” are Python scripts that internally are also referred to as a “quirk” because they fix “quirks”, like deviations from the standard specifications. ZHA Device Handles do this by transparently, acting as a translator, translating and converting non-compliant device messages and instead present them to the application as coming from a virtual compliant device. These ZHA Device Handlers for Home Assistant can thus be used to parse custom messages to and from Zigbee devices. The ZHA Device Handlers that are made can then be reused by all users in future versions of Home Assistant._

_The custom quirks implementations for zigpy implemented as ZHA Device Handlers for Home Assistant are a similar concept to that of [Hub-connected Device Handlers for the SmartThings Classics platform](https://docs.smartthings.com/en/latest/device-type-developers-guide/) as well as that of [Zigbee-Herdsman Converters (formerly Zigbee-Shepherd Converters) as used by Zigbee2mqtt](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html), meaning they are each virtual representations of a physical device that expose additional functionality that is not provided out-of-the-box by the existing integration between these platforms._

### _Reporting issues_

_When reporting issues, please provide the following information in addition to information requested by issue template:_

1. _Debug logs for the issue, see [debug logging](#debug-logging)_
2. _Model of Zigbee radio being used_
3. _If issue is related to a specific Zigbee device, provide device Zigbee signature. Signature is available at
**Configuration** -> **Devices & Services** -> **Zigbee Home Automation** (click **Configure**) -> **Devices** (pick your device) -> **Zigbee Device Signature**_

### _Debug logging_

_To enable debug logging for ZHA component and radio libraries, add the following [logger](/integrations/logger/) configuration to `configuration.yaml`:_

```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```

PS: Also, for ZHA general end-user community support, see here -> https://community.home-assistant.io/c/configuration/zha/48

Hedda, 2022-06-17:
> ZHA sends events only by one key (qbkg03lm), the second key is simply ignored and not fixed in zha_event

Note that you will need a much better description which explain more information about the device and how to replicate issue, however it does sound as if it might maybe be related to https://github.com/zigpy/zha-device-handlers/issues/713 and https://github.com/zigpy/zha-device-handlers/pull/715

closed 2022-06-17 by Shurik85