zigpy/zigpy #1018: General commands on client cluster

ngillen, 2022-07-26:
Hi,

I need to  discover and read attributes on device's client cluster.
I manage to send general command to a server cluster and even to send general command to a client cluster with `cluster.reply(True, 0x0C, foundation.GENERAL_COMMANDS[0x0C].schema, 0, 200, manufacturer=manufacturer_code)`. However, I don't have any response on zigpy even if the device does respond (verified with a capture).
Is there an "easy" way to perform those general commands on client clusters ?

Thanks!

puddly, 2022-07-26:
What coordinator are you using?

ngillen, 2022-07-26:
I use a dongle with Silabs EFR32, so I use bellows on top of zigpy.

puddly, 2022-07-26:
Try this:

```Python
from zigpy.clusters.general import Scenes

cluster = device.endpoints[1].out_clusters[Scenes.cluster_id]

await cluster.discover_attributes(
    start_attribute_id=0x0000,
    max_attribute_ids=0xFF,
)
```

It works for me:
```Python
Discover_Attributes_rsp(
    discovery_complete=<Bool.true: 1>,
    attribute_info=[
        DiscoverAttributesResponseRecord(attrid=0, datatype=32),
        DiscoverAttributesResponseRecord(attrid=1, datatype=32),
        DiscoverAttributesResponseRecord(attrid=2, datatype=33),
        DiscoverAttributesResponseRecord(attrid=3, datatype=16),
        DiscoverAttributesResponseRecord(attrid=4, datatype=24),
        DiscoverAttributesResponseRecord(attrid=65533, datatype=33)
    ]
)
```

ngillen, 2022-07-27:
Thanks for your response.
In fact, I want to discover attributes on a remote product (08:6b:d7:ff:fe:d2:71:eb) from my dongle (cc:cc:cc:ff:fe:9f:6a:7f).
Here are my devices:
```
Device:
  NWK: 0x0000
  IEEE: cc:cc:cc:ff:fe:9f:6a:7f
  Endpoints:
    1: profile=0x104, device_type=DeviceType.IAS_CONTROL
      Input Clusters:
        Basic (0)
        On/Off (6)
        Time (10)
        Ota (25)
        IAS Ancillary Control Equipment (1281)
      Output Clusters:
        Power Configuration (1)
        Poll Control (32)
        IAS Zone (1280)
        IAS Warning Device (1282)
Device:
  NWK: 0xd1a8
  IEEE: 08:6b:d7:ff:fe:d2:71:eb
  Endpoints:
    1: profile=0x104, device_type=DeviceType.ON_OFF_LIGHT
      Input Clusters:
        Basic (0)
        Identify (3)
        Groups (4)
        Scenes (5)
        On/Off (6)
        Manufacturer Specific (64513)
        Manufacturer Specific (64768)
      Output Clusters:
        Ota (25)
    242: profile=0xa1e0, device_type=102
      Input Clusters:
        GreenPowerProxy (33)
      Output Clusters:
        GreenPowerProxy (33)
```
So, when I write something like:
```python
from zigpy.zcl.clusters.general import Ota

cluster = self.devices[t.EmberEUI64.convert("08:6b:d7:ff:fe:d2:71:eb")].endpoints[1].out_clusters[Ota.cluster_id]

await cluster.discover_attributes(
    start_attribute_id=0x0000,
    max_attribute_ids=0xFF,
)
```
It sends a frame with Direction: Client to Server in Frame Control Field. However, it should be a Server to Client as Ota is in out_clusters. That's why the device responds with no attributes.

puddly, 2022-07-31:
> However, it should be a Server to Client as Ota is in out_clusters. 

I think you're right!

The following patch makes the discovery request to the OTA cluster work properly:

```diff
diff --git a/zigpy/zcl/__init__.py b/zigpy/zcl/__init__.py
index c90d29b..8259931 100644
--- a/zigpy/zcl/__init__.py
+++ b/zigpy/zcl/__init__.py
@@ -276,10 +276,17 @@ class Cluster(util.ListenableMixin, util.CatchingTaskMixin):
         if tsn is None:
             tsn = self._endpoint.device.application.get_sequence()

+        # Set the packet header `direction` to match the cluster direction
+        is_reply = self in self.endpoint.out_clusters.values()
+
         if general:
-            hdr = foundation.ZCLHeader.general(tsn, command_id, manufacturer)
+            hdr = foundation.ZCLHeader.general(
+                tsn, command_id, manufacturer, is_reply=is_reply
+            )
         else:
-            hdr = foundation.ZCLHeader.cluster(tsn, command_id, manufacturer)
+            hdr = foundation.ZCLHeader.cluster(
+                tsn, command_id, manufacturer, is_reply=is_reply
+            )

         self.debug("Sending request header: %r", hdr)
         self.debug("Sending request: %r", request)
```

ngillen, 2022-07-31:
The patch make all commands work as expected :relieved:
Thank you for your help! :+1:

PR #1021 (2022-08-01): Set the request header's `direction` bit according to the cluster type
Fixes #1018

I'm a little wary of merging this change without more runtime testing, especially with quirks. Are there any known devices that rely on the current behavior to work properly?
closed 2022-08-01 by puddly