zigpy/zigpy #1019: "'OnOffChannel' object has no attribute '_state'"

Ulrar, 2022-07-26:
Hi,

I've noticed this repeating sometimes in my HA logs, it appears to be coming from Zigpy : `/usr/local/lib/python3.10/site-packages/zigpy/util.py:54`

```
Error calling listener <bound method OnOffChannel.cluster_command of <homeassistant.components.zha.core.channels.general.OnOffChannel object at 0x150cd070cc70>> with args (38, 66, on_with_timed_off(on_off_control=1, on_time=1800, off_wait_time=0)): AttributeError("'OnOffChannel' object has no attribute '_state'")
```

I have no idea of what's causing this as it doesn't seem to log the device address or message that caused it.
Any tips on how to figure out what's causing this ?

Thanks

dmulcahey, 2022-07-26:
Looks like a HA bug. Can you open an issue please in the HA repo?

Ulrar, 2022-07-27:
@dmulcahey Sure, I opened one now. Thanks

dmulcahey, 2022-07-27:
Thanks. I’ll look at it Friday when I return from vacation.

closed 2022-07-27 by dmulcahey