zigpy/zigpy #1036: Unexpected Error during Initializing Endpoints

prairiesnpr, 2022-08-26:
Custom device fails to join, with the following error.

```
2022-08-25 15:36:02.324 WARNING (MainThread) [zigpy.device] Device failed to initialize due to unexpected error
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 177, in initialize
    await self._initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/util.py", line 130, in retry
    r = await func()
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 233, in _initialize
    await ep.initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 73, in initialize
    self.profile_id = sd.profile
AttributeError: 'NoneType' object has no attribute 'profile'
```

It looks like we can end up in a situation where the simple descriptor request returns successful but without a value.  This causes the previous error since zigpy expects an object and not None.

Home Assistant install:
 - Raspberry Pi 3
 - Home Assistant 2022.8.6
 - Supervisor 2022.08.3
 - Operating System 8.5
 - Frontend 20220802.0 - latest

End device:
 - Garage controller with an xbee 3 connected to an Arduino pro to handle the ZHA endpoints.
 - Root cause is likely tied to this device not correctly responding to the simple descriptor, the device would join to earlier home assistant installs and functions correctly on another identical setup where the install has been updated over time. Error is isolated to the join.

Coordinator: 
 - xbee 3

Full Join Logs:
https://pastebin.com/6cvzwd5Y

Adminiuga, 2022-08-27:
Does the device answer to the simple descriptor request?  Can you make it answer?

prairiesnpr, 2022-09-19:
> Does the device answer to the simple descriptor request? Can you make it answer?

Sorry, just saw this.  No, it's not answering the simple descriptor request on every endpoint and yes, the solution is to rebuild the device so it will.  Full join attempt logs are here, https://pastebin.com/6cvzwd5Y

Just felt that we should probably ensure we had a response before attempting to parse it in zigpy.

closed 2023-03-25 by github-actions[bot]