zigpy/zigpy #1041: Possible rounding issue

TheJulianJES, 2022-09-01:


puddly, 2022-09-01:
All of the `active_power` attributes have type `int16s`. It could be that the value is converted from a float into the correct data type *somewhere*.

puddly, 2022-12-29:
@TheJulianJES is there a specific device that replicates this problem? I tried manually reading the `active_power` attribute of an Inovelli switch and `87` was successfully converted to `8.7` in the frontend.

TheJulianJES, 2022-12-29:
I think the issue only happens when using a ``LocalDataCluster`` and giving it a value like ``99.9``.
I've confirmed the issue with an Aqara MAEU01 plug (same as https://github.com/home-assistant/core/issues/84084 which was closed due to believing it was caused by a reset of the plug which it likely wasn't, as the state only reset a bit and not to complete 0).
I've had the following in my logs: "Triggered by state 26.049 (26.05)" **(so resetting from 26.05 to 26.049)**

The following code calls the ``_update_attribute`` method in ``zigpy``:
https://github.com/zigpy/zha-device-handlers/blob/a6556fea6ab6400984094dcc88d7bd772e322e97/zhaquirks/xiaomi/__init__.py#L624

The ``value * 1000`` still has decimals at this point. In the code snippet posted in OP, (IIRC) the value is correctly rounded for the part that causes the sensor to update immediately (without polling the fake cluster). However, the part where it's saved to the ``LocalDataCluster`` does not round correctly and just cuts off everything behind the decimal point.

So let's assume that ``value * 1000 = 1.9``. As soon as ``_update_attribute`` is called with ``1.9``,  the sensor gets correctly set to ``1.9``. However, when restarting (Z)HA, it'll poll ``1`` from the ``LocalDataCluster`` (causing the warning for total sum sensors). As soon as it gets another attribute report, it's set back to ``1.9`` though.

Actually, it might have been the other way around to or where the HA sensor does the rounding instead of "the part that updates the sensor". (Looking at the bold message at the top (26.05 -> 26.049))

Still, some ZHA part does not correctly round the sensor in all cases.

I've considered just rounding the value in the quirks before sending it to ``_update_attribute`` in the past, (so that ``value * 1000`` has no decimal points) but that seems like a hack (and it's an issue for other devices too (possibly only with a ``LocalDataCluster`` though). (Heh, maybe just explicitly round again in ``_:update_attribute``)

TheJulianJES, 2022-12-29:
So just to summarize what I believe the issue is:

Calling ``_update_attribute`` below with a ``value`` like ``1.9`` (so a decimal number) correctly rounds the number in some places (to ``2.0``) and just cuts off the ``.9`` in others places (so to ``1.0``).
https://github.com/zigpy/zigpy/blob/40e470454c7c7ecc167aa83b9aae72e5b0f9f28b/zigpy/zcl/__init__.py#L762-L764
(One place is where sensors are notified immediately in HA, the other place is likely setting the attribute in the db (and then polling it which shows another value).
Where the cut-off happens vs where the proper rounding happens I'm not sure after this time anymore. I believe the cut-off might have happened when reading/polling the "fake data attribute".)

Should ``_update_attribute`` ever be called with ``value`` being a decimal number though? If not, just round it there?

puddly, 2022-12-29:
Do you think I can replicate this if I buy a `lumi.plug.maus01` device?

This is the only logic I can find in ZHA for rounding: https://github.com/home-assistant/core/blob/0defe97892e088c40f08231e826c3e4b6cead255/homeassistant/components/zha/sensor.py#L280-L287

Tracing through the quirks code in `XiaomiCluster._update_attribute`, the raw value from the Aqara heartbeat is passed straight through to `consumption_reported` (https://github.com/zigpy/zha-device-handlers/blob/a6556fea6ab6400984094dcc88d7bd772e322e97/zhaquirks/xiaomi/__init__.py#L276-L279), which then multiplies it by 1000 and emits the `attribute_updated` event. The only place rounding takes place in this entire pipeline is in ZHA.

I don't think it's possible, but could the application database be implicitly truncating numbers? You should be able to see this on startup, printed in the `[0x%04x:%s:0x%04x] Attribute id: %s value: %s` logs.

TheJulianJES, 2022-12-29:
> Do you think I can replicate this if I buy a lumi.plug.maus01 device?

Actually, I don't think that device is working correctly with the current quirks. From what I've read in other issues, I'm also not sure if it reports differently (only using AnalogClusters instead of the special Xiaomi messages).
(and you probably shouldn't have to buy a device just to fix an issue)

> I don't think it's possible, but could the application database be implicitly truncating numbers? You should be able to see this on startup, printed in the [0x%04x:%s:0x%04x] Attribute id: %s value: %s logs.

I'll test this and report back.

> I tried manually reading the active_power attribute of an Inovelli switch and 87 was successfully converted to 8.7 in the frontend.

Just to confirm, in this case ``87`` is the raw ``active_power`` attribute value also seen when reading it through the clusters UI, while ``8.7`` is the value displayed by the HA sensor entity?

TheJulianJES, 2022-12-29:
Actually, now I've had the following (with ``active_power``, as it's easier to reproduce than with the total sum sensor):

The Aqara MAEU01 plug sometimes sends attribute reports via the AnalogInputCluster (when significant power changes happen) (forgot about this previously but it also calls the same ``_update_attribute`` method):
https://github.com/zigpy/zha-device-handlers/blob/a6556fea6ab6400984094dcc88d7bd772e322e97/zhaquirks/xiaomi/__init__.py#L542-L555
https://github.com/zigpy/zha-device-handlers/blob/a6556fea6ab6400984094dcc88d7bd772e322e97/zhaquirks/xiaomi/__init__.py#L587-L589

This looks like the following in the logs:

```yaml
2022-12-29 23:53:31.652 DEBUG (MainThread) [zigpy.zcl] [0x322A:21:0x000c] Decoded ZCL frame: AnalogInputCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=8.26200008392334))])
2022-12-29 23:53:31.652 DEBUG (MainThread) [zigpy.zcl] [0x322A:21:0x000c] Received command 0x0A (TSN 14): Report_Attributes(attribute_reports=[Attribute(attrid=0x0055, value=TypeValue(type=Single, value=8.26200008392334))])
2022-12-29 23:53:31.653 DEBUG (MainThread) [zigpy.zcl] [0x322A:21:0x000c] Attribute report received: present_value=8.26200008392334
```

The HA sensor displayed ``8.3W`` (so rounded up)

However, as soon as I polled the ``active_power`` (which is a fake/data attribute) using the clusters UI, it returned ``82`` there and the sensor in HA immediately jumped to ``8.2W``.

If I wouldn't have polled it manually, at least for ``active_power`` this also happens every 30 seconds (and on startup):
```yaml
2022-12-29 23:48:27.001 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x322A:1:0x0b04]: async_update
2022-12-29 23:48:27.001 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x322A:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
```

Still testing this (and will look at the DB stuff) and report back

TheJulianJES, 2022-12-29:
Restarting ZHA just shows the following after restarting when I previously read/polled the attribute:
```yaml
[zigpy.appdb] [0x322a:1:0x0b04] Attribute id: 1291 value: 83
```

However, when I immediately shutdown HA, as soon as I get an attribute report from the ``AnalogInputCluster`` (``POWER_REPORTED`` event -> ``_update_attribute``) and don't poll/read the attribute before and then restart HA, I get the following in my logs:
```yaml
[0x322a:1:0x0b04] Attribute id: 1291 value: 85.41999816894531
```

TheJulianJES, 2022-12-29:
Here's an attribute report from the ``AnalogInputCluster`` coming in with db logging enabled: https://dpaste.org/Ebbz2
Seems to keep the number completely there.

Maybe some rounding/cut-off is done when reading from a ``LocalDataCluster``?

TheJulianJES, 2022-12-29:
So I've added some debug logs in zigpy's ``_update_attribute`` and in zhaquirks' `LocalDataCluster`` in ``write_attributes`` and ``read_attributes_raw``:

When the attribute report from the ``AnalogInputCluster`` is received (just look at the ``1291/active_power`` attribute):
```yaml
2022-12-30 00:23:10.773 DEBUG (MainThread) [zigpy.zcl] [0x322A:21:0x000c] Attribute report received: present_value=8.128999710083008
2022-12-30 00:23:10.773 WARNING (MainThread) [zigpy.zcl] [0x322A:21:0x000c] (TEST LOGS) Attribute updated (id = value): 85 = 8.128999710083008
2022-12-30 00:23:10.774 WARNING (MainThread) [zigpy.zcl] [0x322A:21:0x000c] (TEST LOGS) Attribute cache is now: {85: 8.128999710083008}
2022-12-30 00:23:10.775 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute updated (id = value): 1291 = 81.28999710083008
2022-12-30 00:23:10.776 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute cache is now: {1291: 81.28999710083008, 1285: 241, 772: 3.489508992061019, 1541: 10, 1540: 1, 1027: 1, 1026: 1}
```

After some time, ZHA polls the ``active_power`` (and ``rms_voltage``).

```yaml
2022-12-30 00:23:19.155 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x322A:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_voltage']
2022-12-30 00:23:19.156 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Reading attribute in LocalDataCluster id: 1291 value from _attr_cache: 81.28999710083008
2022-12-30 00:23:19.156 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Reading attribute in LocalDataCluster id: 1285 value from _attr_cache: 241
2022-12-30 00:23:19.156 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute updated (id = value): 1291 = 81
2022-12-30 00:23:19.157 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute cache is now: {1291: 81, 1285: 241, 772: 3.489508992061019, 1541: 10, 1540: 1, 1027: 1, 1026: 1}
2022-12-30 00:23:19.157 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute updated (id = value): 1285 = 241
2022-12-30 00:23:19.158 WARNING (MainThread) [zigpy.zcl] [0x322A:1:0x0b04] (TEST LOGS) Attribute cache is now: {1291: 81, 1285: 241, 772: 3.489508992061019, 1541: 10, 1540: 1, 1027: 1, 1026: 1}
```

However, when it "re-updates" the database (during the whole reading/polling process), it looses the numbers after the decimal point. (I'll try to see from where this is called from)

puddly, 2022-12-29:
I think I can see how this can happen.

`LocalDataCluster` indeed uses the attribute cache to prevent attribute "reads" from actually hitting the device, but it overrides `read_attributes_raw`, not `read_attributes`, which causes the attribute read to continue on through zigpy:

https://github.com/zigpy/zigpy/blob/40e470454c7c7ecc167aa83b9aae72e5b0f9f28b/zigpy/zcl/__init__.py#L471-L481

The attribute type is currently this:

https://github.com/zigpy/zigpy/blame/40e470454c7c7ecc167aa83b9aae72e5b0f9f28b/zigpy/zcl/clusters/general.py#L1150-L1152

I think somewhere, in some quirk definition, the type is not `t.Single`.

TheJulianJES, 2022-12-29:
I guess it's just ``int16s``, as ``active_power`` normally can't have decimals (but here it does because of the quirk and what the plug sends):
https://github.com/zigpy/zigpy/blob/e1a7dd91abc6109bc4757781017c629530059e24/zigpy/zcl/clusters/homeautomation.py#L249

So, should all quirks just override this (if that's even possible)?

puddly, 2022-12-30:
Ahh, that explains it!

So it looks like the max wattage of the plug is 2300W. The maximum value of a `uint16s` data type is 32767, so I think a potential fix would be to hard-code `ac_power_divisor = 10` for all these devices and then internally store `active_power = round(present_value * 10)`. That way, the exposed interface is still ZCL compliant while still preserving some decimal places.

---

So as with other quirk stuff, we're running into limitations by treating the ZCL as the public interface. I'd hate to duplicate the entire Home Assistant device interface into quirks, but with the model matching and such already leaking into ZHA's codebase, I think that may be the best solution...

TheJulianJES, 2022-12-30:
> ac_power_divisor = 10 for all these devices and then internally store active_power = round(present_value * 10

Sounds good! ``ac_power_divisor = 10`` was already added for the Aqara plugs, so they get a decimal point in HA.
So just the rounding is needed there I guess to fix the issue.

But should there be some validation in zigpy to make sure that attributes can't be saved with the wrong datatype?
(Or just add the "type conversion" when writing attributes to the LocalDataCluster in zhaquirks?)

puddly, 2022-12-30:
> But should there be some validation in zigpy to make sure that attributes can't be saved with the wrong datatype?

The use of `_update_attribute` within quirks is a problem. It bypasses any logic within zigpy to do type conversion. I propose changing every use of `_update_attribute` to `update_attribute` within quirks and then updating `update_attribute` to enforce the attribute type.

Not sure how well this will work with "fake" attributes.

puddly, 2022-12-30:
To debug with existing quirks and to see what will break, maybe something like this:

```diff
diff --git a/zigpy/zcl/__init__.py b/zigpy/zcl/__init__.py
index 7e5744e..c1b70e0 100644
--- a/zigpy/zcl/__init__.py
+++ b/zigpy/zcl/__init__.py
@@ -760,6 +760,21 @@ class Cluster(util.ListenableMixin, util.CatchingTaskMixin):
         self._update_attribute(attrid, value)

     def _update_attribute(self, attrid, value):
+        try:
+            attrdef = self.find_attribute(attrid)
+        except KeyError:
+            LOGGER.warning("Unknown attribute %s=%r", attrid, value)
+        else:
+            attrid = attrdef.id
+            new_value = attrdef.type(value)
+
+            if new_value != value:
+                LOGGER.warning(
+                    "Attribute %s=%r converted to %r", attrid, value, new_value
+                )
+
+            value = new_value
+
         self._attr_cache[attrid] = value
         self.listener_event("attribute_updated", attrid, value)
```

This would also let you call `update_attribute("string_name", 0x1234)`, without having to pass around the numerical ID. For release, the warnings would be errors.

TheJulianJES, 2022-12-30:
Yep, properly converts the values:

```python
Attribute ZCLAttributeDef(id=0x050B, name='active_power', type=<class 'zigpy.types.basic.int16s'>, access=<ZCLAttributeAccess.Report|Read: 9>, mandatory=False, is_manufacturer_specific=False)=78.75 converted to 78
Attribute ZCLAttributeDef(id=0x0000, name='current_summ_delivered', type=<class 'zigpy.types.basic.uint48_t'>, access=<ZCLAttributeAccess.Read: 1>, mandatory=False, is_manufacturer_specific=False)=6.414828356355429 converted to 6
Attribute ZCLAttributeDef(id=0x0304, name='total_active_power', type=<class 'zigpy.types.basic.int32s'>, access=<ZCLAttributeAccess.Report|Read: 9>, mandatory=False, is_manufacturer_specific=False)=6.414828356355429 converted to 6
```

And adding ``round()`` for those three attributes removes the warnings (and properly rounds them) as expected

closed 2023-09-14 by github-actions[bot]