zigpy/zigpy #1055: backups.create_backup() seems missing information

pipiche38, 2022-10-01:
I'm struggling with the backup implementation and the end result. Am I'm missing something before calling create_backup()

Here is what I got when calling : `backups.create_backup() 
`

> 2022-10-01 18:00:14,868 DEBUG   :Adding a new backup NetworkBackup(backup_time=datetime.datetime(2022, 10, 1, 16, 0, 14, 868146, tzinfo=datetime.timezone.utc), network_info=NetworkInfo(extended_pan_id=67:34:48:45:04:01:5e:10, pan_id=4783, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.CHANNEL_15: 32768>, security_level=5, network_key=Key(key=84:66:24:84:0d:2c:90:ee:60:17:25:a2:32:97:6a:10, tx_counter=1357942, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), key_table=[], children=[], nwk_addresses={}, stack_specific={'zstack': {'tclk_seed': '5209d0aba562e876c3aac2c1ec23032f'}}, metadata={'zstack': {'TransportRev': 2, 'ProductId': 1, 'MajorRel': 2, 'MinorRel': 7, 'MaintRel': 1, 'CodeRevision': 20211217, 'BootloaderBuildType': <BootloaderBuildType.NON_BOOTLOADER_BUILD: 0>, 'BootloaderRevision': None}}, source='zigpy-znp@0.8.2'), node_info=NodeInfo(nwk=0x0000, ieee=00:12:4b:00:24:c1:69:02, logical_type=<LogicalType.Coordinator: 0>))
> 2022-10-01 18:00:14,871 DEBUG   :Replacing NetworkBackup(backup_time=datetime.datetime(2022, 10, 1, 15, 55, 41, 130147, tzinfo=datetime.timezone.utc), network_info=NetworkInfo(extended_pan_id=67:34:48:45:04:01:5e:10, pan_id=4783, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.CHANNEL_15: 32768>, security_level=5, network_key=Key(key=84:66:24:84:0d:2c:90:ee:60:17:25:a2:32:97:6a:10, tx_counter=1357942, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), key_table=[], children=[], nwk_addresses={}, stack_specific={'zstack': {'tclk_seed': '5209d0aba562e876c3aac2c1ec23032f'}}, metadata={'zstack': {'TransportRev': 2, 'ProductId': 1, 'MajorRel': 2, 'MinorRel': 7, 'MaintRel': 1, 'CodeRevision': 20211217, 'BootloaderBuildType': <BootloaderBuildType.NON_BOOTLOADER_BUILD: 0>, 'BootloaderRevision': None}}, source='zigpy-znp@0.8.2'), node_info=NodeInfo(nwk=0x0000, ieee=00:12:4b:00:24:c1:69:02, logical_type=<LogicalType.Coordinator: 0>))
> 

While when running the `zigpy_znp.tools.network_backup`, I got all devices

`[domoticz@rasp Domoticz-Zigbee]$ python3 -m zigpy_znp.tools.network_backup /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_a03c7d0a9329ec119a4b6d7840c9ce8d-if00-port0`

> 2022-10-01 18:04:53.449 rasp zigpy_znp.znp.security WARNING Skipping hashed link key 36:f4:22:2c:2f:ee:1c:d0:5c:bd:63:c7:76:7f:bf:c2 (tx: 6899, rx: 0) for unknown device 00:15:bc:00:2f:01:18:f7
> {
>     "metadata": {
>         "version": 1,
>         "format": "zigpy/open-coordinator-backup",
>         "source": "zigpy-znp@0.8.2",
>         "internal": {
>             "creation_time": "2022-10-01T18:04:56+02:00",
>             "zstack": {
>                 "version": 3.3
>             }
>         }
>     },
>     "coordinator_ieee": "00124b0024c16902",
>     "pan_id": "12af",
>     "extended_pan_id": "6734484504015e10",
>     "nwk_update_id": 0,
>     "security_level": 5,
>     "channel": 15,
>     "channel_mask": [
>         15
>     ],
>     "network_key": {
>         "key": "846624840d2c90ee601725a232976a10",
>         "sequence_number": 0,
>         "frame_counter": 1357942
>     },
>     "devices": [
>         {
>             "ieee_address": "00047400001f09a4",
>             "nwk_address": "ea8a",
>             "is_child": false,
>             "link_key": {
>                 "key": "f600cfaba516ec7667a3ddc1ec57072f",
>                 "tx_counter": 2178,
>                 "rx_counter": 0
>             }
>         },
>         {
>             "ieee_address": "000474000082a54f",
>             "nwk_address": "37d3",
>             "is_child": false,
>             "link_key": {
>                 "key": "8c0f40c1ec57072f1dac52aba516ec76",
>                 "tx_counter": 2970,
>                 "rx_counter": 0
>             }
>         },
>         {
>             "ieee_address": "00047400008aff8b",
>             "nwk_address": "31cd",
>             "is_child": false,
>             "link_key": {
>                 "key": "5b542f62e802c7aa493e6623035b5609",
>                 "tx_counter": 2970,
>                 "rx_counter": 0
>             }
[... 163 lines cut ...]
>             "nwk_address": "84b5",
>             "is_child": false,
>             "link_key": {
>                 "key": "5bd2a5bb784e020e49b8ecfa931793ad",
>                 "tx_counter": 7370,
>                 "rx_counter": 0
>             }
>         },
>         {
>             "ieee_address": "a4c138ca71bfc636",
>             "nwk_address": "a147",
>             "is_child": false,
>             "link_key": {
>                 "key": "3f1614d4a8d0b7679c047e9de93beef6",
>                 "tx_counter": 11715,
>                 "rx_counter": 0
>             }
>         }
>     ],
>     "stack_specific": {
>         "zstack": {
>             "tclk_seed": "5209d0aba562e876c3aac2c1ec23032f"
>         }
>     }
> }

puddly, 2022-10-01:
It has a `load_devices` keyword argument:

https://github.com/zigpy/zigpy/blob/3999ef1a33babe86c6cf256ce44412ff164fdfae/zigpy/backups.py#L127

closed 2022-10-01 by puddly

pipiche38, 2022-10-01:
Thanks, but should I enable it or not for a proper backup / restore ? Why is it not True by default ?

puddly, 2022-10-01:
Yes. It's disabled by default because the backup takes too long to do synchronously and isn't strictly necessary to restore a network.