zigpy/zigpy #1057: Issue sending command with parameter

mdeweerd, 2022-10-02:
I tried to send a Trigger Effect command to a device using zha-toolkit which failed, so I tried using the zha.zcl_cmd service in home assistant which failed the same way.

# Try 1
```yaml
service: zha.issue_zigbee_cluster_command
data:
  ieee: 60:a4:23:ff:fe:91:fc:9a
  endpoint_id: 1
  cluster_id: 3
  command: 64
  command_type: server
  args: 1
```

result:

```plaintext
2022-10-02 15:52:10.514 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Value for field 'effect_variant' is required
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 447, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 680, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 746, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 1273, in issue_zigbee_cluster_command
    response = await zha_device.issue_cluster_command(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 677, in issue_cluster_command
    response = await cluster.command(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 310, in request
    hdr, request = self._create_request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 271, in _create_request
    request.serialize()  # Throw an error before generating a new TSN
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 209, in serialize
    for field, value in self.assigned_fields(strict=True):
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 177, in assigned_fields
    raise ValueError(f"Value for field {field.name!r} is required")
ValueError: Value for field 'effect_variant' is required
2022-10-02 15:52:10.548 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139695561028704] Error handling message: Unknown error (unknown_error)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 646, in handle_execute_script
    await script_obj.async_run(msg.get("variables"), context=context)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1513, in async_run
    await asyncio.shield(run.async_run())
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 405, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 449, in _async_step
    self._handle_exception(
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 472, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 447, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 680, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 746, in admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/api.py", line 1273, in issue_zigbee_cluster_command
    response = await zha_device.issue_cluster_command(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/device.py", line 677, in issue_cluster_command
    response = await cluster.command(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 310, in request
    hdr, request = self._create_request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 271, in _create_request
    request.serialize()  # Throw an error before generating a new TSN
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 209, in serialize
    for field, value in self.assigned_fields(strict=True):
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 177, in assigned_fields
    raise ValueError(f"Value for field {field.name!r} is required")
ValueError: Value for field 'effect_variant' is required
```

# Try 2:
```yaml
service: zha_toolkit.zcl_cmd
data:
  ieee: light.tz3000_dbou1ap4_ts0505a_9afc91fe_level_light_color_on_off
  cluster: 3
  cmd: 64
  args: [1]
```

result:
```plaintext
ValueError: Value for field 'effect_variant' is required```
```

# Other methods:
```yaml
service: zha_toolkit.zcl_cmd
data:
  ieee: light.tz3000_dbou1ap4_ts0505a_9afc91fe_level_light_color_on_off
  cluster: 3
  cmd: 64
  args:
    - effect_variant: 1
```

essentially results in:
```plaintext
ValueError: Failed to convert effect_id={'effect_variant': 1} from type <class 'dict'> to <enum 'EffectIdentifier'>
```

I think in the past at least the second method would have worked.

puddly, 2022-10-03:
Does your device have an applied quirk? If so, this has been fixed by https://github.com/zigpy/zigpy/pull/1056.

mdeweerd, 2022-10-03:
The device I tried does have a quirk.

So I tested another device that does not have a quirk.  The end result is unfortunately the same.

Here are the different errors depending on the "args" parameter (single value, list of single value, dictionary).

```
ValueError: Value for field 'effect_variant' is required
ValueError: Failed to convert effect_id={'trigger_effect': 1} from type <class 'dict'> to <enum 'EffectIdentifier'>
```

puddly, 2022-10-03:
I believe you need to pass both arguments:

```yaml
args: [0x00, 0x00]
```

mdeweerd, 2022-10-03:
@puddly Sorry, apparently I was looking at the wrong table which I took for the list of the options, which it was a list of possible values.

I tried with two values and that worked.

closed 2022-10-03 by mdeweerd