zigpy/zigpy #1059: self._device.application._dblistener is None

naive-HA, 2022-10-03:
zigpy/endpoint.py/add_input_cluster line 108 checks hasattr(self._device.application, "_dblistener")
but never checks if it is None and even if None it passes None to line 109 
listener = zigpy.zcl.ClusterPersistingListener(self._device.application._dblistener, cluster) 
which triggers an exception every time an attribute is updated via 
zigpy/zcl/__init__.py/ClusterPersistingListener/attribute_updated
which is called by
zigpy/util.py/listener_event
which is called by 
zigpy/zcl/__init__.py/read_attributes
which is called by 
zigpy/endpoint.py/read_attributes
which is called every time an attribute is updated

naive-HA, 2022-10-03:
a quick and elegant fix would be to check if self._applistener is not None
in zigpy/zcl/__init__.py/ClusterPersistingListener/attribute_updated
line 907 before 
self._applistener.attribute_updated(self._cluster, attrid, value)

puddly, 2022-10-03:
In what scenario is `app._dblistener = None`? If you disable the zigpy database?

naive-HA, 2022-10-03:
I do not disable any database.
I have started using the code posted by you here:
https://github.com/zigpy/zigpy/issues/452#issuecomment-671441387
which now grew into this:
https://github.com/zigpy/zigpy-znp/issues/180

I have seen others complaining about the same error, when making use of the code you posted in
https://github.com/zigpy/zigpy/issues/452
Like:
https://github.com/zigpy/zigpy/issues/452#issuecomment-671806167

Am I doing something wrong?

naive-HA, 2022-10-03:
The error mentioned in https://github.com/zigpy/zigpy/issues/452#issuecomment-671806167
 Error calling listener.attribute_updated: 'NoneType' object has no attribute 'attribute_updated'
is caused by self._device.application._dblistener being None

puddly, 2022-10-03:
Ah, that is a mistake with the code example. It should be:

```python
app = await ControllerApplication.new({
    "database_path": "/path/to/zigbee.db",
    "device": {
        "path": "/dev/serial/by-id/your-serial-port",
    }
}, auto_form=True)

listener = MainListener(app)
app.add_listener(listener)
```

naive-HA, 2022-10-03:
that ain't it
in the code I am using I first instantiate the app object and then I startup via
await app.startup(auto_form=True)
See: https://github.com/zigpy/zigpy-znp/issues/180
although the indentation is not displayed correctly...
Just like in your code snippet: https://github.com/zigpy/zigpy/issues/452#issuecomment-671441387
So, the startup is awaited, no issues there. But self._device.application._dblistener is still None

this other way of instantiating the object and starting up/forming the network is also posted by you here:
https://github.com/zigpy/zigpy/issues/452#issuecomment-672026121
but my understanding is that these are equivalent as long as the startup is awaited

Am I wrong?

puddly, 2022-10-03:
They are not equivalent: https://github.com/zigpy/zigpy/blob/2a122e8ac61b1fa464813f1bdec94af33f388822/zigpy/application.py#L131-L150

naive-HA, 2022-10-03:
That is true
I have updated my code and all good

closed 2022-10-03 by naive-HA