zigpy/zigpy #1064: zigpy backup . Adding a new backup is incomplete, ignoring

pipiche38, 2022-10-03:
when trying to add a new backup, I'm getting an error message . What is wrong ?

            self.node_info.ieee != t.EUI64.UNKNOWN
            and self.network_info.extended_pan_id != t.EUI64.UNKNOWN
            and self.network_info.pan_id not in (0x0000, 0xFFFF) 
            and self.network_info.channel in range(11, 26 + 1)
            and self.network_info.network_key.key != t.KeyData.UNKNOWN

```
2022-10-03 18:53:43,549 DEBUG   :Adding a new backup NetworkBackup(backup_time="NetworkBackup(backup_time=datetime.datetime(2022, 10, 3, 16, 53, 26, 317009, tzinfo=datetime.timezone.utc), network_info=NetworkInfo(extended_pan_id=67:34:48:45:04:01:5e:10, pan_id=4783, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.CHANNEL_15: 32768>, security_level=5, network_key=Key(key=84:66:24:84:0d:2c:90:ee:60:17:25:a2:32:97:6a:10, tx_counter=1403201, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), key_table=[], children=[], nwk_addresses={}, stack_specific={'zstack': {'tclk_seed': '5209d0aba562e876c3aac2c1ec23032f'}}, metadata={'zstack': {'TransportRev': 2, 'ProductId': 1, 'MajorRel': 2, 'MinorRel': 7, 'MaintRel': 1, 'CodeRevision': 20211217, 'BootloaderBuildType': <BootloaderBuildType.NON_BOOTLOADER_BUILD: 0>, 'BootloaderRevision': None}}, source='zigpy-znp@0.8.2'), node_info=NodeInfo(nwk=0x0000, ieee=00:12:4b:00:24:c1:69:02, logical_type=<LogicalType.Coordinator: 0>))\n", network_info=NetworkInfo(extended_pan_id=ff:ff:ff:ff:ff:ff:ff:ff, pan_id=0xFFFE, nwk_update_id=0, nwk_manager_id=0x0000, channel=0, channel_mask=<Channels.NO_CHANNELS: 0>, security_level=0, network_key=Key(key=ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff, tx_counter=0, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), key_table=[], children=[], nwk_addresses={}, stack_specific={}, metadata={}, source=None), node_info=NodeInfo(nwk=0xFFFE, ieee=ff:ff:ff:ff:ff:ff:ff:ff, logical_type=<LogicalType.EndDevice: 2>))
2022-10-03 18:53:43,552 DEBUG   :Backup is incomplete, ignoring
```

puddly, 2022-10-03:
Please provide the entire debug log and an NVRAM dump.

pipiche38, 2022-10-03:
Here attached,

My understanding is that is_complete() is checking against those informations

```
        return (
            self.node_info.ieee != t.EUI64.UNKNOWN
            and self.network_info.extended_pan_id != t.EUI64.UNKNOWN
            and self.network_info.pan_id not in (0x0000, 0xFFFF)
            and self.network_info.channel in range(11, 26 + 1)
            and self.network_info.network_key.key != t.KeyData.UNKNOWN
        )
```
and 

'extended_pan_id': '67:34:48:45:04:01:5e:10',
'pan_id': '12AF'
'channel': 15,
'network_key': {'key': '84:66:24:84:0d:2c:90:ee:60:17:25:a2:32:97:6a:10'
'node_info': {'nwk': '0000', 'ieee': '00:12:4b:00:24:c1:69:02',

The log file is here
[ll.zip](https://github.com/zigpy/zigpy/files/9699535/ll.zip)

puddly, 2022-10-03:
Did you manually try to create backup objects and insert them into your database? Something is very broken:

```python
2022-10-03 19:16:33,940 INFO    :Last backup retreived: NetworkBackup(backup_time={'backup_time': '2022-10-03T17:16:15.989292+00:00', 'network_info': 
                                                                             ^^^^^^^^^^^
```

```python
NetworkBackup(
    backup_time="NetworkBackup(backup_time=datetime.datetime(2022, 10, 3, 16, 53, 26, 317009, tzinfo=datetime.timezone.utc), network_info=NetworkInfo(extended_pan_id=67:34:48:45:04:01:5e:10, pan_id=4783, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.CHANNEL_15: 32768>, security_level=5, network_key=Key(key=84:66:24:84:0d:2c:90:ee:60:17:25:a2:32:97:6a:10, tx_counter=1403201, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=00:12:4b:00:24:c1:69:02), key_table=[], children=[], nwk_addresses={}, stack_specific={'zstack': {'tclk_seed': '5209d0aba562e876c3aac2c1ec23032f'}}, metadata={'zstack': {'TransportRev': 2, 'ProductId': 1, 'MajorRel': 2, 'MinorRel': 7, 'MaintRel': 1, 'CodeRevision': 20211217, 'BootloaderBuildType': <BootloaderBuildType.NON_BOOTLOADER_BUILD: 0>, 'BootloaderRevision': None}}, source='zigpy-znp@0.8.2'), node_info=NodeInfo(nwk=0x0000, ieee=00:12:4b:00:24:c1:69:02, logical_type=<LogicalType.Coordinator: 0>))\n",
                ^^^ ???
    network_info=NetworkInfo(extended_pan_id=ff:ff:ff:ff:ff:ff:ff:ff, pan_id=0xFFFE, nwk_update_id=0, nwk_manager_id=0x0000, channel=0, channel_mask=<Channels.NO_CHANNELS: 0>, security_level=0, network_key=Key(key=ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff:ff, tx_counter=0, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=0, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), key_table=[], children=[], nwk_addresses={}, stack_specific={}, metadata={}, source=None),
    node_info=NodeInfo(nwk=0xFFFE, ieee=ff:ff:ff:ff:ff:ff:ff:ff, logical_type=<LogicalType.EndDevice: 2>),
)
```

pipiche38, 2022-10-03:
Ok, I'll investigate

closed 2022-10-03 by pipiche38