zigpy/zigpy #1067: Homeassistant zigbee  2022.10.0 not starting with zigpy error

guirem, 2022-10-06:
Hi, 

Zigbee Homeassistant component is no longer starting since latest upgrade yesterday on version 2022.10. 
I'm using a zigate dongle.

Hard to see from the logs if it originates int zigpy or zigpy-zigate as both got recent update.

- ZiGate USB-TTL dongle - 3.1d
- Home Assistant 2022.10.0
- Supervisor 2022.09.1
- Operating System 9.0
- Version requirements for hass 2022.10
```
# homeassistant.components.zha
zigpy-zigate==0.10.0
# homeassistant.components.zha
zigpy==0.51.2
```

I'm opening an issue in zigpy-zigate repo as well.

Screenshot below and full log attached.

[image]

[zigpy-bug.txt](https://github.com/zigpy/zigpy/files/9724012/zigpy-bug.txt)

guirem, 2022-10-06:
Associated zigpy-zigate issue: https://github.com/zigpy/zigpy-zigate/issues/133

puddly, 2022-10-06:
This is a problem with the ZiGate library, let's not duplicate issues across projects.

closed 2022-10-06 by puddly