zigpy/zigpy #1071: Application could not start (due to device timeout)

madchap, 2022-10-08:
Hi,

I am opening an issue here and this might not be the right place, apologies in advance. If you could then point me into the right direction.

I am running the latest to-date hassio version on a raspberry Pi 3+ (Home Assistant 2022.10.1).
I have 2 sonoff USB dongle bridges (1 'P' and 1 'E') linked and 1 UART for elec metering. The brigde used for zha is the following:

```
[ 1299.426977] usb 1-1.3: USB disconnect, device number 5
[ 1303.196120] usb 1-1.3: new full-speed USB device number 8 using dwc_otg
[ 1303.310562] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=55d4, bcdDevice= 4.42
[ 1303.310634] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1303.310656] usb 1-1.3: Product: SONOFF Zigbee 3.0 USB Dongle Plus V2
[ 1303.310675] usb 1-1.3: Manufacturer: ITEAD
[ 1303.310693] usb 1-1.3: SerialNumber: 20220813133528
```
Upon restart, I have to go unplug and plug it back it, then `Migrate radio` in order to reconfigure the zha integration with the device.

[image]

Some errors seen in the logs. 
```
2022-10-08 10:49:05.690 DEBUG (MainThread) [homeassistant.components.zha] ZHA storage file does not exist or was already removed
2022-10-08 10:49:14.043 ERROR (MainThread) [zigpy.application] Couldn't start application
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 124, in startup
    await self.connect()
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 124, in connect
    self._ezsp = await bellows.ezsp.EZSP.initialize(self.config)
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 106, in initialize
    await ezsp._startup_reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 97, in _startup_reset
    await self.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 126, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/uart.py", line 244, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
2022-10-08 10:49:16.785 WARNING (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis coordinator (atte
mpt 1 of 3)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 172, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 144, in new
    await app.startup(auto_form=auto_form)
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 124, in startup
    await self.connect()
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 124, in connect
    self._ezsp = await bellows.ezsp.EZSP.initialize(self.config)
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 106, in initialize
   await ezsp._startup_reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 97, in _startup_reset
    await self.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 126, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/uart.py", line 244, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
2022-10-08 10:49:16.785 WARNING (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis coordinator (atte
mpt 1 of 3)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 172, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 144, in new
    await app.startup(auto_form=auto_form)
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 124, in startup
    await self.connect()
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 124, in connect
    self._ezsp = await bellows.ezsp.EZSP.initialize(self.config)
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 106, in initialize
    await ezsp._startup_reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 97, in _startup_reset
    await self.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 126, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.10/site-packages/bellows/uart.py", line 244, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

Any idea?

Thank you.

madchap, 2022-10-08:
If I unplug the "P" dongle, then it seems to come up as expected.

puddly, 2022-10-08:
> Upon a full restart of the raspberryPI (or after a power outage), I have to go unplug and plug it back the Sonoff dongle, then Migrate radio in order to reconfigure the zha integration with the device.

Interesting. Can you upload the diagnostics information for the ZHA integration? It almost appears like the serial port was not expanded to its `/dev/serial/by-id/...` symlink when you set up ZHA.

closed 2022-11-01 by puddly

madchap, 2022-11-02:
argh, sorry, didn't see the notice before. I am assuming you'd still require the diag info? thanks.