zigpy/zigpy #1078: 3reality OTA provider for Third Reality branded Zigbee devices

Hedda, 2022-10-17:
Great to hear Third Reality (3reality) has joined "Works with Home Assistant" / "Works via Zigbee with Home Assistant" partner-program.

https://www.home-assistant.io/blog/2022/10/13/third-reality-partner/

https://www.home-assistant.io/blog/2022/07/12/partner-program/

Follow up question to that is if Third Reality (3reality) could spare development time to add an OTA provider for itself in the zigpy library?

As they probably do not know, zigpy is the Zigbee framework and hardware radio API/CLI abstraction library that is used as a dependency by a few popular home automation software and having "3reality Zigbee OTA provider" support could potentially enable easier OTA upgrades for end-users of those in the future, and include; [ZHA integration](https://www.home-assistant.io/integrations/zha) for Home Assistant today, as well as probably [Zigbee plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz) ([Domoticz-Zigbee](https://github.com/zigbeefordomoticz/Domoticz-Zigbee) project) and [Zigbee plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/) in the future.

See zigpy OTA provider code ([provider.py](https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py)) in Python which includes the URL links as source of Zigbee OTA files for direct download:

https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py

https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

The zigpy library would need a "Third Reality OTA Firmware provider" to handle image download (see "Skeleton OTA Firmware provider" for reference, as well as note [provider.py](https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py) also already has different code sections to handle the download of OTA files from IKEA, INOVELLI, SONOFF, and LEDVANCE as references).

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

Hopefully Third Reality will consider setting up an OTA provider server or creating a dedicated repository for all officially released Zigbee OTA files for Third Reality branded Zigbee devices?  

That way it will be a single trusted and official primary source of official Zigbee OTA files released by Third Reality, and that OTA provider source repository with your Zigbee OTA files could not only be used by zigpy as it could also be reused by other Zigbee implementations independently (both other open-source implementations as well as closed source and proprietary implementations).

Maybe also consider copying concept of `index.json` with metadata on all files like Koenkk repository has to make scripting easier:

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

zigpy and ZHA does not yet use this or other OTA Index files but [Zigbee2MQTT](https://www.zigbee2mqtt.io/) allows users to use it locally as described here:

https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/guide/usage/ota_updates.md#local-ota-index-and-firmware-files

and discussion in relation to zigpy implementation here:

https://github.com/zigpy/zigpy/issues/535

And of course `README.md` file which among other things could contain some basic information on how to update those:

https://github.com/Koenkk/zigbee-OTA/blob/master/README.md

Perhaps also as keeping some kind of `CHANGELOG.md` with changelog release history for each device firmware version:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

PS: By the way, please also see this somewhat related discussion regarding here Koenkk zigbee-OTA repository -> https://github.com/zigpy/zigpy/issues/535

pipiche38, 2022-10-17:
great news, and happy to integrate Third Reality  objects to the Zigbee for Domoticz plugin

puddly, 2022-11-11:
Implemented in #1096

closed 2022-11-11 by puddly

Hedda, 2023-04-04:
@hwzolin FYI, puddly is working on standard for "Generic OTA providers" + a new JSON format schema:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

  "**_This allows for new providers to be added to zigpy without extra code. This also allows for manufacturers to distribute test feed URLs to customers without deploying them globally._**"

This feature will be introduced via this new pull request that is still a work-in-progress. Please see ->  https://github.com/zigpy/zigpy/pull/1165