zigpy/zigpy #1087: Zigpy application examples

sergorl, 2022-10-31:
Hi, guys. Are there some examples like?:

1. Device enumeration: list all connected devices to my zigbee-adapter
2. Data reading from devices
3. Device command sending

13g10n, 2022-11-02:
+1 here

It will be extra useful to get some examples of basic flow when using as separate library (without HA and/or even quirks).

Hedda, 2022-11-17:
There is unfortunately only some basic info available here:

https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md

Also see this pull request "**_Documenting zigpy API a kind of developer guide_**" that was only partially copied to above document:

https://github.com/zigpy/zigpy/pull/477

Maybe you can submit PR to update API docs in [CONTRIBUTING.md](https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md) or post info in wiki? -> https://github.com/zigpy/zigpy/wiki

(Perhaps for additional inspiration also look at https://github.com/Koenkk/zigbee-herdsman#api-documentation ?).

There no other docs but also check through issues and discussions as some pointers have been given to others who asked, as ex:

https://github.com/zigpy/zigpy/discussions/709

https://github.com/zigpy/zigpy-znp/issues/108

https://github.com/zigpy/zigpy/issues/452

https://github.com/zigpy/zigpy/issues/469

https://github.com/zigpy/zigpy/issues/471 

https://github.com/zigpy/zigpy/issues/715

https://github.com/zigpy/zigpy/issues/7

https://github.com/zigpy/zigpy/discussions/595

as well as existing articles in the zigpy wiki:

https://github.com/zigpy/zigpy/wiki

Other than [ZHA code](https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha) as reference might also want to look at the code of the [Zigbee Plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz) and the [Zigbee Plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/) (competing open-source home automation software) which both partially implements [zigpy libraries](https://github.com/zigpy/) as dependencies:

https://github.com/zigbeefordomoticz/Domoticz-Zigbee/ (dev https://github.com/zigbeefordomoticz/Domoticz-Zigbee/tree/dev).

To get to code of  [Zigbee Plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/) you need to buy and install that plugin (which once installed will give you access to their Phyton code) as they do not distribute the plugin code directly via a Git repository. Recommend read their blog post:

https://blog.jeedom.com/5183-tout-ce-quil-faut-savoir-sur-le-plugin-officiel-zigbee/

Hedda, 2022-11-17:
Oh, also check out zigpy-cli and zha-toolkit

https://github.com/zigpy/zigpy-cli

https://github.com/mdeweerd/zha-toolkit/

FYI, dmulcahey also began working on a web socket [client–server model](https://en.wikipedia.org/wiki/Client%E2%80%93server_model) project but it looks like he at least paused that it for now:

https://github.com/zigpy/zha-websocket-server/

Think need both zhaws-addon + zhaws from fork for Home Assistant core with ZHA WS implementation as custom component:

https://github.com/zigpy/zhaws-addon

https://github.com/dmulcahey/home-assistant/tree/dm/zha-ws

https://github.com/dmulcahey/home-assistant/tree/dm/zha-ws/homeassistant/components/zhaws

garrethcain, 2024-01-06:
I'd be willing to write some stand-alone examples (not HA related) and even get documentation started if I could get some help to ramp up my understanding...

Hedda, 2024-01-12:
> I'd be willing to write some stand-alone examples (not HA related) and even get documentation started if I could get some help to ramp up my understanding...

@garrethcain FYI, probably not up to date but @pipiche38 did start to write a `zigpy_api.md` a few years ago, check out:

https://github.com/zigpy/zigpy/pull/477

https://github.com/zigpy/zigpy/issues/471

IMHO best approach to begin today might be to just start writing one or more short stand-alone examples as a small "draft" pull request in a new section for the `CONTRIBUTING.md` file as then you can get feedback to that pull reuqest from zigpy developers:

https://github.com/zigpy/zigpy/blob/dev/CONTRIBUTING.md

Then once you get started with a format that is acceptable to existing zigpy developers then you could submit new pull requests.

Hedda, 2024-08-14:
This issue/tracker and the high-level/application-level questions in it should now maybe/probably be moved to the repository for the new zha (zha gateway) library:

https://github.com/zigpy/zha

The zigpy library is a low-level library, while the new zha library is a high-level library that is event based and has a simpler API:

https://github.com/zigpy/zha/blob/dev/zha/application/gateway.py

That new zha library is actually parts of Home Assistant's ZHA component that have been split/broken-out as an stand-alone library, which is now what the ZHA integration in Home Assistant now depends on to use as an external Zigbee Gateway library:

https://github.com/home-assistant/core/tree/dev/homeassistant/components/zha

DataBeaver, 2024-12-03:
I wrote this during my adventures of getting zigpy to do stuff.  Perhaps it could be useful as an example program.  I can create a PR if you give some guidance of where the file should go.  Feedback on how to improve it is also welcome.
```py
import asyncio
import logging
import zigpy

# Change bellows to a different module if your radio does not use a
# Silicon Labs chip
from bellows.zigbee.application import ControllerApplication

# Some shortcut names for convenience
from zigpy.zcl.clusters.general import OnOff as OnOffCluster
from zigpy.zcl.clusters.general import LevelControl as LevelControlCluster

# Change these to suit your system
database_path = "zigbee.db"
device_path = "/dev/ttyUSB1"

# Note: You will need to get your zigbee devices into pairing mode in order
# for them to join the network.  The method for that varies by device.  The
# Zigbee2MQTT device database has useful information, for example:
#   https://www.zigbee2mqtt.io/devices/LED2201G8.html

class MainListener:
	"""Monitors new devices and adds them to the light switch"""

	def __init__(self, switch):
		self.switch = switch

	def device_initialized(self, dev):
		"""Called when a device is fully initialized and all of its endpoints
		and clusters are available for inspection."""

		self.switch.process_device(dev)

class OnOffListener:
	"""Listens to on/off commands from button devices and forwards them to
	the light switch object"""

	def __init__(self, switch):
		self.switch = switch

	def cluster_command(self, tsn, cmd, args):
		cmd_defs = OnOffCluster.ServerCommandDefs
		if cmd==cmd_defs.off.id:
			asyncio.create_task(self.switch.turn_off())
		elif cmd==cmd_defs.on.id:
			asyncio.create_task(self.switch.turn_on())

class LevelListener:
	"""Listens to level control commands from control devices and forwards them
	to the light switch object"""

	def __init__(self, switch):
		self.switch = switch

	def cluster_command(self, tsn, cmd, args):
		cmd_defs = LevelControlCluster.ServerCommandDefs
		if cmd==cmd_defs.move.id or cmd==cmd_defs.move_with_on_off.id:
			asyncio.create_task(self.switch.adjust_level(args.move_mode, args.rate, cmd==cmd_defs.move_with_on_off.id))
		elif cmd==cmd_defs.stop.id or cmd==cmd_defs.stop_with_on_off.id:
			asyncio.create_task(self.switch.stop_level(cmd==cmd_defs.stop_with_on_off.id))

class LightSwitch:
	"""Controls lights (or other devices with on/off and level controls) based
	on commands from control devices"""

	def __init__(self):
		self.onoff_clusters = []
		self.onoff_listener = OnOffListener(self)
[... 95 lines cut ...]
			print("Allowing devices to join")
			await app.permit(60)
			await asyncio.sleep(60)
			print("Join period ended")
		asyncio.create_task(permit_join())

		# Run the event loop forever by awaiting a future which will never get
		# a result
		await asyncio.get_running_loop().create_future()
	except asyncio.exceptions.CancelledError:
		# Pressing Ctrl-C will cause a CancelledError to be raised
		pass

	# If the app is not shutdown, the process will hang waiting for a background
	# thread to finish
	await app.shutdown()

# Get some basic information from zigpy printed to the console.  Changing the
# level to DEBUG can be useful to discover what kinds of events your control
# devices send but will produce a lot of spam.
logging.basicConfig(level=logging.INFO)

# Start the application
asyncio.run(main())
```

amosrohe, 2024-12-13:
[DataBeaver](https://github.com/DataBeaver) 

tnx for the code 
I manage to add devices and i can see response to using them (for example pushing a button) when i am in DEBUG  logging  
how do i use the response in the application level? is there any whey to run a function when event is set ?

robsmith11, 2025-05-08:
The example from @DataBeaver is the first I've seen that shows a clear and simple demonstration of how to use zigpy to do something useful.  Could it be added to the docs somewhere and linked in the main README?

Hedda, 2025-05-11:
Note that today you more likely want to use the zha library (which in turn imports zigpy) instead of using the zigpy library directly without the zha library, see:

- https://github.com/zigpy/zha

PS: Also be aware of the zigpy-cli tool as well which can be used in combination in an implementation of the zha library:

- https://github.com/zigpy/zigpy-cli