zigpy/zigpy #1088: Src_ep seems to be overwritten when sending a command

pipiche38, 2022-11-03:
When trying to send a request from src_ep 0x0b it looks like at a point time the src_ep is overwritten with 0x01

Is there anything I should do to avoid that ?

```
2022-11-03 22:28:37,985 DEBUG   :request device: <Device model=None manuf=None nwk=0xDB3C ieee=00:12:4b:00:04:26:ac:ad is_initialized=False> src_ep: 11 dst_ep: 11 data: b'\x11\x02\xe0\x00\x01\x02\x08\xff'
2022-11-03 22:28:37,988 DEBUG   :Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=11, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xDB3C), dst_ep=11, source_route=None, extended_timeout=False, tsn=32, profile_id=260, cluster_id=513, data=Serialized[b'\x11\x02\xe0\x00\x01\x02\x08\xff'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2022-11-03 22:28:37,992 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xDB3C), DstEndpoint=11, DstPanId=0x0000, SrcEndpoint=1, ClusterId=513, TSN=32, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x11\x02\xE0\x00\x01\x02\x08\xFF')
```

puddly, 2022-11-03:
zigpy-znp does not allow you to send data from unregistered endpoints. You must register endpoint 0x0b and then disable the endpoint 1 preference in zigpy-znp config: https://github.com/zigpy/zigpy-znp/pull/181

pipiche38, 2022-11-03:
Do you mean that we can have only One source Endpoint ?

In our initial integration we were able to use several source end point by using the add_endpoint

```
await self.add_endpoint(
                zdo_types.SimpleDescriptor(
                    endpoint=controller_ep,
                    profile=zigpy.profiles.zha.PROFILE_ID,
                    device_type=zigpy.profiles.zll.DeviceType.CONTROLLER,
                    device_version=0b0000,
                    input_clusters=[clusters.general.Basic.cluster_id,],
                    output_clusters=[],
                )
```

We are doing that because, we have some devices like Schneider using 0x0b or 0x03 as source point. Been restricted to only 1 end point will make that some Devices won't work any more

puddly, 2022-11-03:
See https://github.com/zigpy/zigpy-znp/blob/e6e2c42981c8a04303db6d28ef503a6b0778d667/zigpy_znp/zigbee/application.py#L751-L787. This is how zigpy-znp has always worked, though this logic will need to be updated to work better with the packet API.

pipiche38, 2022-11-03:
Do you mind, if I submit a PR against zigpy-znp to enhance _find_endpoint() ?
My id is to provide an optional parameter which is a candidate src_ep
If we find teh src_ep in the registered endpoint and the profileid and cluster are also registered then we return it otherwise we use the current default ?

zigbeefordomoticz, 2023-10-30:
This is still an open issue

pipiche38, 2024-04-27:
Still an open issue

pipiche38, 2024-10-24:
Still open issue.

closed 2025-04-29 by github-actions[bot]