zigpy/zigpy #1098: Ikea Tradfri repeater ota loop

Jsl-1, 2022-11-12:
Hi, 

I experience issues updating ikea repeaters. Logs let me think that upgrade is succesfull,  but repeater is never updated and restart updating indefinitely. 

I think it's exactly the same problem as reported in this issue :
https://github.com/dresden-elektronik/deconz-ota-plugin/issues/37

and resolved in :
https://github.com/manup/deconz-ota-plugin/commit/d9f096e9724ba0f6c50ecc92aab9afed9cf79f7b#diff-c00c2125ea3a8f90c7c4bde108c37012a7f74ce2f2c6fc3515bdbcbfc0566ae2

Can someone take a look and find a way to fix ikea file provider ?

Hardware used :  slaesh CC2652RB with zigpy-znp
Home-assistant : 2022.11.2
Can extract some logs, but ota logs don't show any error, all update appeared as success.

puddly, 2022-11-12:
> Can extract some logs, but ota logs don't show any error, all update appeared as success.

Logs would be useful. Can you post the debug log of the device completing the OTA update and rebooting?

Jsl-1, 2022-11-12:
Hi,

Which logs do i need to enable in configuration ?
- zigpy
- zigpy_znp
- zigpy.ota
- zigpy.zcl

Anything else ?

puddly, 2022-11-12:
That should be enough.

MattWestb, 2022-11-12:
@puddly In the IKEA OTA release we have swd flashing all IKEA controllers with help of your extracting and converting the images to GBL the many users was having the problem in ZHA and Z2M.

I have problems updating my repeater and was swd flashing one older firmware then the OTA was working to current version.
The device was original saying it have downloaded the OTA OK and applying the upgrade but was rolling it back ten booted.

If you is not stripping some meta data in the headers (like deCONZ was doing) i think this is one "undocumented future" (bug) in the firmware or the data setting in it that is making the problem.

Jsl-1, 2022-11-12:
Hi, 
Here is the log from first "progress: 100.0" to next "progress: 0.0"

https://paste.ubuntu.com/p/QJWFNMVRmQ/

puddly, 2022-11-13:
Have you tried power cycling the repeater? This isn't going to be possible to fix zigpy is fixed to perform OTA updates on demand, as opposed to in the background.

Jsl-1, 2022-11-13:
> Have you tried power cycling the repeater? This isn't going to be possible to fix until zigpy is fixed to perform OTA updates on demand, as opposed to in the background.

Hi, yes, i tried to unplug and replug just after success many times, restarting home assistant/computer after  success, querying file version,  firmware version, rebooting computer, restarting homeassistant... but nothing changed.

So i disabled ikea provider for now to stop updating and flooding network.

Just to know, is it a big deal to fix on demand update ?
Feel free to close this issue if nothing can be done before fixing it ;)

Thank you for your support !

closed 2023-05-19 by github-actions[bot]