zigpy/zigpy #1121: [zigpy.zcl] Unknown cluster command 225

mbo18, 2022-12-02:
Hi,

I've just got a Tuya plug (TS011F _TZ3000_gvn91tmx) and every night at 00:52 I have these logs in HA (10 same lines):

Today:
`2022-12-02 00:52:08.994 WARNING (MainThread) [zigpy.zcl] [0x05E9:1:0xe001] Unknown cluster command 225 b'\x07u\xfa\x1b+&\x00\x00\x00"\x00\x00\x00\x1d\x00\x00\x00\x19\x00\x00\x00\x14\x00\x00\x00\x0c\x00\x00\x00\x08\x00\x00\x00\x00'
`
Yesterday:
`2022-12-01 00:52:13.629 WARNING (MainThread) [zigpy.zcl] [0x05E9:1:0xe001] Unknown cluster command 225 b'\x06\xfd\xa8\x1a+"\x00\x00\x00\x1d\x00\x00\x00\x19\x00\x00\x00\x14\x00\x00\x00\x0c\x00\x00\x00\x08\x00\x00\x00\x00'
`
Any hint what this is? I'm just wondering.

Thank you

puddly, 2022-12-04:
If the Tuya plug doesn't have a quirk applied, ZHA won't know what it's proprietary messages mean. Tuya unfortunately decided to send data over the 0xE001 cluster, which falls outside of the manufacturer-specific range, and thus will cause a warning to be logged.

See if you can find your Tuya model (`_TZ3000_gvn91tmx`) in the quirks repo: https://github.com/zigpy/zha-device-handlers/. If not, you will may be able to add it to an existing quirk.

mbo18, 2022-12-04:
There is currently a quirk applied which is `zhaquirks.tuya.ts011f_plug.Plug`. It seems to be a "generic" quirk applied to all TS011F plugs. 
If I understand correctly, this quirk needs to be extended to cover this case?

gudvinr, 2023-02-05:
I have similar issue with another plug (TS011F by _TZ3000_zloso4jk). I have a bunch of them and they also send stuff at around 00:52:

<details><summary>logs</summary>

```
hass-homeassistant-1  | 2023-02-04 00:51:15.228 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:15.435 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:15.609 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:15.809 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:16.028 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:16.211 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:16.411 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:16.612 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:16.813 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:17.025 WARNING (MainThread) [zigpy.zcl] [0xD6B1:1:0xe001] Unknown cluster command 225 b'\ngZp+@\x03\x00\x00>\x03\x00\x00;\x03\x00\x009\x03\x00\x006\x03\x00\x004\x03\x00\x001\x03\x00\x00/\x03\x00\x00,\x03\x00\x00*\x03\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:38.322 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:38.503 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:38.704 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:38.905 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:39.111 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:39.306 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:39.508 WARNING (MainThread) [zigpy.zcl] [0x6017:1:0xe001] Unknown cluster command 225 b'\nnZp+\xfd\xc5\x00\x00\xe8\xc4\x00\x00\xb5\xc3\x00\x00\x98\xc2\x00\x00d\xc1\x00\x00L\xc0\x00\x002\xbf\x00\x00\xfc\xbd\x00\x00\xcb\xbc\x00\x00\xb9\xbb\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:39.514 WARNING (MainThread) [zigpy.zcl] [0xB7F1:1:0xe001] Unknown cluster command 225 b'\njZp+|\x16\x00\x00r\x16\x00\x00i\x16\x00\x00_\x16\x00\x00U\x16\x00\x00K\x16\x00\x00A\x16\x00\x007\x16\x00\x00+\x16\x00\x00!\x16\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:51:39.708 WARNING (MainThread) [zigpy.zcl] [0x6017:1:0xe001] Unknown cluster command 225 b'\nnZp+\xfd\xc5\x00\x00\xe8\xc4\x00\x00\xb5\xc3\x00\x00\x98\xc2\x00\x00d\xc1\x00\x00L\xc0\x00\x002\xbf\x00\x00\xfc\xbd\x00\x00\xcb\xbc\x00\x00\xb9\xbb\x00\x00\x01'
[... cut ...]
hass-homeassistant-1  | 2023-02-04 00:52:27.674 WARNING (MainThread) [zigpy.zcl] [0xF8DA:1:0xe001] Unknown cluster command 225 b'\n}Zp+\x7f^\x00\x00\xfd]\x00\x00Y]\x00\x00\xc3\\\x00\x00C\\\x00\x00\x84[\x00\x00\x11[\x00\x00eZ\x00\x00\xc9Y\x00\x00-Y\x00\x00\x01'
hass-homeassistant-1  | 2023-02-04 00:52:27.876 WARNING (MainThread) [zigpy.zcl] [0xF8DA:1:0xe001] Unknown cluster command 225 b'\n}Zp+\x7f^\x00\x00\xfd]\x00\x00Y]\x00\x00\xc3\\\x00\x00C\\\x00\x00\x84[\x00\x00\x11[\x00\x00eZ\x00\x00\xc9Y\x00\x00-Y\x00\x00\x01'
```
</details>

puddly, 2023-02-17:
You can fix it in one of two ways:

1. Set the level of the `zigpy.zcl` logger to `ERROR`.
2. Help document the cluster command by writing a device quirk.

fjolliton, 2023-03-04:
I have similar issues around the same time of the night (00:51:26 .. 00:52:12). What is so special about this time? Also I noticed that all my automations in Home Assistant were turned off at this exact moment.

Joo01, 2023-04-16:
I have also the warnings at same time!

Log looks:
```2023-04-16 00:04:37.526 WARNING (MainThread) [custom_components.samsungtv_smart.media_player] media_player.samsung_tv_gq32q50r - Connection to SmartThings restored
2023-04-16 00:51:14.354 WARNING (MainThread) [zigpy.zcl] [0x3742:1:0xe001] Unknown cluster command 225 b'\n\xed\xf4\xcd+w\x07\x00\x00P\x07\x00\x00)\x07\x00\x00\x02\x07\x00\x00\xda\x06\x00\x00\xb2\x06\x00\x00\x8d\x06\x00\x00f\x06\x00\x00;\x06\x00\x00\x12\x06\x00\x00\x01'
2023-04-16 00:51:14.553 WARNING (MainThread) [zigpy.zcl] [0x3742:1:0xe001] Unknown cluster command 225 b'\n\xed\xf4\xcd+w\x07\x00\x00P\x07\x00\x00)\x07\x00\x00\x02\x07\x00\x00\xda\x06\x00\x00\xb2\x06\x00\x00\x8d\x06\x00\x00f\x06\x00\x00;\x06\x00\x00\x12\x06\x00\x00\x01' ```

tonns of this warnings... at the end 30 minuts later the error messages '/store return code 500' ...

``` 2023-04-16 00:52:27.966 WARNING (MainThread) [zigpy.zcl] [0x60C0:1:0xe001] Unknown cluster command 225 b'\n\xfd\xf4\xcd+\x80\x07\x00\x00R\x07\x00\x00&\x07\x00\x00\xfa\x06\x00\x00\xd6\x06\x00\x00\xa8\x06\x00\x00\x84\x06\x00\x00_\x06\x00\x002\x06\x00\x00\x06\x06\x00\x00\x01'
2023-04-16 01:22:45.303 ERROR (MainThread) [homeassistant.components.hassio.handler] /store return code 500
2023-04-16 01:22:45.306 WARNING (MainThread) [homeassistant.components.hassio] Can't read Supervisor data: 
2023-04-16 01:24:22.489 ERROR (MainThread) [homeassistant.components.devolo_home_network] Error fetching switch_leds data: The device is currently not available. Maybe on standby?
2023-04-16 01:38:27.181 WARNING (MainThread) [homeassistant.helpers.entity] Update of climate.wohnzimmer is taking over 10 seconds ```

mbo18, 2023-04-16:
Someone needs to enable debug for ZHA/zlinky (see doc) at 00:50 and wait. Then put debug logs here

GuatrauHA, 2023-04-23:
I have exactly the same problem.

How can i 'Set the level of the zigpy.zcl logger to ERROR.' ?

guidocioni, 2023-04-24:
> I have exactly the same problem.
> 
> How can i 'Set the level of the zigpy.zcl logger to ERROR.' ?

Just add this to your `configuration.yaml`

```
logger:
  default: warning
  logs:
    zigpy.zcl: error
```

closed 2023-10-28 by github-actions[bot]