zigpy/zigpy #1125: Bool type can store more values than needed

Shulyaka, 2022-12-06:
Most notably, t.Bool(1) != t.Bool(2). It is a different behavior than python bool type. It may create unexpected results.
Unfortunately I don't have enough python knowledge to fix that myself, a hint would be helpful. I tried to overload Bool.__new__() function without success.

Proposed tests:
```python
def test_bool():
    assert t.Bool(0) == t.Bool.false
    assert t.Bool("0") == False
    with pytest.raises(ValueError):
        t.Bool(-1)
    assert t.Bool.true == True
    assert t.Bool(2) == t.Bool.true
    assert t.Bool("0xFF") == t.Bool(1)
    assert t.Bool(0xFF).serialize() == b"\x01"
```

puddly, 2022-12-06:
I believe some devices send non-binary numbers over boolean data types like this.

Is there a specific device being affected by this behavior?

puddly, 2022-12-06:
Also, we currently don't handle the "unknown" or "invalid" value for many types: `0xff` for `uint8_t`, for example. They're treated as just a normal value for the type. I believe `0xff` is similarly used for `Bool`.

Shulyaka, 2022-12-06:
What I mean is that we should be able to parse anything into Bool (or at least anything accepted by uint8_t), but cast it to 1 (unless it is 0). At least this is my personal expectation and I might be outnumbered, I have no idea:)

closed 2023-06-11 by github-actions[bot]