zigpy/zigpy #1131: Failed to deserialize ZDO packet / Data is too short to contain 1 bytes

nocin, 2022-12-25:
Hi all,

I'm trying to fix my Zigbee network, as I have some lights turning on randomly or become unavailable. Therefore, I enabled the following logs:
```logger:
  default: info
  logs:
    # ZHA
    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: debug
    zigpy: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug
```
When starting Home Assistant, I noticed the following lines in the log:
[image]

From top to bottom:
```

Logger: zigpy.device
Source: runner.py:120
First occurred: 10:45:05 AM (1 occurrences)
Last logged: 10:45:05 AM

Failed to parse message (b'038403dc') on cluster 32799, because Data is too short to contain 1 bytes
```

```
Logger: zigpy_znp.zigbee.application
Source: /usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py:489
First occurred: 10:45:05 AM (1 occurrences)
Last logged: 10:45:05 AM

Failed to deserialize ZDO packet
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 485, in on_zdo_message
    zdo_hdr, zdo_args = self._device.zdo.deserialize(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zdo/__init__.py", line 44, in deserialize
    args, data = t.deserialize(data, param_types)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 716, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 742, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 184, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```

```
Logger: zigpy_znp.api
Source: runner.py:120
First occurred: 10:45:05 AM (1 occurrences)
Last logged: 10:45:05 AM

Failed to parse broken GeneralFrame(header=CommandHeader(id=0x9F, subsystem=Subsystem.ZDO, type=CommandType.AREQ), data=b'\x03\xD7\x84\x03\xDC') as <class 'zigpy_znp.types.commands.ZDO.ParentAnnceRsp.Callback'>
```

Seems like all three are related to parsing a **ZDO** packet. 
Can anyone tell what this is for and if it is something relevant? If yes, how can I fix it? 

Thanks & Regards
Nico

puddly, 2022-12-25:
You just have some router on your network broadcasting bogus announcement data. Nothing you can do to fix this other than figuring out which router it is and either rebooting it or removing it from your  network. From the corrupted data present, it seems the NWK of the router might be 0xD703.

nocin, 2022-12-26:
Hi @puddly,

thanks to you, I could identify the device. It's a **Classic A60 W clear - LIGHTIFY by OSRAM** bulb.
[image]

I'm curious, how you got the NWK 0xD703 from the posted logs? 
Does this corrupted announcement data leads to any further problems in the network? Or is it just a problem with the device itself?

The strange thing is, that this light is one of the few lights that I have never had any problems with. But when checking the network graph, it seems like many devices go through it. So it would make sense, that it leads to issues on the network.
[image]

I will try to reconnect it to my network and if the error still occurs, I will replace it in the next days. I will report if it has led to an improvement.

BR
Nico

MattWestb, 2022-12-26:
Was looking and your device is running the last posted firmware from https://update.ledvance.com/firmware-overview?submit=all (put `A60 W` in the search criteraia). 
I think its one first gen OSRAM device and the Zigbee chip / Zigbee stack is very buggy. For example is the nice outdoor plug impossible having the the network without corrupting the traffic and degrading the mesh network and the best is putting them in the black box for bad Zigbee devices.

tube0013, 2022-12-26:
Yup Osram devices from that era are a nightmare and frustration waiting to happen.

nocin, 2023-01-03:
Hi all,

after removing the OSRAM light for a few days, there still appears some ZDO errors in the log. Not sure to what device they are related:
[image]

```Logger: zigpy.zcl
Source: runner.py:120
First occurred: 12:52:19 AM (10 occurrences)
Last logged: 12:52:21 AM

[0xE52F:1:0xe001] Unknown cluster command 225 b"\n\x7f*F+:\x00\x00\x008\x00\x00\x006\x00\x00\x004\x00\x00\x000\x00\x00\x00.\x00\x00\x00,\x00\x00\x00*\x00\x00\x00'\x00\x00\x00%\x00\x00\x00\x01"
```

```Logger: zigpy_znp.uart
Source: /usr/local/lib/python3.10/site-packages/zigpy_znp/uart.py:74
First occurred: January 2, 2023 at 11:45:51 PM (1 occurrences)
Last logged: January 2, 2023 at 11:45:51 PM

Received an exception while passing frame to API: TransportFrame(payload=GeneralFrame(header=CommandHeader(id=0xB2, subsystem=Subsystem.ZDO, type=CommandType.AREQ), data=b'\x0B\x42\x84\x00\x32'))
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/uart.py", line 72, in data_received
    self._api.frame_received(frame.payload)
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 810, in frame_received
    command = command_cls.from_frame(frame, align=self.nvram.align_structs)
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/types/commands.py", line 417, in from_frame
    params[param.name], data = param.type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 309, in deserialize
    value, data = field.type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 713, in deserialize
    length, data = cls._length_type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 184, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```
Unfortunately, there are still some lights which become randomly unavailable for a few hours and one light randomly turns on 1 or 2 times a day. Seems like removing the OSRAM light didn't bring any improvement to my zigbee network.

There is also a timeout error in the log, which I have no idea where it is coming from.
[image]
```
Logger: homeassistant.components.zha.core.channels.base
Source: components/zha/core/channels/base.py:486
Integration: Zigbee Home Automation (documentation, issues)
First occurred: January 2, 2023 at 7:44:55 PM (12 occurrences)
Last logged: January 2, 2023 at 7:45:03 PM

[0x38A9:1:0x0702]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
[0x82B4:1:0x0702]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts:<Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0x82B4:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0x82B4:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'), DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')]
[0x38A9:1:0x0b04]: async_initialize: all attempts have failed: [TimeoutError(), TimeoutError(), TimeoutError(), TimeoutError()]
```

Can anyone give me a hint what I have to look for?
Thanks & Regards
Nico

MattWestb, 2023-01-03:
Wot device is `0xE52F` and `0x38A9` ?

nocin, 2023-01-03:
Hi @MattWestb,

I guess 0xE52F and 0x38A9 are the NwK? 
Can you give me a tip on the fastest way to find the device for the given ID's? So far I have always clicked through all my devices, but that takes forever.

Thanks!

MattWestb, 2023-01-03:
They is not famous as Osram plugs and likely working well but you was getting time out from it but it can being one before that is making the problems.
Also device `0x82B4` is having no ACK and i think its have the same problem only not knowing witch is doing the problems.

Kan you finding `0x82B4` 2 ??

nocin, 2023-01-03:
Hi @MattWestb,

thanks for your insane response time. :-)

I got both Blitzwolf Plugs in a single order (it was a package of 4 plugs). So I'm a bit confused why they have different chips in it. I would expect all 4 were coming from the same charge.

Found `0x82B4`. It's a [GERMA Relay](https://www.aliexpress.com/item/1005004099871471.html?spm=a2g0o.order_list.order_list_main.5.79391802R1Xj76) I got in December. I set it up with Home Assistant, but until now, I'm not using it. That's why it's last seen on 2022-12-16.
[image]
Since my problems have been going on for several months, I don't think they are related to this GERMA relay.

puddly, 2023-01-03:
> Can you give me a tip on the fastest way to find the device for the given ID's? So far I have always clicked through all my devices, but that takes forever.

You can use the Visualization in ZHA and type in `0x1234` to search for that specific device. It's unfortunately kind of hard to tell which devices are matching the query but they will light up a little bit. You can then double click on the node and it will take you to the device.

> [image]

The first warning is likely not a problem. The second one is being sent by device `0x420B` (the first two bytes of most of those ZDO commands are the NWK address encoded, so `\x0B\x42` in your case).

MattWestb, 2023-01-03:
The GERMA is also one Telink Semiconductor :-((
Can you looking if you finding it on the network map ?  
(I think you can doing it from the device card)
Very likely its only floating around then ts have no parent for the moment.
Repower it or resetting it or make it connecting to the network.

Also looking if you is having low link quality from to one device with all its neighbors.
I also having one TS0004 that is having one Tel... and its having very bad signal to all neighbors but its not having any children and all is being routed around it in my production network.

puddly, 2023-01-03:
> The GERMA is also one Telink Semiconductor :-((

Interesting. Are there any major stack bugs with Telink's Zigbee SDK?  ThirdReality uses them and I've not had any issues (so far) with their devices on my network, though I have mostly Aqara and IKEA.

MattWestb, 2023-01-03:
My TS0004 is working great but is have only 10 dB output power and bad antenna.
Many new tuya plugs is using it and the power chip is using frequencies for sending data and i think its bad code or bad hardware making them reporting over voltage and zero voltage without its real.

I think more bad signal then bugs in this devices then is more normal ones.

nocin, 2023-01-04:
> You can use the Visualization in ZHA and type in `0x1234` to search for that specific device. It's unfortunately kind of hard to tell which devices are matching the query but they will light up a [...]

Hi @puddly,

thanks for the hint with the visualization! I would not have thought of that.

I looked for the `0x420B` device and again, it's an OSRAM light bulb. I totally forgot that I had bought two of these. :-(
So both OSRAM light bulbs were producing a ZDO error. Makes sense in some way.
[image]

Looks like I will have to replace this bulb as well. Will report, if it leads to any improvements.

Thank you all so far!
BR
Nico

MattWestb, 2023-01-04:
Your Osram A60 W looks being on the last firmware then looking on the firmware list at https://update.ledvance.com/firmware-overview?submit=all.
Was having one user with the same device for some weeks ago but i dont knowing if hi have replacing it or not for getting the network working OK.
Look on the network map and see if its between your coordinator and the other device that is having problems then its likely its not doing the routing OK and then the other devices is not getting the commands sent to them.

pipiche38, 2023-01-07:
I have a similar issue and here are some inputs data as I was able to capture the ZDO frame

Coordinator is a SonOff TI 
The Device  0x8347 is a Dimmer switch wo neutral  from Legrand

```
2023-01-07 19:16:49,739 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32769, SecurityUse=0, TSN=11, MacDst=0x0000, Data=b'\x81\x83\x47')
2023-01-07 19:16:49,745 WARNING :Failed to deserialize ZDO packet
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 485, in on_zdo_message
    zdo_hdr, zdo_args = self._device.zdo.deserialize(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/zdo/__init__.py", line 44, in deserialize
    args, data = t.deserialize(data, param_types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 734, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 183, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```

[image]

puddly, 2023-01-07:
Nothing can be done about this. The packet is corrupted (or seemingly the wrong one? it shares no bytes with the one in your screenshot).

pipiche38, 2023-01-07:
I'm surprise as I can make it systematic.

pipiche38, 2023-01-07:
I have the feeling this is happening on ZDOCmd.IEEE_addr_req when done to broadcats, and it looks like the coordinator is sending back something uncorrect

```
2023-01-07 19:38:46,914 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.IEEE_addr_req: [0xAD1D, <AddrRequestType.Single: 0>, 0]
2023-01-07 19:38:46,918 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.Broadcast: 15>, address=0xFFFD), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=1, TSN=15, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x0F\x1D\xAD\x00\x00')
2023-01-07 19:38:46,943 DEBUG   :Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=15, profile_id=0, cluster_id=32769, data=Serialized[b'\x0f\x81\xad\x1d'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-01-07 19:38:46,950 DEBUG   :Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=15, profile_id=0, cluster_id=32769, data=Serialized[b'\x0f\x81\xad\x1d'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-01-07 19:38:46,954 DEBUG   :Could not parse ZDO message from packet
2023-01-07 19:38:46,959 ERROR   :Failed to parse message (b'0f81ad1d') on cluster 32769, because Data is too short to contain 1 bytes
2023-01-07 19:38:46,967 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-01-07 19:38:46,972 DEBUG   :Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x0000), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=32769, TSN=15, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x0F\x81\xAD\x1D')
2023-01-07 19:38:46,990 DEBUG   :Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-01-07 19:38:46,993 DEBUG   :Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32769, SecurityUse=0, TSN=15, MacDst=0x0000, Data=b'\x81\xAD\x1D')
2023-01-07 19:38:46,998 WARNING :Failed to deserialize ZDO packet
Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy_znp/zigbee/application.py", line 485, in on_zdo_message
    zdo_hdr, zdo_args = self._device.zdo.deserialize(
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/zdo/__init__.py", line 44, in deserialize
    args, data = t.deserialize(data, param_types)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/__init__.py", line 9, in deserialize
    value, data = type_.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 734, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/types/basic.py", line 183, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
2023-01-07 19:38:47,002 DEBUG   :Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=15, profile_id=0, cluster_id=32769, data=Serialized[b'\x0f\x81\xad\x1d'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-01-07 19:38:47,006 DEBUG   :Could not parse ZDO message from packet
2023-01-07 19:38:47,009 ERROR   :Failed to parse message (b'0f81ad1d') on cluster 32769, because Data is too short to contain 1 bytes
```

closed 2023-09-03 by github-actions[bot]

pipiche38, 2024-12-07:
I have added a debug in the zigpy_znp application on_zdo_message to capture the packet value when having this deserialize issue

@puddly When we look to the data, the hex decoding give 33818523, 

where 0x33 I guess is the SQN
where 0x81 is an error "The requested device did not exist on a device following a child descriptor request to a parent."

where 0x8523 is indeeed a NwkId in my network, and it is an  Aqara (lumi.sensor_motion.aq2) for which we have done a IEEE_Addr_req

So my question is why https://github.com/zigpy/zigpy-znp/blob/b75c65657c6c7bf1d0b73a6d8eda787195746eaf/zigpy_znp/zigbee/application.py#L473 are we not checking the status before doing the deserialize as indeed the data are not correct if status is != 0

If I look to clusterId 32769 this is 0x8001, which If I'm not wrong is IEEE address response

`2024-12-07 17:26:21,634 WARNING :Failed to deserialize ZDO packet ZigbeePacket(timestamp=datetime.datetime(2024, 12, 7, 16, 26, 21, 633380, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=51, profile_id=0, cluster_id=32769, data=Serialized[b'3\x81\x85#'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)`
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 473, in on_zdo_message
    zdo_hdr, zdo_args = self._device.zdo.deserialize(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zdo/__init__.py", line 46, in deserialize
    args, data = t.deserialize(data, param_types)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/__init__.py", line 11, in deserialize
    value, data = type_.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 924, in deserialize
    item, data = cls._item_type.deserialize(data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 199, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```