zigpy/zigpy #1146: Connecting to Ethernet Coordinator

majurgens, 2023-01-23:
I'm not sure if I am understanding zigpy properly, but can I use it to write a python script to talk to the ethernet interface (with an IP address obviously) of the Zigbee coordinator like this one https://paulbanks.org/projects/lidl-zigbee/?

puddly, 2023-01-23:
You connect to `socket://1.2.3.4:5678` instead of `/dev/serial/by-id/device-id`, it's handled automatically.

Be aware that Zigpy is a low-level Zigbee library so it won't be directly useful in a script unless you're familiar with Zigbee primitives and the ZCL.

closed 2023-01-23 by puddly

majurgens, 2023-01-23:
Thanks. Is there something else around that implements at a higher level. It would be nice for me to be able to do something high level like "get device xyz status"