zigpy/zigpy #1147: Proposal: Rate Limits

jzielke84, 2023-01-26:
I have noticed that when triggering many lights in a scene using home assistant sometimes a light randomly misses the on or off command and gets “stuck”. Zigpy correctly reports the light as still being on because it hasn’t gotten any acknowledgment from it. Triggering the scene again results in the hung lang switching as it should.

My assumption is, that on a large number or parallel changes the Zigbee network can’t handle it, resulting in some kind of package loss.

My proposal is to add some rate limits so when any app like Homeassistant sends x numbers of commands to zigpy, a certain number of commands are send out directly while others get into some queue for a couple of milliseconds, thus allowing the network to digest all messages without getting flooded.

Zigbee Groups won’t serve that purpose because if on light in a group is on, the whole Zigbee group returns on while it’s probably not (someone turned a light of manually by another Zigbee switch).

puddly, 2023-01-26:
The current concurrency limits for each radio type are ones that I've found to be stable. What radio are you using?

jzielke84, 2023-01-29:
What do you mean by radio? Coordinator hardware? Zigbee Channel?

puddly, 2023-01-29:
Yes, coordinator.

jzielke84, 2023-01-30:
A Sonoff USB 3.0 coordinator on a 5m extension cable. Controlling small groups works fine but when triggering 18 lights at once some get stuck.

puddly, 2023-01-30:
Which one? There are two Sonoff coordinators, each using different chips.

jzielke84, 2023-01-30:
Angebot: SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N) Zigbee USB-Stick für ZHA in Home Assistant or Zigbee2MQTT,Open HAB etc. https://amzn.eu/d/1qGEaIV

puddly, 2023-01-30:
18 concurrent requests is a lot. Reduce the concurrency a little if it's too much for your environment (the default for your radio type is 16):

```python
zha:
  zigpy_config:
    max_concurrent_requests: 8
```

jzielke84, 2023-01-30:
Thanks I will give it a try and report back to you.

jzielke84, 2023-01-31:
Set it to 8. still one random light hangs. Even at 4.

MattWestb, 2023-01-31:
@jzielke84 Do you getting some interesting the HA log like router problems ?
Also if you can sniffing the Zigbee traffic it wold being great then its sounds you is having routing problem in your Zigbee network.
If having 4 and sending to 18 devices its taking 4.5 seconds sending one command to all 18 device that must looking like one Skye with stars flickering.

It shall being possible sending very high rate uni cast in one Zigbe network but its need good routers and if one is bad its destroying all and its not working well (Philips HUE is doing it). But for light is Zigbee groups one must if sending commands to many device or the network is being jammed or routing requests between commands.

I can thinking remaking Zigbee light groups in ZHA with 3 states instead of 2,
All off and can toggle all on, all on can toggle all off, mixed on /off can toggle all on **or** off.
What is you thinking @puddly ??

jzielke84, 2023-01-31:
> Do you getting some interesting the HA log like router problems ?

None that I'm aware of. Everything works flawlessly except mass on/off switching.

puddly, 2023-01-31:
I'm able to toggle all 40 of my lights concurrently with zero failures (albeit with a few seconds delay) so I suspect you either have an interference issue, or a bad router on your network.

jzielke84, 2023-01-31:
Well, how we go from here?

Hedda, 2023-02-03:
First tip is to follow this guide for Zigbee interference avoidance (as well as improve network range and coverage optimization):

https://community.home-assistant.io/t/guide-for-zigbee-interference-avoidance-and-network-range-coverage-optimization/515752

By the way, regarding your mentioned "5m extension cable"; suggest making sure that is a properly shielded and grounded cable.

Also, among other things that also include the recommendation to upgrade the Zigbee Coordinator firmware. For TI CC2652P see:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Adapters based on Texas Instruments CC2652 and CC1352 chips can be flashed by putting them in the bootloader (see your adapter manual on how to do this). After you have done this one of the following tools can be used to flash firmware to it:

- **[ZigStar GW Multi tool ](https://github.com/xyzroe/ZigStarGW-MT)(multi platform GUI tool)**
- **[CC2538-BSL](https://github.com/JelmerT/cc2538-bsl)** (multi platform Python based command line tool) ([instructions](https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_via_cc2538-bsl.html))
- **[Texas Instruments FLASH PROGRAMMER 2](https://www.ti.com/tool/FLASH-PROGRAMMER)** (Windows only) 

If your OS unable to find the device chances are that your OS can't communicate with its virtual COM port because lacking drivers:

- https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

If you still have problems after that then you need to provide debug logs and additional information mentioned in the ZHA docs:

- https://www.home-assistant.io/integrations/zha#debug-logging

- https://www.home-assistant.io/integrations/zha#reporting-issues

closed 2023-09-02 by github-actions[bot]