zigpy/zigpy #1157: Failed to convert attribute 0xD030 from ExternalSwitchType.Momentary (<class 'str'>) to type <enum 'ExternalSwitchType'>: invalid literal for int() with base 10: 'ExternalSwitchType.Momentary'

wonsky007, 2023-02-03:
Hi,

Im using TS0002 relay [Quirk: zhaquirks.tuya.ts000x.Switch_2G_GPP] and want change ExternalSwitchType from Toggle to Momentary. Its failing every time because conversion error, even when trying to write value readed from device.
HA version:  2023.1.7 and current ZHA

Debug logs below:

```
2023-02-03 12:10:22.246 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=zha, service=set_zigbee_cluster_attribute, service_data=ieee=a4:c1:38:de:a0:ab:18:d8, endpoint_id=1, cluster_id=57345, cluster_type=in, attribute=53296, value=ExternalSwitchType.Momentary>

2023-02-03 12:10:22.248 ERROR (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Failed to convert attribute 0xD030 from ExternalSwitchType.Momentary (<class 'str'>) to type <enum 'ExternalSwitchType'>: invalid literal for int() with base 10: 'ExternalSwitchType.Momentary'

2023-02-03 12:10:22.270 DEBUG (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True, *is_reply=False), tsn=51, command_id=<GeneralCommand.Write_Attributes: 2>, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2023-02-03 12:10:22.271 DEBUG (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Sending request: Write_Attributes(attributes=[])

2023-02-03 12:10:22.272 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBD36), dst_ep=1, source_route=None, extended_timeout=False, tsn=51, profile_id=260, cluster_id=57345, data=Serialized[b'\x003\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2023-02-03 12:10:22.278 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0xBD36), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=57345, TSN=51, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x00\x33\x02')

2023-02-03 12:10:22.301 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)

2023-02-03 12:10:22.304 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=51)

2023-02-03 12:10:22.325 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=57345, SrcAddr=0xBD36, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=94, SecurityUse=<Bool.false: 0>, TimeStamp=5522199, TSN=0, Data=b'\x18\x33\x04\x00', MacSrcAddr=0xBD36, MsgResultRadius=29)

2023-02-03 12:10:22.328 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xBD36), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=57345, data=Serialized[b'\x183\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=94, rssi=None)

2023-02-03 12:10:22.329 DEBUG (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Received ZCL frame: b'\x183\x04\x00'

2023-02-03 12:10:22.330 DEBUG (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=51, command_id=4, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)

2023-02-03 12:10:22.332 DEBUG (MainThread) [zigpy.zcl] [0xBD36:1:0xe001] Decoded ZCL frame: TuyaZBExternalSwitchTypeCluster:Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)])

2023-02-03 12:10:22.333 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xBD36](TS0002): set: ExternalSwitchType.Momentary for attr: 53296 to cluster: 57345 for ept: 1 - res: Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)])

2023-02-03 12:10:22.334 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [57345] cluster_type: [in] endpoint_id: [1] attribute: [53296] value: [ExternalSwitchType.Momentary] manufacturer: [None] response: [Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)])]
```

[image]

[image]

MattWestb, 2023-02-03:
Then sending you must using dec or hex so `2` or `0x02` shall working.
Try first reading the attribute for getting the current string so you is knowing how it was from the start.

wonsky007, 2023-02-03:
I've tried set value to `2` or `0x02` yesterday without luck.

This is what I read from device:
[image]

In next step I've change value to 2
[image]

Readed again and attribut was still ExternalSwitchType.Toggle. 
Solved by restarting home assistant and write value 2. It looks like ExternalSwitchType.Momentary was cached somewhere by HA or web browser. Sometimes restart is best solution.

puddly, 2023-05-26:
Fixed by #1214.

closed 2023-05-26 by puddly