zigpy/zigpy #1163: [REQUEST] Leviton OTA provider for Leviton branded Zigbee devices

Hedda, 2023-02-21:
Missed that [Leviton](https://www.leviton.com/en) was the first to join the "**_Works with Home Assistant_**" partner program (Nabu Casa device compatibility partnership):

https://www.home-assistant.io/blog/2022/07/27/leviton-partner/

Or rather, I missed that [Leviton](https://www.leviton.com/en) then solely partnered to the "**_Works via Z-Wave with Home Assistant_**", only agreeing to work with Nabu Casa to make sure that their line of Z-Wave series devices will work great with Home Assistant, and not also partnered for the "**_Works via Zigbee with Home Assistant_**" to also make sure that those and future [Leviton](https://leviton.com/) branded Zigbee devices will also fully work out-of-the-box as well as get automatic firmware updates in via Home Assistant, such as Leviton’s Decora Smart family of Zigbee enabled lighting control products.

* https://www.leviton.com/en/all-products-new/automation/by-technology/decora-digital-controls-with-zigbee-technology#t=Products&sort=%40wcs_site_tree_rank%20ascending&layout=card
  * https://www.leviton.com/en/products/dg6hd-1bw
  * https://www.leviton.com/en/products/dg15s-1bw
  * https://www.leviton.com/en/products/dg3hl-1bw
  * https://www.leviton.com/en/products/dg15a-1bw

"_**There are manufacturers that are creating products that integrate into Home Assistant using standards like Z-Wave, Zigbee, or Matter (soon). In these cases, the integration is maintained by the Home Assistant community and Nabu Casa. These companies can still become a member of the Works with Home Assistant program but are relieved from integration maintenance.**_" "_**With Home Assistant we are always working on educating our users about preferring local control and open standards when acquiring new products. This is also reflected in the “Works with Home Assistant” badges.**_"

https://www.home-assistant.io/blog/2022/07/12/partner-program/

[image]

Thus this open letter question [Leviton](https://www.leviton.com/en) as a company also will provide public Zigbee OTA firmware update images for those devices?

https://www.leviton.com/en

The Follow-up question to that or rather an open letter feature request is if [Leviton](https://www.leviton.com/en) as a company is willing to spare development time to add a publicly available OTA (Over-The-Air) Provider Server or Repository where it publishes/hosts Zigbee firmware update image files for its devices as well as contribute some OTA Provider downloader code for this [zigpy library](https://github.com/zigpy/zigpy)?

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

As [Leviton](https://www.leviton.com/en) as a company probably do not know yet, zigpy is the Zigbee framework and hardware radio API/CLI abstraction code library that is used by Home Assistant's ZHA integration as well as a few other popular home automation software as a dependency in their Zigbee implementations and providing a "Leviton Zigbee OTA provider" and matching downloader code support for Zigbee devices from [Leviton](https://www.leviton.com/en) in this zigpy library could potentially enable much easier OTA firmware upgrades for end-users of those in the future, and include; [for Home Assistant's ZHA integration](https://www.home-assistant.io/integrations/zha) today, as well as probably [Zigbee plugin for Domoticz](https://www.domoticz.com/wiki/ZigbeeForDomoticz) (e.i. the [Domoticz-Zigbee](https://github.com/zigbeefordomoticz/Domoticz-Zigbee) project) and the [official Zigbee Plugin developed by Jeedom SAS](https://market.jeedom.com/index.php?v=d&p=market_display&id=4050) (i.e. the [Zigbee plugin for Jeedom](https://doc.jeedom.com/en_US/plugins/automation%20protocol/zigbee/)) in the future.

If Leviton already provides a  publicly available OTA (Over-The-Air) Provider Server or Repository where it publishes/hosts Zigbee firmware update image files then all that should be missing is an OTA provider code in zigpy.

Please see zigpy's existing OTA provider code ([provider.py](https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py)) in Python which includes the URL links as source and downloader code for Zigbee OTA firmware image files for direct download:

https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py

https://github.com/zigpy/zigpy/tree/dev/zigpy/ota

That is, this zigpy library would need a new "Leviton OTA Firmware provider" section in that OTA provider code ([provider.py](https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py)) to handle image download for Leviton OTA firmware update image files, (please see the "Skeleton OTA Firmware provider" for reference, as well as for referencing also studying [provider.py](https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/provider.py) to also look at the other OTA providers that already have different code sections to handle the download of OTA files, such as; IKEA, INOVELLI, ThirdReality (3reality), SONOFF/ITEAD, SALUS/Computime, and LEDVANCE/OSRAM.

https://github.com/zigpy/zigpy/blob/dev/zigpy/ota/OTA_URLs.md

Hopefully, Leviton management will consider setting up an OTA provider server or creating a dedicated repository for all officially released Zigbee OTA files for Leviton branded Zigbee devices.  

That way it will be a single trusted and official primary source of official Zigbee OTA files released by Leviton, and that OTA provider source repository with your Zigbee OTA files could not only be used by zigpy as it could also be reused by other Zigbee implementations independently (both other open-source implementations as well as closed source and proprietary implementations).

FYI, zigpy developers has recently added a new feature to the "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool that manufacturers can use to produce a skeleton JSON index from a given set of OTA files, and that way make sure manufacturers OTA files are correctly-structured so that they will be compatible with zigpy out-of-the-box. See the README.md for that "[zigpy-cli](https://github.com/zigpy/zigpy-cli)" tool:

https://github.com/zigpy/zigpy-cli/blob/dev/README.md#ota

Maybe also consider copying concept of `index.json` with metadata on all files like Koenkk repository has to make scripting easier:

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

zigpy and ZHA does not yet use this or other OTA Index files but [Zigbee2MQTT](https://www.zigbee2mqtt.io/) allows users to use it locally as described here:

https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/guide/usage/ota_updates.md#local-ota-index-and-firmware-files

and discussion in relation to zigpy implementation here:

https://github.com/zigpy/zigpy/issues/535

And of course `README.md` file which among other things could contain some basic information on how to update those:

https://github.com/Koenkk/zigbee-OTA/blob/master/README.md

Perhaps also as keeping some kind of `CHANGELOG.md` with changelog release history for each device firmware version:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

PS: By the way, please also see this somewhat related discussion regarding here Koenkk zigbee-OTA repository -> https://github.com/zigpy/zigpy/issues/535

puddly, 2023-08-16:
@Hedda please don't copy/paste comments like this, it generates a lot of unnecessary noise. Most people subscribed to one issue will be subscribed to the others.

Hedda, 2023-08-16:
OK, though I think developers from Leviton, dresden elektronik, Ubisys, and LiXee would only follow threads about their products.

closed 2026-04-24 by Hedda