zigpy/zigpy #1208: ValueError: invalid literal for int() with base 10: 'StartUpOnOff.Off'

rrooggiieerr, 2023-05-09:
I'm seeing this zigpee relater error in my Home Assistant log file

```
2023-05-09 22:35:22.268 ERROR (MainThread) [homeassistant.util.logging] Exception in async_state_changed when dispatching '_TZE200_la2c2uo9 TS0601_84:ba:20:ff:fe:67:35:b1_available_entity': ()
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/entity.py", line 96, in async_state_changed
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 147, in state
    if self.current_option is None or self.current_option not in self.options:
  File "/usr/src/homeassistant/homeassistant/components/zha/select.py", line 208, in current_option
    option = self._enum(option)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 366, in __call__
    value = int(value)
ValueError: invalid literal for int() with base 10: 'StartUpOnOff.Off'
```

It's related to a device identified by `TS0601 by _TZE200_la2c2uo9`

This is the field it relates to. If I change the vallue to `StartUpOnOff.On` I get a similar message in my log file but then `ValueError: invalid literal for int() with base 10: 'StartUpOnOff.On'`

[image]

dmulcahey, 2023-05-09:
I have a branch of HA ZHA / frontend that makes this work similar to how commands work. It will correct this issue. use the numerical equivalent for now.

rrooggiieerr, 2023-05-09:
Where 0 is Off and 1 is On I'm guessing?

dmulcahey, 2023-05-09:
[image]

puddly, 2023-05-26:
Fixed by #1214.