zigpy/zigpy #1210: group membership error

stu247, 2023-05-15:
I am using Home Assistant 2023.5.1 with ZHA.  I started using ZHA about 2 months ago.  Since I started using it I have been seeing the following error in HA log:
```
2023-05-07 08:23:42.494 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 135, in group_membership_scan
    await ep.group_membership_scan()
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 174, in group_membership_scan
    groups = set(res[1])
TypeError: 'Status' object is not iterable
```
It varies how often the error message is generated, but it occurs around 10 times a day.  I have not set up any groups and I do not have any groups listed in CONFIGURATION->Groups.  Please let me know if you need any additional information to resolve this issue.

stu247, 2023-05-17:
Here is some log output surrounding the error:
```
2023-05-17 13:41:32.387 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD0A5](E1C-NB7): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:41:43.381 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6597](WM25/L-Z): Device seen - marking the device available and resetting counter
2023-05-17 13:41:43.381 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x6597](WM25/L-Z): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:41:43.392 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB1D7](WM25/L-Z): Device seen - marking the device available and resetting counter
2023-05-17 13:41:43.392 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xB1D7](WM25/L-Z): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:41:45.674 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2023-05-17 13:41:45.679 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.APP_CNF|GP|UTIL|ZDO|AF|SYS: 1625>)
2023-05-17 13:41:52.388 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x502B](WM25/L-Z): Device seen - marking the device available and resetting counter
2023-05-17 13:41:52.388 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x502B](WM25/L-Z): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:41:52.388 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x3313](WM25/L-Z): Device seen - marking the device available and resetting counter
2023-05-17 13:41:52.388 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x3313](WM25/L-Z): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:41:55.859 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0xF254, Relays=[])
2023-05-17 13:41:55.859 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x9432, SrcIEEE=0c:43:14:ff:fe:8a:c9:12, ParentNwk=0xF254)
2023-05-17 13:41:55.859 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x9432, SrcIEEE=0c:43:14:ff:fe:8a:c9:12, ParentNwk=0xF254)
2023-05-17 13:41:55.859 INFO (MainThread) [zigpy.application] Device 0x9432 (0c:43:14:ff:fe:8a:c9:12) joined the network
2023-05-17 13:41:55.859 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 135, in group_membership_scan
    await ep.group_membership_scan()
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 174, in group_membership_scan
    groups = set(res[1])
TypeError: 'Status' object is not iterable
2023-05-17 13:41:55.866 DEBUG (MainThread) [zigpy.zcl] [0x9432:1:0x0004] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0
, *is_cluster=True, *is_general=False), tsn=21, command_id=2, *direction=<Direction.Server_to_Client: 0>)
2023-05-17 13:41:55.867 DEBUG (MainThread) [zigpy.zcl] [0x9432:1:0x0004] Sending request: get_membership(groups=[])
2023-05-17 13:41:55.867 DEBUG (MainThread) [zigpy.device] [0x9432] Extending timeout for 0x15 request
2023-05-17 13:41:55.867 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9432), dst_ep=1, source_route=None, extended_timeout=
True, tsn=21, profile_id=260, cluster_id=4, data=Serialized[b'\x01\x15\x02\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-05-17 13:41:55.867 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x9432, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2023-05-17 13:41:55.873 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2023-05-17 13:41:55.873 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x9432), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=4, TSN=21, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NET
WORK: 32>, Radius=0, Data=b'\x01\x15\x02\x00')
2023-05-17 13:41:55.905 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-05-17 13:41:55.909 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=21)
2023-05-17 13:41:56.083 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x9432, NWK=0x9432, IEEE=0c:43:14:ff:fe:8a:c9:12, Capabilities=<MACCapabilities.AllocateShortAddrDuringAssocNeeded: 128>)
2023-05-17 13:41:56.083 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
[... cut ...]
2023-05-17 13:42:04.372 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x726B](WM25/L-Z): Device seen - marking the device available and resetting counter
2023-05-17 13:42:04.372 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x726B](WM25/L-Z): Update device availability -  device available: True - new availability: True - changed: False
2023-05-17 13:42:05.268 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x726B, Relays=[0xF254])
```

puddly, 2023-05-19:
What is this device? According to the log, its NWK is `0x9432` and its IEEE is `0c:43:14:ff:fe:8a:c9:12`.

stu247, 2023-05-19:
It is a roller shade, WM25/L-Z by Smartwings.

tube0013, 2023-05-19:
Smartwings don't support groups even though the cluster is there.

tube0013, 2023-05-19:
I do think ZHA shouldn't assume they go into a group successfully when the unsupported cluster command comes back.

PR #1212 (2023-05-26): Handle `UNSUP_CLUSTER_COMMAND` when scanning groups
Looks like some devices erroneously have a `Groups` cluster (0x0004) but do not actually support any commands.

Maybe we should raise a `zigpy.exceptions.UnsupportedCommand` error instead of returning the response?

Fixes #1210
closed 2023-05-26 by puddly