zigpy/zigpy #1211: Update zigpy for compliance with new Zigbee Protocol Specification Revision 23 / Zigbee R23 ?

Hedda, 2023-05-16:
Connectivity Standards Alliance (CSA) released Zigbee Pro Specification Revision 23 / R23 (a.k.a. "Zigbee PRO 2023") March 15th of 2023:

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

(This link was originally posted by @MattWestb in https://github.com/zigpy/zigpy/discussions/595 )

Also check out these other references for more high-level summaries: 

https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

https://www.silabs.com/blog/zigbee-pro-2023-spec-released-increases-security

https://staceyoniot.com/with-its-pro-2023-release-zigbee-tries-to-stay-relevant/

https://news.tmcnet.com/usubmit/2023/04/26/9802739.htm

brumik, 2023-05-22:
Hello, there was a lots of talk around this and I see that there is some new activity. Is there an approximate timeline for this feature? (month, half a year, year, more?). Thank you for your effort and answer.

MattWestb, 2023-05-22:
I think our devs is not starting working on this then its need updating all coordinators to R23 and most is never getting it but EZSP is having it but not stable and very likely TI is getting it also but only for current production hardware and no old ones.
Also for getting most out of R23 you need having all your routers running on Zigbee stack R23 and i think you can finding any devices on the market that is doing it today.
All IKEA device is updated to Zigbe stack R22 and ZCL R8 and perhaps latest update for Vindrikning can being R23 but is no high priority for them then all no current device (MG21) cant being updated to R23 then the hardware is having to less flash and RAM (MG1P) for running it.

brumik, 2023-05-23:
Thank you. I thought ZGP is part of this ticket.

Hedda, 2023-05-23:
> I thought ZGP is part of this ticket.

It is not, there is a separate issue open as a feature request for ZGP (Zigbee Green Power) support inside zigpy, see here -> https://github.com/zigpy/zigpy/issues/341

Hedda, 2023-06-09:
Silicon Labs Gecko SDK v4.3 (4.3.0.0) / Zigbee EmberZNet v7.3 (7.3.0.0) makes the Silabs Zigbee stack compatible with this new Zigbee Protocol Specification Revision 23 / R23 (Zigbee PRO 2023), but users of Silicon Labs firmware should be warned against flashing EZSP v12 firmware as of yet as it is not backwards compatible with EZSP v11 or older EZSP so canät yet be used with the bellows radio library for zigpy -> https://github.com/zigpy/bellows/issues/563

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.0.0.pdf

https://www.silabs.com/wireless/zigbee?tab=software#software

**Zigbee**
- Zigbee R23 compliance, with these Security enhancements among others:
  - Dynamic link key negotiation
  - Device interview to query devices before they are allowed to join
  - Trust Center Swap Out to replace an existing Trust Center with a new one
  - Frame Counter Synchronization
- Zigbee Direct Device (ZDD) support for:
  - Onboarding/commissioning
  - Communication to all Zigbee devices without a hub (Alpha), using Bluetooth LE
- Zigbee Smart Energy 1.4a compliance (Alpha)
- Enhancements to Zigbee Green Power (ZGP) APIs
- New Zigbee Security upgrade component for moving encryption keys from cleartext NVM3 tokens into secure storage

**Multiprotocol**
- Zigbee/OpenThread Concurrent Multiprotocol SoC sample app
- CPC GPIO expander module
- Zigbeed enhancements

PS: Silabs EZSP protocol version has also been incremented to EZSP v12 in Silicon Labs Zigbee EmberZNet v7.3 (7.3.0.0) firmware:

https://github.com/zigpy/bellows/issues/563

Hedda, 2023-12-06:
Has this already been implemented?

Hedda, 2024-11-07:
> Zigbee PRO 2023 introduces the coexistence of Zigbee and Zigbee Smart Energy devices, adds SubGHz support to North American and EU markets, and offers device onboarding and control via Bluetooth Low Energy.

zigpy does not yet support Zigbee Smart Energy, but FYI, the Connectivity Standards Alliance ratified Zigbee Smart Energy 1.4a

* https://csa-iot.org/newsroom/smart-energy-1-4a-for-zigbee-based-energy-management-technology/

  * https://csa-iot.org/developer-resource/specifications-download-request/

Hedda, 2025-05-06:
Still wanted as Zigbee PRO 2023 introduces the coexistence of Zigbee and Zigbee Smart Energy (ZSE) profile devices

puddly, 2025-05-06:
If there's anything specific you think needs to be implemented by zigpy *the library* to communicate with real devices, let me know. Otherwise, this is all automatically handled by coordinator firmware once R23 is enabled, it's all a layer below zigpy.