zigpy/zigpy #1220: 'OSRAM LIGHTIFY RT Tunable White' OTA failure

backcountrymountains, 2023-06-08:
I have an OSRAM recessed ceiling light:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=48042, maximum_buffer_size=64, maximum_incoming_transfer_size=0, server_mask=0, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.ExtendedActiveEndpointListAvailable|ExtendedSimpleDescriptorListAvailable: 3>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0102",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b04",
        "0xfc0f"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "LIGHTIFY RT Tunable White",
  "class": "osram.tunablewhite.OsramTunableWhite"
}
```
I can't get it to OTA update. I've downloaded the appropriate files from the LEDVANCE site as well as from Koenkk and put them in my ```/config/zigpy_ota/``` folder. 
The file is: ```ZLL_MK_0x01020510_CEILING_TW_OSRAM.ota```
I did the zigpy ota decode thing and it showed:
```
ZLL_MK_0x01020510_CEILING_TW_OSRAM.ota
Type: <class 'zigpy.ota.image.OTAImage'>
Header: OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl: 0>, manufacturer_id=4364, image_type=107, file_version=16909584, stack_version=2, header_string='NULL', image_size=132672, *device_specific_file=False, *hardware_versions_present=False, *key=ImageKey(manufacturer_id=4364, image_type=107), *security_credential_version_present=False)
Number of subelements: 4
Validation result: ValidationResult.UNKNOWN
```
The update has ```file_version=16909584``` and the ```image_type=107```.

My configuration.yaml is:
```
zha:
  database_path: /config/zigbee.db
  enable_quirks: true
  custom_quirks_path: /config/zhaquirks/
  source_routing: true
  ezsp_config:
      CONFIG_MAX_END_DEVICE_CHILDREN: 16
      CONFIG_SOURCE_ROUTE_TABLE_SIZE: 24
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
```

I've tried the ```Manage Zigbee Device:Ota:image_notify``` with ```QueryJitter=100``` sometimes with the ManufacturerCode and other stuff, sometimes not. But the logs always show:
```
2023-06-08 10:44:24.137 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=4, command_id=1, *direction=<Direction.Server_to_Client: 0>)
2023-06-08 10:44:24.138 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] Decoded ZCL frame: Ota:QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4364, image_type=12, current_file_version=16908629)
2023-06-08 10:44:24.138 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] Received command 0x01 (TSN 4): QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4364, image_type=12, current_file_version=16908629)
2023-06-08 10:44:24.139 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] OTA query_next_image handler for 'OSRAM LIGHTIFY RT Tunable White': field_control=0, manufacturer_id=4364, image_type=12, current_file_version=16908629, hardware_version=None, model='LIGHTIFY RT Tunable White'
2023-06-08 10:44:24.140 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] No OTA image is available
2023-06-08 10:44:24.140 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=4, command_id=2, *direction=<Direction.Client_to_Server: 1>)
2023-06-08 10:44:24.141 DEBUG (MainThread) [zigpy.zcl] [0xAFBB:3:0x0019] Sending reply: query_next_image_response(status=<Status.NO_IMAGE_AVAILABLE: 152>, manufacturer_code=None, image_type=None, file_version=None, image_size=None)
```
The logs show that the light has an ```image_type=12``` and ```current_file_version=16908629```.

Is the mismatch in ```image_type``` preventing the update? Is it possible to force the update? This thing is virtually useless (frequently drops off network) and I'm willing to risk it.

The OSRAM device seems identical to other  Ledvance recessed ceiling lights that I have:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4489, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0102",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x0b05",
        "0xfc01"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "LEDVANCE",
  "model": "RT TW",
  "class": "zigpy.device.Device"
}
```
If I were truly daring, would it be possible to force the OSRAM device to accept the LEDVANCE firmware?

Thanks.

puddly, 2023-06-08:
You can disable the code in zigpy to match images but it will at best result in the device rejecting the image at the very end, or the device bricking itself at the end of the update. Generally, the image type not matching means that the image is incompatible.

MattWestb, 2023-06-08:
From the log is the device need` image_type =12` = `0x000C` and the file is having the name `A19_W_10_year_IMG000C_00102428-encrypted.ota`

backcountrymountains, 2023-06-08:
@MattWestb That's my crisis, there is manufacturer LEDVANCE firmware that might match the image_type and there is OSRAM firmware that really seems like it should be correct because it is a Ceiling Tunable White but there is no firmware that is both OSRAM manufacturer id and image type 12.

Thanks for the idea, though. It was pretty slick. I didn't notice the image type in the name of the other firmware.

backcountrymountains, 2023-06-08:
Alternatively, where/how does zigpy locate/make the cache/dictionary of images? Is it supposed to be in my ```/config/zigpy_ota``` couldn't I modify the cache to point to whatever file I wanted to try and upload?

@MattWestb following your inspection of the file names, I found the ```RT_TW_IMG001C_00102428-encrypted.ota``` which I think is the file that zigpy used to update the LEDVANCE version of the same light. 

I think that either the Ceiling TW firmware or the LEDVANCE RT_TW firmware will work with this light but I don't know how to modify the zigpy matching logic.

puddly, 2023-06-08:
Forcing the upload of a specific image just isn't something that's supported without modifying *something*. You'll likely have better luck directly modifying the OTA image so that both zigpy and the device treat it like a normal OTA upgrade:

```Python
In [1]: import pathlib

In [2]: data = pathlib.Path("RT_TW_IMG001C_00102428-encrypted.ota").read_bytes()

In [3]: from zigpy.ota.image import parse_ota_image

In [4]: image, rest = parse_ota_image(data)

In [5]: image.header
Out[5]: OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl.0: 0>, manufacturer_id=4489, image_type=28, file_version=1057832, stack_version=2, header_string='EBL RGBW', image_size=170776, *device_specific_file=False, *hardware_versions_present=False, *key=ImageKey(manufacturer_id=4489, image_type=28), *security_credential_version_present=False)

In [6]: image.header.image_type = 107

In [7]: pathlib.Path("RT_TW_IMG001C_00102428-encrypted.patched.ota").write_bytes(image.serialize())
Out[7]: 170776
```

Here's the output file. Small chance, but it might work: [RT_TW_IMG001C_00102428-encrypted.patched.ota.zip](https://github.com/zigpy/zigpy/files/11692739/RT_TW_IMG001C_00102428-encrypted.patched.ota.zip)

backcountrymountains, 2023-06-08:
@puddly TL;DR: can you change the manufacturer code and image type of ```RT_TW_IMG001C_00102428-encrypted``` to ```manufacturer_id=4364``` ```image_type=12```, the patched firmware you posted is still 4489 and 107.

So, I think you're right that modifying the OTA image so that the device thinks it's the correct update is the way to go.

I tried initiating the update using the patched firmware file, but zigpy gave "Update needed: False"
```
2023-06-08 15:28:14.695 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Received command 0x01 (TSN 4): QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4364, image_type=12, current_file_version=16908629)
2023-06-08 15:28:14.697 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] OTA query_next_image handler for 'OSRAM LIGHTIFY RT Tunable White': field_control=0, manufacturer_id=4364, image_type=12, current_file_version=16908629, hardware_version=None, model='LIGHTIFY RT Tunable White'
2023-06-08 15:28:14.714 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] OTA image version: 16909584, size: 132672. Update needed: False
2023-06-08 15:28:14.714 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=4, command_id=2, *direction=<Direction.Client_to_Server: 1>)
2023-06-08 15:28:14.714 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Sending reply: query_next_image_response(status=<Status.NO_IMAGE_AVAILABLE: 152>, manufacturer_code=None, image_type=None, file_version=None, image_size=None)
```

I did the zigpy ota decode thing on the firmware and it shows ```image_type=107``` and ```manufacturer_id=4489``` instead of the OSRAM values(```manufacturer_id=4364``` ```image_type=12```) that the light expects.

```
homeassistant:/config/zigpy_ota# zigpy ota info RT_TW_IMG001C_00102428-encrypted.patched.ota 
RT_TW_IMG001C_00102428-encrypted.patched.ota
Type: <class 'zigpy.ota.image.OTAImage'>
Header: OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl: 0>, manufacturer_id=4489, image_type=107, file_version=1057832, stack_version=2, header_string='EBL RGBW', image_size=170776, *device_specific_file=False, *hardware_versions_present=False, *key=ImageKey(manufacturer_id=4489, image_type=107), *security_credential_version_present=False)
Number of subelements: 1
Validation result: ValidationResult.UNKNOWN
```
To inveigle the OSRAM light, which currently has ```manufacturer_id=4364``` ```image_type=12``` and ```current_file_version=16908629```, into accepting the LEDVANCE ```RT_TW_IMG001C_00102428``` firmware, I think those keys have to be changed.

I have no idea how you did your magic there. Can you just type ```python``` at a console and then put in those commands? What sorcery is this?

Thanks for the help!
Also, I tried modifying the ```__init.py__``` in the ```zigpy/ota``` directory from
```
 if self.key != ImageKey(manufacturer_id, img_type):
            return False
```
to
```
 if self.key != ImageKey(manufacturer_id, 107):
            return False
```
And zigpy did try to send the ```ZLL_MK_0x01020510_CEILING_TW_OSRAM.ota```, which I believe is the correct OSRAM firmware, but has the wrong image type (107 instead of 12).
```
2023-06-08 14:49:35.497 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Received command 0x01 (TSN 2): QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4364, image_type=12, current_file_version=16908629)
2023-06-08 14:49:35.499 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] OTA query_next_image handler for 'OSRAM LIGHTIFY RT Tunable White': field_control=0, manufacturer_id=4364, image_type=12, current_file_version=16908629, hardware_version=None, model='LIGHTIFY RT Tunable White'
2023-06-08 14:49:35.499 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] OTA image version: 16909584, size: 132672. Update needed: True
2023-06-08 14:49:35.499 INFO (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Updating: OSRAM LIGHTIFY RT Tunable White
2023-06-08 14:49:35.500 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=2, command_id=2, *direction=<Direction.Client_to_Server: 1>)
2023-06-08 14:49:35.501 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] Sending reply: query_next_image_response(status=<Status.SUCCESS: 0>, manufacturer_code=4364, image_type=107, file_version=16909584, image_size=132672)
```
I think the light must've done a simple check and refused the update because a the firmware was never sent from what I could see in the logs. I'm pretty sure zigpy was ready to send it because I moved it out of the ```zigpy_ota``` folder and then tried the update again and zigpy freaked out in the logs saying that the file was gone.

backcountrymountains, 2023-06-09:
So the answer to doing the magic is, indeed, just typing that stuff in. I tried in my HA docker but I kept getting some kind of circular import error ```from zigpy.ota.image import parse_ota_image: ImportError: cannot import name 'TYPE_CHECKING' from partially initialized module 'typing'``` probably because I have no idea what I'm doing.

Anyway, when I finally figured out some way of using ```from zigpy.ota.image import parse_ota_image``` I managed to modify the LEDVANCE firmware with ```manufacturer_id=4364 image_type=12```. I disabled version number checking in zigpy by using the all powerful ```#```. Then I sent the ```image_notify```.

And boom, it's uploading the firmware right now.
....
Twice now the update has stopped at 86%...
```2023-06-08 18:43:40.682 DEBUG (MainThread) [zigpy.zcl] [0x03E3:3:0x0019] OTA upgrade progress: 86.2```
Perchance the LEDVANCE firmware at 160kb is too big for the OSRAM which might need to use the ZLL firmware that is 130kb...

backcountrymountains, 2023-06-09:
so I did the puddly magic on the OTA file header for the ```ZLL_MK_0x01020510_CEILING_TW_OSRAM.ota``` and then sent the ```image_notify``` and it worked!
[image]
The update was successful! Now I have a crappy OSRAM recessed can light on the latest firmware from 2019! Exciting!

side note: it's important to remove the ```#``` on the verification logic in ```/ota/__init.py__```:
```
     if ver >= self.version:
            return False
```
or else zigpy will just keep updating forever.

Now I just have to do it for my ```Flex RGBW Pro``` led strip, which somehow also has mismatched ```manufacturer_id``` and ```image_type```

Side, side note: How can it be that the devices will accept whatever firmware that fits as long as it has the correct ```manufacturer_id``` and ```image_type```, which can be easily modified? What's the point in encrypting it, if it can be so trivially changed and uploaded? Doesn't seem super secure.

Thanks for the help!

closed 2023-06-09 by backcountrymountains

MattWestb, 2023-06-09:
OSRAM have encrypted the firmware and both types is using the same key for it.
The meta data is wrapped outside the real flash file so changing it is not braking encryption of the file (IKEA is signing there files but its the same changing the meta data is working and not braking the OTA update).

As long the hardware is the same it shall not being any problems but its one large braking warning on doing this then is likely gets different hardware versions that is not compatible.

MattWestb, 2023-06-09:
Opa was forgetting:
## Great thanks our @puddly for the hack !!!!

puddly, 2023-06-09:
Awesome job @backcountrymountains!!! I'm really surprised this worked.

I wonder if you could modify the image header to bump up the file version as well (to something very high), so that the file can be utilized without changing zigpy at all.

MattWestb, 2023-06-09:
I think that shall working OK like we was doing one IKEA Styrbar with wrong version in the metadata.