zigpy/zigpy #1226: Merten Schneider  -  Wiser wireless switch 1-gang or 2-gang FLS/SYSTEM-M/4 - Model WDE002906

tech2mar, 2023-06-28:
Hey friends,

First of all, thanks for the great work,
I am looking for help on this: 
https://www.zigbee2mqtt.io/devices/WDE002906.html#schneider%2520electric-wde002906

On the z2m the push button seems to work fine even with the dim function.
On the ZHA, it shows me 4 battery states and no push or button function.

Can anyone help?

Hedda, 2023-07-07:
Zigbee devices that do not only use standard configuration and parameters (default ZCL clusters and attributes) but instead also implement custom manufacturer clusters and attributes (also known as "quirks") will need a custom handler/converter/parser/translator as Python script code in the upstream "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/)" library repository to extend custom manufacturer device "quirk" support for specific non-standard ZCL clusters and attributes (which Zigbee gateway implementations that depends on zigpy and the [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/) library, like Home Assistant s ZHA integration, then can make use of).

https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices

https://github.com/zigpy/zha-device-handlers/#readme

If you can not code that Python script code yourself to write the needed custom handler/converter/translator for adding that specific device to the "ZHA Device Handlers" repository/library then suggest submitting a "device support request" (with device signature and diagnostic information) as new issues -> [https://github.com/zigpy/zha-device-handlers/issues](https://github.com/zigpy/zha-device-handlers/issues)

Without a "device support request" as a new issue posted to the "ZHA Device Handlers" repository/library requesting support for an unsupported device or device feature with Device signature and Diagnostic information the ZHA integration developers will not even know about the device unless they by random chance happened to have bought it themselves.

The reason is why non-standard devices need a custom handler/converter/translator is explained in ZHA integration documentation here -> [https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling](https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling)

Zigbee devices that use clusters and attributes that are standard in the official ZCL (Zigbee Cluster Library) do not need custom handlers/converters/translators as explained in the ZHA integration documentation here -> [https://www.home-assistant.io/integrations/zha#knowing-which-devices-are-supported](https://www.home-assistant.io/integrations/zha#knowing-which-devices-are-supported)

PS: Off-topic but FYI, this also works kind of similar for Zigbee2MQTT which also requires a custom handler/converter/translator for specific devices -> [https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)

tech2mar, 2023-07-07:
Thanks for the explanation, will it be possible to merge what in the z2m side is done with the zha? 🤔

I think working on the same thing in 2 difrent directions a great way to lose time and resources.

Thanks a lot I will do my best.

Hedda, 2023-07-08:
> will it be possible to merge what in the z2m side is done with the zha?

No, Zigbee2MQTT and zigbee-herdsman are not even coded in the same programming-language as ZHA and zigpy.

Again, if you can not code Python the follow ZHA docs on where and where to post a new device request -> https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices

closed 2024-01-12 by github-actions[bot]