zigpy/zigpy #1227: Recent changes to "retryable"

mdeweerd, 2023-07-06:
The recent changes to "retryable" impacts at least one functionnality in zha-toolkit.  I suppose that this was published in HA 2023.7.0 .

While, I will fix/workaround in zha-toolkit, I am also reporting it here in case the impact might reach further than just zha-toolkit.

Basically zha-toolkit is reusing the decorator "retryable", but it looks like the "tries" option is propagated to the decorated function.

I haven't checked that my self in detail yet.

See:
https://github.com/mdeweerd/zha-toolkit/pull/171#discussion_r1254109865

mdeweerd, 2023-07-06:
Thenks @puddly for your fast reply.

FYI, here is the code that worked before but does not work anymore:

```python
from zigpy.exceptions import ControllerException, DeliveryError
from zigpy.util import retryable

@retryable(
    (
        DeliveryError,
        ControllerException,
        asyncio.CancelledError,
        asyncio.TimeoutError,
    ),
    tries=1,
)
async def cluster_read_attributes(
    cluster, attrs, manufacturer=None
) -> tuple[list, list]:
    """Read attributes from cluster, retryable"""
    return await cluster.read_attributes(attrs, manufacturer=manufacturer)
```

It appears that when calling the above cluster_read_attributes from the zigpy, the tries argument is propagated to cluster_read_attributes where "tries" does not exist.

```plaintext
#
# Call from zha_toolkit to decorated cluster_read_attributes
#
res = await u.cluster_read_attributes(basic_cluster,[4, 0, 1, 5, 7, 0xFFFE],tries=1)
^^^^^^^^^^^^^^^^^^^^^^^^^^
#
# zigpy's retryable calling the original function,
#
File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 153, in wrapper
return func(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^
#
# Apparently with "tries" in kwargs
#
TypeError: cluster_read_attributes() got an unexpected keyword argument 'tries'
```

https://github.com/zigpy/zigpy/blob/0.56.1/zigpy/util.py#L153

So it looks like "tries" (and possibly delay) needs to be unset in kwargs.

puddly, 2023-07-06:
In the code you just posted, `tries` is being passed in explicitly as a keyword argument:

```python
res = await u.cluster_read_attributes(basic_cluster,[4, 0, 1, 5, 7, 0xFFFE],tries=1)
```

It no longer exists as a kwarg provided by the decorator so you just need to remove it.

mdeweerd, 2023-07-06:
Thanks for pointing that out.  I integrated the old version of retryable in zha-toolkit.