zigpy/zigpy #1231: Invalid value support

galmok, 2023-07-16:
Disclaimer: I use Home Assistant which has ZigBee integration provided by zigpy. 

How does zigpy handle invalid values?

According to the ZCL documentation I located here:

https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf

In section 2.6.2 (page 84+), the data types are listed and with each data type, there is also a specific value that indicates an invalid value.

For data type 0x25 this is 0xFFFFFFFFFFFF. 
For data type 0x2A this is 0x800000.

I mention those as those two are values reported by a ZigBee device I use and because the values are logged in Home Assistant as valid values.

I don't know if this is a problem with zigpy or with Home Assistant, i.e. I don't know if HA gets enough information about the data type to know if a certain value is an invalid value.

From the HA log I get this:

```
2023-07-15 16:39:11.196 DEBUG (MainThread) [zigpy_znp.api] 
Received command: AF.IncomingMsg.Callback(
	GroupId=0x0000, 
	ClusterId=1794, 
	SrcAddr=0x9355, 
	SrcEndpoint=2, 
	DstEndpoint=1, 
	WasBroadcast=<Bool.false: 0>, 
	LQI=36, 
	SecurityUse=<Bool.false: 0>, 
	TimeStamp=194464, 
	TSN=0, 
	Data=b'\x18\x97\x0A\x00\x00\x25\x56\xDD\x33\x02\x00\x00\x00\x04\x2A\x97\x01\x00\x00\x02\x18\x00\x00\x01\x25\xFF\xFF\xFF\xFF\xFF\xFF\x02\x01\x25\xFF\xFF\xFF\xFF\xFF\xFF', 
	MacSrcAddr=0xC3C4, 
	MsgResultRadius=28)
2023-07-15 16:39:11.198 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(
	src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x9355), 
	src_ep=2, 
	dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), 
	dst_ep=1, 
	source_route=None, 
	extended_timeout=False, 
	tsn=0, 
	profile_id=260, 
	cluster_id=1794, 
	data=Serialized[b'\x18\x97\n\x00\x00%V\xdd3\x02\x00\x00\x00\x04*\x97\x01\x00\x00\x02\x18\x00\x00\x01%\xff\xff\xff\xff\xff\xff\x02\x01%\xff\xff\xff\xff\xff\xff'], 
	tx_options=<TransmitOptions.NONE: 0>, 
	radius=28, 
	non_member_radius=0, 
	lqi=36, 
	rssi=None)
2023-07-15 16:39:11.198 DEBUG (MainThread) [zigpy.zcl] [0x9355:2:0x0702] Received ZCL frame: 
	b'\x18\x97\n\x00\x00%V\xdd3\x02\x00\x00\x00\x04*\x97\x01\x00\x00\x02\x18\x00\x00\x01%\xff\xff\xff\xff\xff\xff\x02\x01%\xff\xff\xff\xff\xff\xff'
2023-07-15 16:39:11.199 DEBUG (MainThread) [zigpy.zcl] [0x9355:2:0x0702] Decoded ZCL frame header: ZCLHeader(
	frame_control=FrameControl(
		frame_type=<FrameType.GLOBAL_COMMAND: 0>, 
		is_manufacturer_specific=0, 
		direction=<Direction.Client_to_Server: 1>, 
		disable_default_response=1, 
		reserved=0, 
		*is_cluster=False, 
		*is_general=True), 
	tsn=151, 
	command_id=10, 
	*direction=<Direction.Client_to_Server: 1>)
2023-07-15 16:39:11.200 DEBUG (MainThread) [zigpy.zcl] [0x9355:2:0x0702] Decoded ZCL frame: Metering:Report_Attributes(
	attribute_reports=[
		Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=36953430)), 
		Attribute(attrid=0x0400, value=TypeValue(type=int24s, value=407)), 
		Attribute(attrid=0x0200, value=TypeValue(type=bitmap8, value=<bitmap8: 0>)), 
		Attribute(attrid=0x0100, value=TypeValue(type=uint48_t, value=281474976710655)), 
		Attribute(attrid=0x0102, value=TypeValue(type=uint48_t, value=281474976710655))])
2023-07-15 16:39:11.201 DEBUG (MainThread) [zigpy.zcl] [0x9355:2:0x0702] Received command 0x0A (TSN 151): Report_Attributes(
	attribute_reports=[
		Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=36953430)), 
		Attribute(attrid=0x0400, value=TypeValue(type=int24s, value=407)), 
		Attribute(attrid=0x0200, value=TypeValue(type=bitmap8, value=<bitmap8: 0>)), 
		Attribute(attrid=0x0100, value=TypeValue(type=uint48_t, value=281474976710655)), 
		Attribute(attrid=0x0102, value=TypeValue(type=uint48_t, value=281474976710655))])
2023-07-15 16:39:11.201 DEBUG (MainThread) [zigpy.zcl] [0x9355:2:0x0702] Attribute report received: 
	current_summ_delivered=36953430, 
	instantaneous_demand=407, 
	status=<bitmap8: 0>, 
	current_tier1_summ_delivered=281474976710655, 
	current_tier2_summ_delivered=281474976710655
```

It seems zigpy reports the values back to HA. It should not have reported the current_tier1_summ_delivered nor the current_tier2_summ_delivered attribute as they are invalid. Or it should have conveyed the invalid state of the values somehow (which it may have and HA failed to pick it up).

galmok, 2024-05-27:
Is it possible to add this inside value support using a quirk? It should be a quirk that affects all devices.

puddly, 2024-05-27:
This would need to be implemented in zigpy. Quirks are for device specific problems. General "invalid value" checking belongs in the library.

galmok, 2024-11-24:
This is still unresolved. I still have problems with false readings due to this.

galmok, 2025-05-24:
I think this is being handled here, but if these are different issues, this issue is this open: 

https://github.com/home-assistant/core/issues/96894

closed 2025-11-27 by github-actions[bot]