zigpy/zigpy #1234: [Future request] Filtering commands on ZCL level for both uni and broadcast frames

MattWestb, 2023-07-25:
Broadcast can being done in classic  mode = broadcast the command 3 time or ZB3 = sending one broadcast and and listening if all neighbors is repeating it and if OK no more broadcast of the frame if not retry up to 3 times (also called passive acking).
In the network layer all is being done with different frame sequences but in ZCL all is having the same = its the same command sent from the device.
We was having this problem with IKEA Symfonisk first gen that is multi spammer of the network but the last firmware is only using unicast so its better.

The new RODRET OnOff switch is default doing unucast to the coordinator and if binding it to one light group is doing both.

From point of view the controller is making it very nice but ZHA is not.
From Wireschark:
```
No.	Time	PAN	Protocol	IEEE Src	IEEE Dst	Zigbee Src	Zigbee Dst	ZBN Dst	Group Nr	ZBN Seq	ZBA Seq	ZDP Seq	Nwk Seq	Src EP	Dst EP	Command	Info
885	08:21:01,544091	0x2af0	ZigBee HA	0x86ac	0x4548	0x86ac	0x0000	0x0000		44	135		20	1	1		ZCL Level Control: Stop with OnOff, Seq: 83
886	08:21:01,551156	0x2af0	ZigBee HA	0x4548	0x0000	0x86ac	0x0000	0x0000		44	135		16	1	1		ZCL Level Control: Stop with OnOff, Seq: 83
887	08:21:01,561349	0x2af0	ZigBee HA	0x86ac	0x4548	0x86ac	Broadcast	0xfffd	0x0000	45	136		21	1			ZCL Level Control: Stop with OnOff, Seq: 83
888	08:21:01,570109	0x2af0	ZigBee HA	0x4548	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		17	1			ZCL Level Control: Stop with OnOff, Seq: 83
889	08:21:01,583948	0x2af0	ZigBee	0x0000	0x4548	0x0000	0x86ac	0x86ac		189	135		27	1	1		APS: Ack, Dst Endpt: 1, Src Endpt: 1
890	08:21:01,591513	0x2af0	ZigBee HA	0xd855	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		126	1			ZCL Level Control: Stop with OnOff, Seq: 83
891	08:21:01,601097	0x2af0	ZigBee HA	0x7421	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		233	1			ZCL Level Control: Stop with OnOff, Seq: 83
892	08:21:01,617255	0x2af0	ZigBee HA	0xb824	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		60	1			ZCL Level Control: Stop with OnOff, Seq: 83
893	08:21:01,620779	0x2af0	ZigBee HA	0x24d6	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		231	1			ZCL Level Control: Stop with OnOff, Seq: 83
894	08:21:01,630972	0x2af0	ZigBee HA	0xfea9	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		220	1			ZCL Level Control: Stop with OnOff, Seq: 83
895	08:21:01,640709	0x2af0	ZigBee HA	0x4e1f	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		32	1			ZCL Level Control: Stop with OnOff, Seq: 83
896	08:21:01,650499	0x2af0	ZigBee HA	0xadc3	0xffff	0x86ac	Broadcast	0xfffd	0x0000	45	136		180	1			ZCL Level Control: Stop with OnOff, Seq: 83
```
and the long on the device card in ZHA:
```
Logbook
July 25, 2023
IKEA RODRET Dimmer Test Off event was fired
10:33:19 - 12 minutes ago
IKEA RODRET Dimmer Test Off event was fired
10:33:19 - 12 minutes ago
```
As can see the device is first sending one command with ZCL sequence number 83 to its parent addressed to 0x0000 and the parent is sending it to the 0x0000 and also sending one application ack back to the controllers parent = looks very nice.
Then the device is sending one broadcast to its parent to group 0x0000 and the parent is broadcasting it and all neighbors is relaying it OK. All is having different sequence number in the network layer but the same in the ZCL layer (83) also with the first unnicast to the coordinator.
What i see is the controller doing it OK and also the routers and coordinator in the network.
But in ZHA is firing 2 identical events in the log = if doing  one automation on it its being triggered 2 times that is not good.

This is tested on Bellows with latest EZSP running zigbeed on RCP. I think TI is doing the filtering in the firmware but EZSP have never doing i what i have seen.

For getting it working OK and not making bad looking logs and events and also making problems in automatons we need filtering all frames from one device on ZCL sequence number so not detecting one command sent from the device and multiple ones if being sent with uni and / or broadcast.

PS: RODRET is (from FCC papers) running on EZSP 7.1.X so very current Zigbee stack and is not needing quirks for sending all command and reporting its status OK as expected its good ZCL compatible so it have not DA and is one very nice device and the OpenCloce and Chortcut button is in the pipe and shall running the same firmware and only having different data setting.

Thanks i advance.

puddly, 2023-07-25:
Is there some simple window that we can use for filtering, like 100ms? So if we receive a frame with `zcl.seq == 123` and then 99ms later a second frame with `zcl.seq == 123`, we would silently drop the second one?

MattWestb, 2023-07-25:
I have not looking how long one broadcast is "waveing" around and if its classic broadcast its being repeated 3 times in my production network only in one RCP test that is not so large.
If i remember is you having one pritty large production network can you looking how long its taking until its having decay after the 3 repeating by sniffing it.

Also then all counters is for frames (not the main counters is much higher) is 0Xff and if doing it / device it can being very long without problems before one command is being sent with the same number. (one IKEA controller OTA is reusing it around 5 times doing one upgrade is taking around 2 hours but i probably having wrong then you counting how large the frame is and the OTA file).

Also this is not doing anything with the mesh network its only what EZSP is sending to the host system and all broadcast is only "piped" to it and not filtered that other coordinators is doing.

Thanks for looking on it **P** !

MattWestb, 2024-01-21:
Still importation getting implanted then its making problems with automatons.

MattWestb, 2024-07-19:
Still happening with some uni and broadcast from devices and making problems doing automatons  of them.

puddly, 2024-07-19:
We added packet filtering a while ago. Does it not help with those devices? If not, could you attach a debug log?

MattWestb, 2024-07-19:
Thanks for feedback.
I shall doing some test in the weekend and see if its working OK with some IKEA devices and also the famous tuya TS04F and reporting back.

zigbeefordomoticz, 2025-02-17:
for information, as mentioned here #1553 , I found a device which is doing its report with always the sun 0x00