zigpy/zigpy #1235: Wrong parsed ZCL R8 attribute

MattWestb, 2023-07-25:
All updated IKEA device that support `product_code` on basic cluster is wrongly parsed of Zigpy and is having extra charters:
```
Clusters
Basic (Endpoint id: 1, Id: 0x0000, Type: in)

ATTRIBUTES

COMMANDS
Attributes of the selected cluster
product_code (id: 0x000a)
 
Value
b'E2201'
 
Manufacturer Code Override
Value
```
The return data shall being `'E2201'` without  the leading `b`.
In sniff the device is sending 5 octets and not 6 as Zigpy is "thinning":
```
IEEE 802.15.4 Data, Dst: 0x4548, Src: 0x86ac
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x86ac
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Read Attributes Response, Seq: 166
    Frame Control Field: Profile-wide (0x18)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...1 .... = Disable Default Response: True
    Sequence Number: 166
    Command: Read Attributes Response (0x01)
    Status Record, Octets: 45:32:32:30:31
        Attribute: Unknown (0x000a)
        Status: Success (0x00)
        Data Type: Octet String (0x41)
        Octet String: 45:32:32:30:31
```
Then its one new attribute and is not being used of may device is not high priority but it wold being nice if it was right done in the future (IKEA have starting adding more ZCL R8 attributes but its looks OK what  i have seen).

Thaks in advance.

puddly, 2023-07-25:
I think the problem is that the frontend is showing a bytestring. The `b` prefix is just a Python object being converted into a string in the wrong way.

More specifically, the `b` prefix indicates that it's a bytestring: it's not considered text until we decode it with a specific codec:

```Python
>>> b'Hello' == 'Hello'
False
>>> b'München'
  File "<stdin>", line 1
    b'München'
    ^^^^^^^^^^
SyntaxError: bytes can only contain ASCII literal characters
>>> 'München'
'München'
>>> 'München'.encode('utf-8')
b'M\xc3\xbcnchen'
>>> b'M\xc3\xbcnchen'.decode('utf-8')
'München'
```

So the fix for this probably belongs in ZHA.

MattWestb, 2023-07-25:
Thanks for looking at it our Puddly !!

Can you moving it to the ZHA re-pro or must i reposing it there, then you is member of both organisations ?

puddly, 2023-07-25:
You can leave it here for now, it'll be fixed (eventually) either way

MattWestb, 2023-07-25:
Thanks **P** !!

closed 2024-01-28 by github-actions[bot]