zigpy/zigpy #1242: Wrong KeyData representation

DamKast, 2023-08-07:
I noticed that the KeyData type class defined in the named.py file is representing the object using a different byte order than other type objects.
For example the EUI64 type represents the object taking account of the little endianness, while the KeyData type is using the same byte order for the representation and when serialize() method is used.

```
# That seems wrong
>>> obj = t.KeyData(b"\x03\x04\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x02")
>>> obj
03:04:01:01:01:01:01:01:01:01:01:01:01:01:01:02
>>> obj.serialize()
b'\x03\x04\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x01\x02'

# That seems correct
>>> obj = t.EUI64(b"\x01\x02\x03\x04\x05\x06\x07\x08")
>>> obj
08:07:06:05:04:03:02:01
>>> obj.serialize()
b'\x01\x02\x03\x04\x05\x06\x07\x08'

```

puddly, 2023-08-07:
I can't actually find a single device that uses the `KeyData` data type over-the-wire for me to confirm (the device would need a `Commissioning` / 0x0015 cluster), but I think you may be right. The spec describes it as follows:

[image]

Wireshark also dissects the same way as it does the EUI64/IEEE addresses: https://github.com/wireshark/wireshark/blob/f9846ecb7c167088c8f8b1f2d75a3a90d168730f/epan/dissectors/packet-zbee-zcl.c#L2356-L2369

Realistically, we only use this cluster in zigpy as a data type for representing key data so the endianness doesn't make a difference at the moment.

DamKast, 2023-08-07:
It is actually not a big deal, but I was really confused when I looked at the serial data going through UART. I was used to see data in little endian, as to say the python representation in the opposite order than the order in which data is serialized (like EUI64 or even uint16_t).
I am using the KeyData type to deserialize/serialize keys from zboss to zigpy, this is why I was looking at this.

MattWestb, 2023-08-07:
Wireshark is not ZCL R7 compatible then some function is not implemented and some small things is not working OK so not taking it for 110% as correct reference better the Zigbee spec.

konistehrad, 2023-11-24:
If you're still looking for a test case on this, the Green Power `pairing` command sends key data for OOB keys found in the RunLessWire switches.

closed 2024-05-29 by github-actions[bot]