zigpy/zigpy #1248: Electrical Measurement Cluster not correctly supported?

Tropaion, 2023-08-23:
I created my own [zigbee device](https://github.com/Tropaion/ZigBee_SmartMeter_Reader/blob/main/software/zigbee/components/zigbee_electricity_meter/src/zb_electricity_meter.c#L198) which reads phase a, b, and c data from an smartmeter.
In detail, these attributes are sent:
```
Attribute Set 0x03: AC (Non-phase Specific) Measurements (S. 303)
electrical power P+ (W): 0x0304 TotalActivePower

Attribute Set 0x05: AC (Single Phase or Phase A) Measurements (S. 306)
voltage L1: 0x0505 RMSVoltage
current L1: 0x0508 RMSCurrent

Attribute Set 0x09: AC Phase B Measurements (S. 313)
voltage L2: 0x0905 RMSVoltagePhB
current L2: 0x0908 RMSCurrentPhB

Attribute Set 0x0A: AC Phase C Measurements
voltage L3: 0x0A05 RMSVoltagePhB
current L3: 0x0A08 RMSCurrentPhB
```
But ZHA only creates the three entities power factor, phase A current and phase A voltage even though I don't even report power factor. How does ZHA decide which entities are to be provided?
[image]

Thanks for your help.

TheJulianJES, 2023-08-23:
The sensors are defined in https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py

Tropaion, 2023-08-24:
@TheJulianJES Thank you, but this didn't really answer my question and I'm pretty bad in python. How does ZHA decide for which attribute in a cluster it creates an entity? Because it doesn't really make sensor that it creates an power factor entity when I don't even send this. And it also looks like the MeasurementType Attribute is ignored. I set it to `111000` to indicate that Phase A, B and C is supported.

puddly, 2023-08-25:
Can you upload the diagnostics JSON for your device?

Tropaion, 2023-08-25:
Yes, here: [zha-066c8dbded61cfafe1400bbfe2fa75da-github.com_Tropaion SmartMeter-be799845130af87e0a4c5eabbb445262.json.txt](https://github.com/zigpy/zigpy/files/12443039/zha-066c8dbded61cfafe1400bbfe2fa75da-github.com_Tropaion.SmartMeter-be799845130af87e0a4c5eabbb445262.json.txt)

Hedda, 2023-08-29:
> How does ZHA decide which entities are to be provided?

FYI, Zigbee devices that do not only use common configuration and parameters standard (default ZCL clusters and attributes in the Zigbee Cluster Libary specification) but instead also implement or custom manufacturer clusters and attributes will usually need a custom handler/converter/parser/translator (also known as "quirks") as Python script code in the upstream "[ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/)" library repository to extend custom manufacturer device "quirk" support for specific non-standard ZCL clusters and attributes (which Zigbee gateway implementations that depends on zigpy and the [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/) library, like Home Assistant s ZHA integration, then can make use of). 

https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices

https://github.com/zigpy/zha-device-handlers/#readme

If you can not code that Python script code yourself to write the needed custom handler/converter/translator (a.k.a. quirk) for adding that specific device to the "ZHA Device Handlers" repository/library then suggest submitting a "device support request" (with device signature and diagnostic information) as new issues -> https://github.com/zigpy/zha-device-handlers/issues

Without a "device support request" as a new issue posted to the "ZHA Device Handlers" repository/library requesting support for an unsupported device or device feature with Device signature and Diagnostic information the ZHA integration developers will not even know about the device unless they by random chance happened to have bought it themselves.

> How does ZHA decide for which attribute in a cluster it creates an entity? 

Normally there is no need to also add new or extend the existing ZHA device types, however, sometimes it can be needed in addition to implementing support for uncommon ZCL clusters and attributes as entities in the ZHA integration that have not been seen before.

The reason why non-standard devices need a custom handler/converter/translator is explained in layman's terms in ZHA integration documentation here -> https://www.home-assistant.io/integrations/zha#how-to-add-support-for-new-and-unsupported-devices and the reason why common devices work without ZHA Device Handlers is explained here -> https://www.home-assistant.io/integrations/zha#knowing-which-devices-are-supported (that is Zigbee devices that use clusters and attributes that are included by default in the official ZCL (Zigbee Cluster Library) specification standard do not need custom handlers/converters/translators).

PS: Off-topic but another FYI, this also works kind of similar for Zigbee2MQTT which also requires a custom handler/converter/translator for specific devices. See -> https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Tropaion, 2023-08-29:
> will usually need a custom handler/converter/parser/translator

@Hedda I know, I have written a few of these device handlers myself, but this doesn't apply to my self-developed device, since I did everything according to the ZCL specification (doesn't mean I didn't possibly make some mistakes).

> Normally there is no need to also add new or extend the existing ZHA device types, however, sometimes it can be needed in addition to implementing support for uncommon ZCL clusters and attributes as entities in the ZHA integration that have not been seen before.

I think the cluster I'm using isn't that uncommon either, just, instead of measuring one phase I'm measuring all three, like every smartmeter. For me, it looks like ZHA is not acting according to the standard and ignores the mandatory attribute `Attribute Set 0x00: Basic Information (S. 299)` value, which specifies the type of measurements to expect from the cluster.

TheJulianJES, 2023-08-29:
> it doesn't really make sense that it creates an power factor entity when I don't even provide it

Yes, that probably shouldn't happen. The check whether an entity should be created or not is here:
https://github.com/home-assistant/core/blob/fe713cec8f563c3171926d30e015bcfb78dd69cf/homeassistant/components/zha/sensor.py#L149-L155

In the future, we might want to change that check to see if a value is in the attribute cache. Similar to this:
https://github.com/home-assistant/core/blob/fe713cec8f563c3171926d30e015bcfb78dd69cf/homeassistant/components/zha/number.py#L398-L410

However, then we'd need to initialize the attribute in the cluster handler, so it's read when pairing or restarting HA.
It might also break battery-powered devices with sensors though.
Maybe we should just add that check to the `power_factor` entity. I've never had a device that actually supports power factor.

---

Like mentioned in [my previous comment](https://github.com/zigpy/zigpy/issues/1248#issuecomment-1690724103), all attributes for which a sensor entity could possibly be created (like `active_power`, `power_factor`, ...) are defined in that file with a `MULTI_MATCH`.

Tropaion, 2023-08-29:
Sadly, I'm not good in python so even if you show me the code, I don't really understand it.

> In the future, we might want to change that check to see if a value is in the attribute cache.

I don't thing that's necessary, isn't pretty simple, at least for data that is sent. When an attribute is updated by the device just create an entity for it.

power factor is not really the problem, since me device provides that, just haven't implemented it yet. 

My real problem is that my smartmeter is periodically sending voltage and current of ALL three phases, but ZHA only creates an entity for the first phase. Shouldn't this problem have already occurred a few times? Three phase devices are not that uncommon. The cluster also provides and attribute which says which phases are provided, but ZHA ignores that. Is there a way to force the creation of entities? Without it I can't really use my measurements.

TheJulianJES, 2023-10-25:
FYI, at least the power factor entity being created unnecessarily will be [fixed](https://github.com/home-assistant/core/pull/102133) in HA 2023.11.0.

> ALL three phases, but ZHA only creates an entity for the first phase. Shouldn't this problem have already occurred a few times?

Most people are just using smart plugs (with a single phase).

> Is there a way to force the creation of entities? 

You'll need to modify ZHA's `sensor.py` and add "matches" for the attributes to create sensors. You might also need to initialize the attributes here: [zha/core/cluster_handlers/homeautomation.py#L78-L91](https://github.com/home-assistant/core/blob/7c93d4fccf75dd486355ab4728d568bec7af3ff8/homeassistant/components/zha/core/cluster_handlers/homeautomation.py#L78-L91)

Tropaion, 2023-10-27:
@TheJulianJES Okay, thank you for the information. I will try to look into it as soon as I have time, but I'm pretty bad at python.
Is there a guide somewhere how to modify and test python code of an Plugin with an local HA instance?

closed 2024-05-02 by github-actions[bot]