zigpy/zigpy #1253: [3.12.0rc2 ] New enum changes

cdce8p, 2023-09-06:
Just a heads up, `3.12.0rc2` seems to include a breaking change for enums.
In particular initializing enums without members will now raise a `TypeError`.

Haven't had time to fully understand it yet but this at least causes the Home Assistant test suite for `zha` to break again.

- https://github.com/python/cpython/issues/108682
- https://github.com/python/cpython/pull/108704

/CC @puddly

ethanfurman, 2023-09-07:
I thought this was resolved in #1236 ?

puddly, 2023-09-07:
Unfortunately, unit tests that passed in rc1 now fail with rc2. We rely on empty enums with a `_missing_` hook to still be usable. Here's a minimal example to reproduce:

```Python
import enum

class MyEnum(enum.IntEnum):
    # No members

    @classmethod
    def _missing_(cls, value):
        member = cls._member_type_.__new__(cls, value)
        member._name_ = f"missing_{value}"
        member._value_ = value

        return member

assert repr(MyEnum(123)) == "<MyEnum.missing_123: 123>"
```

ethanfurman, 2023-09-07:
My apologies.  I tried adjusting the last fix to still work with your use-case*, but that allowed the previous bug to still exist.

I would suggest that you call the `_missing_` method directly in `AlwaysCreateEnumType.__call__()`, instead of calling `cls.__new__()`.

\* To be clear: `_missing_` was intended to return existing members for enums/flags, and create only composite members for flags.  The ability to create new members using the stdlib Enum family was a bug.  Customising `EnumMeta` is the right thing to do.

cdce8p, 2023-09-07:
Just calling `_missing_` doesn't seem to be enough as it's also used for enums which do have values. I ended up using the implementation from `3.11.5`. That seems to work in all cases #1254.

puddly, 2023-09-22:
Sure! I've released a new version, it'll be in HA Core `dev` in time for the beta.

PR #1254 (2023-09-11): Add custom implementation of Enum.__new__
Fixes: #1253
/CC @puddly