zigpy/zigpy #1259: Legrand OTA issue: "invalid image"

muzzy124, 2023-09-25:
There is a problem with OTA updates within zigpy/zha in home assistant for Legrand zigbee devices.

An example of trying to update 0x003843ff ("38" in terms of legrand) legrand micromodule firmware to the latest 1021-0010-003a4203-NLM.zigbee firmware from legrand website using zha in HA 2023.9.3:

2023-09-25 21:37:35.377 DEBUG (MainThread) [zigpy.zcl] [0x5C87:1:0x0019] Decoded ZCL frame: Ota:upgrade_end(status=<Status.INVALID_IMAGE: 150>, manufacturer_code=4129, image_type=16, file_version=3817987)
2023-09-25 21:37:35.377 DEBUG (MainThread) [zigpy.zcl] [0x5C87:1:0x0019] Received command 0x06 (TSN 173): upgrade_end(status=<Status.INVALID_IMAGE: 150>, manufacturer_code=4129, image_type=16, file_version=3817987)
2023-09-25 21:37:35.378 DEBUG (MainThread) [zigpy.zcl] [0x5C87:1:0x0019] OTA upgrade_end handler for ' Legrand  Micromodule switch': status=Status.INVALID_IMAGE, manufacturer_id=4129, image_type=16, file_version=3817987

The same issue is with z2m and they've found the solution here:
https://github.com/Koenkk/zigbee-OTA/issues/328
with such an explanation:
https://github.com/Koenkk/zigbee-OTA/issues/328#issuecomment-1732671224

stevenwfoley, 2024-04-18:
I'm now also receiving the same INVALID_IMAGE error with the Legrand Smart Outlet - WNRR15 (NLPM)
Attempting to install firmware version 83 (1021-0011-005343FF-NLP.zigbee)

stevenwfoley, 2024-05-08:
Anything I can do or provide to help debug this?

PR #1504 (2024-11-26): Adds a quirk to support OTA firmware update for Legrand devices
As mention in this [issue](https://github.com/home-assistant/core/issues/114375), Legrand OTA implementation is broken.

According to [this conversation](https://developer.legrand.com/forums/topic/connected-outlet-nlp-78-firmware/) on Legrand forum, it does not comply with [Zigbee specification](https://csa-iot.org/wp-content/uploads/2022/01/07-5123-08-Zigbee-Cluster-Library-1.pdf#page1051) regarding payload maximum data size rules.

Nevertheless, a workaround has been implemented in [Zigbee2MQTT](https://github.com/Koenkk/zigbee- [... cut]
closed 2024-11-26 by puddly