zigpy/zigpy #1279: Feature request: Support Bosch OTA

jclsn, 2023-11-08:
Bosch has released a new set of Zigbee devices last year and it would be great if the OTA updates for them were supported.

puddly, 2023-11-15:
OTA via file upload has always been supported for devices without HTTP OTA providers.

In the future, OTA will be centralized and independent of individual OTA providers. Until this is implemented, file-based OTA is the way to go.

Hedda, 2023-11-23:
@jclsn do you mean that you wish that zigpy should include an OTA provider for Bosch devices by default? If so then you must understand that for that to happen then Bosch first needs to provide unencrypted OTAU images and publish URLs publicly. See:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

That is, the reason that zigpy already provide OTA providers for some manufacturers by default is a combination of the manufacturer already publishes OTAU images and OTA URLs publicly and a developer have added download code for it:

https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates

For reference see these other feature requests for OTA providers where the manufacturer already publishes OTA URLs publicly:

https://github.com/zigpy/zigpy/issues/992

https://github.com/zigpy/zigpy/issues/1060

https://github.com/zigpy/zigpy/issues/988

https://github.com/zigpy/zigpy/issues/1163

jclsn, 2023-11-23:
@Hedda I am not sure. I just know that Z2M can already update the Bosch devices. I updated my TRV with its help now. 

@puddly No idea how file upload would work. I couldn't find a download portal for firmware updates for Bosch.

Hedda, 2023-11-23:
> I just know that Z2M can already update the Bosch devices. I updated my TRV with its help now.

I don't believe Zigbee2MQTT users get Bosch Zigbee OTAU firmware images directly from URLs that Bosch publishes publicly. or? 

As far as I know, the OTA providers with "official" sources that Zigbee2MQTT has more than zigpy has are [Lixee](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/lixee.ts) and [Ubisys](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/ubisys.ts). See: 

https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/src/lib/ota

The reason why zigbee-herdsman (library that Zigbee2MQTT depends on) provides OTA images from many other manufucturers than zigpy by default is that the Z2M community have just gone ahead to done the dirty work of collecting Zigbee OTA firmware images from various sources (such as example, Zigbee traffic sniffing and SoC firmware dumps) and then [cached, uploaded and shared those Zigbee OTAU images via an unofficial GitHub repository](https://github.com/Koenkk/zigbee-OTA) which then Zigbee2MQTT users then simply download their Zigbee OTAU images from (without most users knowing or caring where that OTA image actually got downloaded from). 

See this related feature request discussion here for more information about this indirect route method -> https://github.com/zigpy/zigpy/issues/535

(Bosch as example -> https://github.com/Koenkk/zigbee-OTA/tree/master/images/Bosch which Zigbee2MQTT's zigbee-herdsman library downloads indirectly via their unofficial zigbee-OTA repository on GitHub -> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/zigbeeOTA.ts )

As such the Zigbee2MQTT implementation have in this chosen to interpret their usage as "[fair use](https://en.wikipedia.org/wiki/Fair_use)" and claim rights to do so even if manufacturer's copyrights and end-user license agreement might technically state something else. I am no lawer so no idea if they would win in court with that argument however personally I think that the worse thing that would probably happen is that they sooner of later get sent a cease and desist letter that request them to remove 

PS: Note that there are several benefits for manufacturers to publish their OTAs publicly that should motivate some of them.

For example, you can let companies and their marketing department know that they should be made aware of Home Assistant's "**_Works with Home Assistant_**" partner program with its matching "**_Works via Zigbee with Home Assistant_**" compatibility badges and ask them as a company if they would not consider joining that program, and by extension also provide Zigbee OTA images publicly to show that they approve of Home Assistant's ZHA integration as a supported Zigbee gateway implementation for their devices:

https://www.home-assistant.io/blog/2022/07/12/partner-program/

There it says "**_Works with Home Assistant partners are expected to provide firmware updates via user-friendly services provided by Home Assistant_**" under **_Responsibilities_** in a section there called "**_Provide easy firmware updates_**" which explains that "**_Works with Home Assistant partners are expected to provide firmware updates via user-friendly services provided by Home Assistant_**".

https://partner.home-assistant.io/

"_**There are manufacturers that are creating products that integrate into Home Assistant using standards like Z-Wave, Zigbee, or Matter (soon). In these cases, the integration is maintained by the Home Assistant community and Nabu Casa. These companies can still become a member of the Works with Home Assistant program but are relieved from integration maintenance.**_"

"_**With Home Assistant we are always working on educating our users about preferring local control and open standards when acquiring new products. This is also reflected in the “Works with Home Assistant” badges.**_"

[image]

FYI, see example ThirdReality, a Zigbee device making company that joined their partner program + having a zigpy OTA provider:

https://www.home-assistant.io/blog/2022/10/13/third-reality-partner/

PPS: This relates to zigpy as it may imply some commitment from companies to help with OTA updates and ZHA Device Handlers.

https://www.home-assistant.io/integrations/zha#ota-firmware-updates

https://github.com/zigpy/zigpy/blob/dev/README.md#zigbee-device-ota-updates

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

https://www.home-assistant.io/integrations/zha#zha-exception-and-deviation-handling

https://github.com/zigpy/zha-device-handlers

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

jclsn, 2023-11-25:
Can't we make use of Z2M's OTA repo then? I mean they are already taking the risk, so we don't have to!

Hedda, 2023-11-26:
> Can't we make use of Z2M's OTA repo then? I mean they are already taking the risk, so we don't have to!

Again, refer to this related feature request discussion here for more information about this indirect route method -> https://github.com/zigpy/zigpy/issues/535

Summery; you as an individual user can choose to add whatever unofficial sourve you want in you Home Assistant ZHA installation instance, that is already possible (check out for example the ZHA-Toolkit), but simply adding such an unofficial repository without vetting each and every OTA image could possibly have the conseqvense of ruining the reputation of zigpy/zha if a bad firmware brick users devices. That is at least one of the reasons why unofficial sources from a third-party is not considered, however zigpy developers does mention there that they might consider creating thier own Zigbee OTA firmware repository where they could properly vet and test flashing each firmware image to such a device before adding it, however maintaining such a repository would require a lot of work, so until then users just have to get by with pointing to their own third-party repositroy of choice or adding firmware images manualy (which of course is also already possible).

Regardless, all that is a different and seperate request from getting Bosch OTA images directly from an official source published to the public by Bosch themselves, so suggest that you close this request unless you actually have URLs to such official sources.

closed 2023-12-06 by jclsn