zigpy/zigpy #1280: Ikea TRADFRI GU10 stuck at firmware 0x00010012

yiancar, 2023-11-08:
Hello all,
Please let me know if this is not the write place to ask for help.

It seems like my device: TRADFRIbulbGU10WS345lm is stuck in firmware 0x00010012.
According to http://fw.ota.homesmart.ikea.net/feed/version_info.json the latest version should be 2.3.087.
I think this is because the current firmware installed reports itself as current build ID: 65554 which is always going to be larger than the one in the json above.

Any ideas how to force the update? I am using HA with the Sky Connect dongle.

Cheers!

MattWestb, 2023-11-09:
What is the image number the device is requesting from the system for OTA update `
2.3.087 is one first / second gen device and the 1.0.012 is one third gen firmware so likely not the same device type.

yiancar, 2023-11-09:
Hello!
Thank you @MattWestb you are right.

`2023-11-08 22:36:24.843 DEBUG (MainThread) [zigpy.zcl] [0x9BC0:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRIbulbGU10WS345lm': field_control=1, manufacturer_id=4476, image_type=8709, current_file_version=65554, hardware_version=1, model='TRADFRIbulbGU10WS345lm'
2023-11-08 22:36:24.844 DEBUG (MainThread) [zigpy.zcl] [0x9BC0:1:0x0019] OTA image version: 65554, size: 236794. Update needed: False
`

Indeed the image type is 8709 and according to your matrix (https://github.com/zigpy/zigpy/discussions/660) 1.0.012 is the latest firmware available.

Sorry for the troubles

closed 2023-11-09 by yiancar