zigpy/zigpy #1292: Can't read zigbee attribute value for type array

claudegel, 2023-11-29:
Since about two week it is impossible to read an array type attribute. We get KeyError: 256
See the log below in this log the attribute 257 is of type Array and device scan stop there. it is the same with attribute read service which give the same keyError: 256
```
023-11-27 21:46:00.136 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Sending request: Read_Attributes(attribute_ids=[160, 161, 176, 257])
2023-11-27 21:46:00.137 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(timestamp=datetime.datetime(2023, 11, 28, 2, 46, 0, 137116, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x448B), dst_ep=1, source_route=None, extended_timeout=False, tsn=225, profile_id=260, cluster_id=65281, data=Serialized[b'\x04\x9c\x11\xe1\x00\xa0\x00\xa1\x00\xb0\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-11-27 21:46:00.137 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (28, 226, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, 260, 65281, 1, b'\x04\x9c\x11\xe1\x00\xa0\x00\xa1\x00\xb0\x00\x01\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2023-11-27 21:46:00.140 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 226]
2023-11-27 21:46:00.141 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 226]
2023-11-27 21:46:00.149 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.APSDE_DATA_CONFIRM|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 166>, 0]
2023-11-27 21:46:00.149 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.APSDE_DATA_CONFIRM|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 166>, 0]
2023-11-27 21:46:00.150 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2023-11-27 21:46:00.152 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 226, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-11-27 21:46:00.153 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 226: 00
2023-11-27 21:46:00.153 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0xe2 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, status: 0x00
2023-11-27 21:46:00.186 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-11-27 21:46:00.186 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-11-27 21:46:00.187 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2023-11-27 21:46:00.190 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[66, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x448b>, 1, 260, 65281, b'\x1c\x9c\x11\xe1\x01\xa0\x00\x00#\x00\x00\x00\x00\xa1\x00\x00#\x00\x00\x00\x00\xb0\x00\x00\x10\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x01\x00\x00\x01', 0, 175, 255, 166, 228, 200, 9, -43]
2023-11-27 21:46:00.190 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [66, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x448b>, 1, 260, 65281, b'\x1c\x9c\x11\xe1\x01\xa0\x00\x00#\x00\x00\x00\x00\xa1\x00\x00#\x00\x00\x00\x00\xb0\x00\x00\x10\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x01\x00\x00\x01', 0, 175, 255, 166, 228, 200, 9, -43]
2023-11-27 21:46:00.191 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 11, 28, 2, 46, 0, 191169, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x448B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=65281, data=Serialized[b'\x1c\x9c\x11\xe1\x01\xa0\x00\x00#\x00\x00\x00\x00\xa1\x00\x00#\x00\x00\x00\x00\xb0\x00\x00\x10\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x01\x00\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-43)
2023-11-27 21:46:00.192 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Received ZCL frame: b'\x1c\x9c\x11\xe1\x01\xa0\x00\x00#\x00\x00\x00\x00\xa1\x00\x00#\x00\x00\x00\x00\xb0\x00\x00\x10\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x01\x00\x00\x01'
2023-11-27 21:46:00.194 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4508, tsn=225, command_id=1, *direction=<Direction.Client_to_Server: 1>)
2023-11-27 21:46:00.198 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Decoded ZCL frame: SinopeManufacturerCluster:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x00A0, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint32_t, value=0)), ReadAttributeRecord(attrid=0x00A1, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint32_t, value=0)), ReadAttributeRecord(attrid=0x00B0, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.true: 1>)), ReadAttributeRecord(attrid=0x0101, status=<Status.SUCCESS: 0>, value=TypeValue(type=Array, value=Array(type=AnonymousLVList, value=[]))), ReadAttributeRecord(attrid=0x0100, status=<Status.SUCCESS: 0>, value=TypeValue(type=NoData, value=<zigpy.types.named.NoData object at 0x7f54526290>)), ReadAttributeRecord(attrid=0x0000, status=<Status.FAILURE: 1>), ReadAttributeRecord(attrid=0x0000, status=<Status.FAILURE: 1>)])
2023-11-27 21:46:00.201 DEBUG (MainThread) [zigpy_deconz.api] 'aps_data_indication' response from <DeconzAddress address_mode=AddressMode.NWK address=0x448b>, ep: 1, profile: 0x0104, cluster_id: 0xff01, data: b'1c9c11e101a000002300000000a100002300000000b000001001010100480a000000010000000001000001'
2023-11-27 21:46:00.202 DEBUG (MainThread) [custom_components.zha_toolkit] event_data {'zha_toolkit_version': 'v1.1.7', 'zigpy_version': '0.59.0', 'zigpy_rf_version': '0.21.1', 'ieee_org': 94:34:69:ff:fe:38:87:aa, 'ieee': '94:34:69:ff:fe:38:87:aa', 'command': 'scan_device', 'command_data': None, 'start_time': '2023-11-28T02:45:01.661145+00:00', 'errors': ['KeyError(256)'], 'params': {'endpoint_id': 1, 'dir': 0, 'tries': 1, 'expect_reply': True, 'args': [], 'event_success': 'my_read_success_trigger_event', 'event_fail': 'my_read_fail_trigger_event', 'event_done': 'my_read_done_trigger_event', 'read_before_write': True, 'read_after_write': True}, 'success': False}
2023-11-27 21:46:00.203 DEBUG (MainThread) [custom_components.zha_toolkit] Fire my_read_fail_trigger_event -> {'zha_toolkit_version': 'v1.1.7', 'zigpy_version': '0.59.0', 'zigpy_rf_version': '0.21.1', 'ieee_org': 94:34:69:ff:fe:38:87:aa, 'ieee': '94:34:69:ff:fe:38:87:aa', 'command': 'scan_device', 'command_data': None, 'start_time': '2023-11-28T02:45:01.661145+00:00', 'errors': ['KeyError(256)'], 'params': {'endpoint_id': 1, 'dir': 0, 'tries': 1, 'expect_reply': True, 'args': [], 'event_success': 'my_read_success_trigger_event', 'event_fail': 'my_read_fail_trigger_event', 'event_done': 'my_read_done_trigger_event', 'read_before_write': True, 'read_after_write': True}, 'success': False}
2023-11-27 21:46:00.203 DEBUG (MainThread) [custom_components.zha_toolkit] Fire my_read_done_trigger_event -> {'zha_toolkit_version': 'v1.1.7', 'zigpy_version': '0.59.0', 'zigpy_rf_version': '0.21.1', 'ieee_org': 94:34:69:ff:fe:38:87:aa, 'ieee': '94:34:69:ff:fe:38:87:aa', 'command': 'scan_device', 'command_data': None, 'start_time': '2023-11-28T02: [... cut ...]
  File "/config/custom_components/zha_toolkit/scan_device.py", line 266, in discover_attributes_extended
    success, failed = await read_attr(
                      ^^^^^^^^^^^^^^^^
  File "/config/custom_components/zha_toolkit/scan_device.py", line 29, in read_attr
    return await cluster.read_attributes(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 531, in read_attributes
    orig_attribute = orig_attributes[record.attrid]
                     ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 256
```

claudegel, 2023-11-29:
It was working fine about two weeks ago

puddly, 2023-11-29:
Can you include a PCAP?

---

You're reading the following attributes:
```Python
Read_Attributes(attribute_ids=[0x00A0, 0x00A1, 0x00B0, 0x0101])
```

The device responds with:
```Python
SinopeManufacturerCluster:Read_Attributes_rsp(status_records=[
	# OK: 0x00A0
	ReadAttributeRecord(attrid=0x00A0, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint32_t, value=0)),
	# OK: 0x00A1
	ReadAttributeRecord(attrid=0x00A1, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint32_t, value=0)),
	# OK: 0x00B0
	ReadAttributeRecord(attrid=0x00B0, status=<Status.SUCCESS: 0>, value=TypeValue(type=Bool, value=<Bool.true: 1>)),
	# OK: 0x0101
	ReadAttributeRecord(attrid=0x0101, status=<Status.SUCCESS: 0>, value=TypeValue(type=Array, value=Array(type=AnonymousLVList, value=[]))),

	# Where is `0x0100` coming from? It was not requested.
	ReadAttributeRecord(attrid=0x0100, status=<Status.SUCCESS: 0>, value=TypeValue(type=NoData, value=<zigpy.types.named.NoData object at 0x7f54526290>)),
	# Same with this
	ReadAttributeRecord(attrid=0x0000, status=<Status.FAILURE: 1>),
	# And this
	ReadAttributeRecord(attrid=0x0000, status=<Status.FAILURE: 1>),
])
```

PR #1270 fixed parsing for the `Array` type. If I revert the PR, the attribute read response from your device fails to parse entirely.

claudegel, 2023-11-29:
What is a PCAP
For 0x0100 I don't know as it is not part of the cluster I'm scanning.

claudegel, 2023-11-29:
The next attribute in that cluster is 0x012A

claudegel, 2023-11-29:
I've tested with another device I was able to read Aray type attribute and it is working in ZHA for reading just that attribute.
So I've tried to read the attribute 0x0101 in zha and it fail
```
2023-11-28 23:22:17.450 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Sending request: Read_Attributes(attribute_ids=[257])
2023-11-28 23:22:17.451 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(timestamp=datetime.datetime(2023, 11, 29, 4, 22, 17, 451870, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x448B), dst_ep=1, source_route=None, extended_timeout=False, tsn=13, profile_id=260, cluster_id=65281, data=Serialized[b'\x04\x9c\x11\r\x00\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-11-28 23:22:17.452 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (22, 14, <DeconzSendDataFlags.NONE: 0>, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, 260, 65281, 1, b'\x04\x9c\x11\r\x00\x01\x01', <DeconzTransmitOptions.USE_NWK_KEY_SECURITY: 2>, 0)
2023-11-28 23:22:17.455 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_request[2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 14]
2023-11-28 23:22:17.455 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 14]
2023-11-28 23:22:17.463 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.APSDE_DATA_CONFIRM|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 166>, 0]
2023-11-28 23:22:17.463 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.APSDE_DATA_CONFIRM|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 166>, 0]
2023-11-28 23:22:17.464 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2023-11-28 23:22:17.467 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_confirm[12, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 14, <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, 1, <TXStatus.SUCCESS: 0>, 0, 0, 0, 0]
2023-11-28 23:22:17.467 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 14: 00
2023-11-28 23:22:17.467 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x0e 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=AddressMode.NWK address=0x448b endpoint=1>, status: 0x00
2023-11-28 23:22:17.531 DEBUG (MainThread) [zigpy_deconz.api] Received command device_state_changed[<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-11-28 23:22:17.532 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.APSDE_DATA_INDICATION|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|130: 170>, 0]
2023-11-28 23:22:17.534 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_indication (1, <DataIndicationFlags.Always_Use_NWK_Source_Addr: 1>)
2023-11-28 23:22:17.537 DEBUG (MainThread) [zigpy_deconz.api] Received command aps_data_indication[45, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x448b>, 1, 260, 65281, b'\x1c\x9c\x11\r\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01', 0, 175, 255, 206, 86, 205, 9, -48]
2023-11-28 23:22:17.538 DEBUG (MainThread) [zigpy_deconz.api] APS data indication response: [45, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, <DeconzAddress address_mode=AddressMode.NWK address=0x0000>, 1, <DeconzAddress address_mode=AddressMode.NWK address=0x448b>, 1, 260, 65281, b'\x1c\x9c\x11\r\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01', 0, 175, 255, 206, 86, 205, 9, -48]
2023-11-28 23:22:17.538 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 11, 29, 4, 22, 17, 538592, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x448B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=65281, data=Serialized[b'\x1c\x9c\x11\r\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-48)
2023-11-28 23:22:17.540 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Received ZCL frame: b'\x1c\x9c\x11\r\x01\x01\x01\x00H\n\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01'
2023-11-28 23:22:17.542 DEBUG (MainThread) [zigpy.zcl] [0x448B:1:0xff01] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4508, tsn=13, command_id=1, *direction=<Direction.Client_to_Server: 1>)
2023-11-28 23:22:17.545 DEBUG (MainThread) [zigpy.device] [0x448b] Failed to parse packet ZigbeePacket(timestamp=datetime.datetime(2023, 11, 29, 4, 22, 17, 538592, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x448B), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_ [... cut ...]
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 377, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 314, in request
    return await req.result
           ^^^^^^^^^^^^^^^^
zigpy.exceptions.ParsingError
```

puddly, 2023-11-29:
> What is a PCAP

A PCAP is a packet capture. If you have a Silicon Labs radio like a SkyConnect or a HUSBZB-1, you can create a packet capture:

```bash
pip install bellows
bellows --baudrate 115200 -d /dev/ttyUSB1 dump -c 20 -w capture.pcap
```

Change `20` to your Zigbee network's channel and post your Zigbee network key so that it can be decrypted.

claudegel, 2023-11-29:
I have a conbee II but I also have a zigbee sniffer. I can see all traffic when scanning a device

claudegel, 2023-11-29:
My sniffer produce a pcapng file. So I'll sniff during the device scan and load the file with the network key. I'm away for three days but will do this when back on friday. Thank you

claudegel, 2023-12-02:
here is my PCAP. I have two. one for when I try to read array type attribute 0x0101 and 0x012c, cluster 0xff01 in ZHA and one when I'm doing a zha_toolkit.scan_device. both fail.
device ieee is 94:34:69:ff:fe:38:87:aa
key bfaa6457733b88185b173089636238f6
channel 15
[array 0101-012c.pcapng.zip](https://github.com/zigpy/zigpy/files/13536700/array.0101-012c.pcapng.zip)

MattWestb, 2023-12-02:
Dont trust wireschark for 100% then some Zigbee things in not (complete) implanted or not working OK like some ZLL frames. Do one check with frame raw data i think is the best.

puddly, 2023-12-02:
Zigpy and Wireshark agree here, unfortunately, and neither parses the data correctly. I think it's a device issue.

What data is represented here? It looks like the `Array` type is being misused by the manufacturer. Here are the three malformed ZCL attribute read responses:

```Perl
# ZCL header   Attribute report up until Array type                Attribute report after Array type
1c9c113f01     010100                                              480a000000010000000000000001
1c9c115c01     2c0100                                              484820000a98271e0000008001fe
1c9c110e01     a000002300000000a100002300000000b000001001010100    480a000000010000000000000001
```

claudegel, 2023-12-02:
I've no idea of the use of those attributes. That what I try to find. This device is a water heater tank controller. I also suspect a manufacturer bug as there are other devices where I can read array value.
Is there a way to make zha_toolkit.scan_device finish the scan because it crash when it reach that attribute

closed 2024-06-06 by github-actions[bot]