zigpy/zigpy #1299: EZSP fail to sendUnicast

brodock, 2023-12-19:
Logger: zigpy.zcl
Source: runner.py:188
First occurred: 06:35:51 (63 occurrences)
Last logged: 18:37:18

```
[0x2B8B:1:0x0702] Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply return await self._endpoint.reply( ^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply return await self.device.reply( ^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply return await self.request( ^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request await self._application.request( File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request await self.send_packet( File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet status, _ = await self._ezsp.sendUnicast( ^^^^^^^^^^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'sendUnicast'
[0x2B8B:1:0x0b04] Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply return await self._endpoint.reply( ^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply return await self.device.reply( ^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply return await self.request( ^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request await self._application.request( File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request await self.send_packet( File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet status, _ = await self._ezsp.sendUnicast( ^^^^^^^^^^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'sendUnicast'
[0xC577:1:0x0702] Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply return await self._endpoint.reply( ^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply return await self.device.reply( ^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply return await self.request( ^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request await self._application.request( File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request await self.send_packet( File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet status, _ = await self._ezsp.sendUnicast( ^^^^^^^^^^^^^^^^^^^^^^ AttributeError: 'NoneType' object has no attribute 'sendUnicast'
[0xC652:1:0x0020] AssertionError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 377, in request return await self._endpoint.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 298, in request with self._pending.new(sequence) as req: ^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 297, in new raise ControllerException(f"duplicate {sequence} TSN") from AssertionError zigpy.exceptions.ControllerException: duplicate 24 TSN The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/general.py", line 504, in check_in_response await self.checkin_response(True, self.CHECKIN_FAST_POLL_TIMEOUT, tsn=tsn) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper with wrap_zigpy_exceptions(): File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__ self.gen.throw(typ, value, traceback) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions raise HomeAssistantError(message) from exc homeassistant.exceptions.HomeAssistantError: Failed to send request: duplicate 24 TSN
[0xC652:1:0x0020] Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request await self._application.request( File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request await self.send_packet( File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 825, in send_packet raise zigpy.exceptions.DeliveryError( zigpy.exceptions.DeliveryError: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102> The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/general.py", line 504, in check_in_response await self.checkin_response(True, self.CHECKIN_FAST_POLL_TIMEOUT, tsn=tsn) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper with wrap_zigpy_exceptions(): File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__ self.gen.throw(typ, value, traceback) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions raise HomeAssistantError(message) from exc homeassistant.exceptions.HomeAssistantError: Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
```

puddly, 2023-12-19:
Please include a complete debug log

brodock, 2023-12-19:
```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.12.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "petkit": {
      "version": "0.1.10",
      "requirements": [
        "petkitaio==0.1.10",
        "tzlocal>=4.2"
      ]
    },
    "afvalbeheer": {
      "version": "5.2.11",
      "requirements": [
        "rsa",
        "pycryptodome"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly",
      "@TheJulianJES"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp",
      "universal_silabs_flasher"
    ],
[... 477 lines cut ...]
      "15": 88.70042934643088,
      "16": 97.04162534718327,
      "17": 96.64469941013013,
      "18": 70.89933442360993,
      "19": 12.244260188723507,
      "20": 70.89933442360993,
      "21": 39.90320178295578,
      "22": 55.9836862725909,
      "23": 70.89933442360993,
      "24": 73.50699819621309,
      "25": 68.14622793558128,
      "26": 49.512515447068886
    },
    "versions": {
      "bellows": "0.37.3",
      "zigpy": "0.60.1",
      "zigpy_deconz": "0.22.2",
      "zigpy_xbee": "0.20.1",
      "zigpy_znp": "0.12.0",
      "zigpy_zigate": "0.12.0",
      "zhaquirks": "0.0.107"
    }
  }
}
```

brodock, 2023-12-19:
```
2023-12-19 18:37:06.497 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=32, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=19), 156, -61, 0xc652, 255, 255, b'\t\x18\x00']
2023-12-19 18:37:06.497 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=32, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=19), 156, -61, 0xc652, 255, 255, b'\t\x18\x00']
2023-12-19 18:37:06.497 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 19, 17, 37, 6, 497871, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC652), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=19, profile_id=260, cluster_id=32, data=Serialized[b'\t\x18\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=156, rssi=-61)
2023-12-19 18:37:06.498 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Received ZCL frame: b'\t\x18\x00'
2023-12-19 18:37:06.499 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=24, command_id=0, *direction=<Direction.Client_to_Server: 1>)
2023-12-19 18:37:06.500 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Decoded ZCL frame: PollControl:checkin()
2023-12-19 18:37:06.500 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Received command 0x00 (TSN 24): checkin()
2023-12-19 18:37:06.501 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] No explicit handler for cluster command 0x00: checkin()
2023-12-19 18:37:06.501 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xC652:1:0x0020]: Received 24 tsn command 'checkin': checkin()
2023-12-19 18:37:06.503 DEBUG (MainThread) [zigpy.util] Tries remaining: 3
2023-12-19 18:37:06.503 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=24, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-12-19 18:37:06.504 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Sending request: checkin_response(start_fast_polling=<Bool.true: 1>, fast_poll_timeout=8)
2023-12-19 18:37:06.505 DEBUG (MainThread) [zigpy.device] [0xc652] Extending timeout for 0x18 request
2023-12-19 18:37:06.505 DEBUG (MainThread) [zigpy.util] Duplicate 24 TSN
2023-12-19 18:37:06.571 DEBUG (MainThread) [bellows.uart] Data frame: b'3262b1a96b2a134fa6944a25aa559249984e27bf12ce67e2857e'
2023-12-19 18:37:06.572 DEBUG (MainThread) [bellows.uart] Sending: b'8430fc7e'
2023-12-19 18:37:06.572 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65533, EmberApsFrame(profileId=0, clusterId=0, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=20), 255, <EmberStatus.SUCCESS: 0>, b'']
2023-12-19 18:37:06.573 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_BROADCAST: 6>, 65533, EmberApsFrame(profileId=0, clusterId=0, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_SOURCE_EUI64: 1024>, groupId=0, sequence=20), 255, <EmberStatus.SUCCESS: 0>, b'']
2023-12-19 18:37:06.573 DEBUG (MainThread) [bellows.zigbee.application] Unexpected message send notification tag: 255
2023-12-19 18:37:06.606 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2023-12-19 18:37:06.606 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=24, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-12-19 18:37:06.607 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Sending request: checkin_response(start_fast_polling=<Bool.true: 1>, fast_poll_timeout=8)
2023-12-19 18:37:06.608 DEBUG (MainThread) [zigpy.device] [0xc652] Extending timeout for 0x18 request
2023-12-19 18:37:06.608 DEBUG (MainThread) [zigpy.util] Duplicate 24 TSN
2023-12-19 18:37:06.709 DEBUG (MainThread) [zigpy.util] Tries remaining: 1
2023-12-19 18:37:06.710 DEBUG (MainThread) [zigpy.zcl] [0xC652:1:0x0020] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=24, command_id=0, *direction=<Direction.Server_to_ [... cut ...]
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper
    with wrap_zigpy_exceptions():
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions
    raise HomeAssistantError(message) from exc
homeassistant.exceptions.HomeAssistantError: Failed to send request: duplicate 24 TSN
```

(this is from the last exception)

brodock, 2023-12-19:
this is from the sendUnicast error:

```
2023-12-19 10:50:26.416 ERROR (MainThread) [zigpy.zcl] [0xC577:1:0x0702] Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply
    return await self._endpoint.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply
    return await self.device.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request
    await self._application.request(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request
    await self.send_packet(
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet
    status, _ = await self._ezsp.sendUnicast(
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendUnicast'

2023-12-19 10:50:26.426 ERROR (MainThread) [zigpy.zcl] [0x4986:1:0x0019] Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/clusters/general.py", line 2187, in _handle_cluster_request
    await self._handle_query_next_image(
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/clusters/general.py", line 2259, in _handle_query_next_image
    await self.query_next_image_response(
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply
    return await self._endpoint.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply
    return await self.device.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request
    await self._application.request(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request
    await self.send_packet(
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet
    status, _ = await self._ezsp.sendUnicast(
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendUnicast'

2023-12-19 10:50:26.431 ERROR (MainThread) [zigpy.zcl] [0x6399:1:0x0300] Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply
    return await self._endpoint.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply
    return await self.device.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request
    await self._application.request(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request
    await self.send_packet(
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet
    status, _ = await self._ezsp.sendUnicast(
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendUnicast'

2023-12-19 10:50:26.439 ERROR (MainThread) [zigpy.zcl] [0x4986:1:0x0019] Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/clusters/general.py", line 2187, in _handle_cluster_request
    await self._handle_query_next_image(
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/clusters/general.py", line 2259, in _handle_query_next_image
    await self.query_next_image_response(
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 411, in reply
    return await self._endpoint.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 278, in reply
    return await self.device.reply(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 474, in reply
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 299, in request
    await self._application.request(
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 829, in request
    await self.send_packet(
  File "/usr/local/lib/python3.11/site-packages/bellows/zigbee/application.py", line 762, in send_packet
    status, _ = await self._ezsp.sendUnicast(
                      ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendUnicast'
```

puddly, 2023-12-19:
Can you enable ZHA debug logging, let it run for about 15 minutes (or however long it takes to reproduce this error), and then upload the whole log?

brodock, 2023-12-19:
[home-assistant_zha_2023-12-19T17-43-45.088Z.log](https://github.com/zigpy/zigpy/files/13718844/home-assistant_zha_2023-12-19T17-43-45.088Z.log)

puddly, 2023-12-19:
In this log, are you trying to reconfigure the ZHA integration through a config flow?

brodock, 2023-12-19:
@puddly yes, I was trying to update the ITEAD Dongle-E firmware to see if it fixes the issues I'm having with an Aqara thermostat. So I had to restore the network a few times after upgrading the zigbee dongle, and was trying to reconfigure/re-add the thermostat

MattWestb, 2023-12-20:
@puddly I have getting this for longer times on both production (NCP) and test systems (RCP).
After updating to latest HA in the night i was getting it on both.
I mailing you the long from my RCP test system then its not very large as the production one.

puddly, 2023-12-20:
Thank you, @MattWestb. It looks like there is a load issue, the NCP resets because it did not receive ACKs in time (20 messages in in 77ms, they probably got stuck in a buffer somewhere):

```Python
2023-12-18 19:43:42.983 DEBUG (MainThread) [bellows.uart] Data frame: b'4599b1a96b2a156347904b23aa5493099c4e278379a86759817e'
2023-12-18 19:43:42.984 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2023-12-18 19:43:42.986 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 7889, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=40), 148, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:42.987 DEBUG (MainThread) [bellows.uart] Data frame: b'5599b1a96b2a158361904b25a95493099c4e278278a867d2857e'
2023-12-18 19:43:42.988 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2023-12-18 19:43:42.992 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 14385, EmberApsFrame(profileId=260, clusterId=768, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=41), 149, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:42.993 DEBUG (MainThread) [bellows.uart] Data frame: b'6599b1a96b2a1580d0904b25a95493099c4e27817ba86781ae7e'
2023-12-18 19:43:42.994 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2023-12-18 19:43:42.997 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 35122, EmberApsFrame(profileId=260, clusterId=768, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=42), 150, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:42.999 DEBUG (MainThread) [bellows.uart] Data frame: b'7599b1a96b2a1580d0904b23aa5493099c4e27807aa86783987e'
2023-12-18 19:43:42.999 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2023-12-18 19:43:43.003 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 35122, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=43), 151, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:43.004 DEBUG (MainThread) [bellows.uart] Data frame: b'0599b1a96b2a158361904b2daa5493099c4e278775a867be6d7e'
2023-12-18 19:43:43.004 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2023-12-18 19:43:43.008 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 14385, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=44), 152, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:43.009 DEBUG (MainThread) [bellows.uart] Data frame: b'4d99b1a96b2a156347904b23aa5493099c4e278379a86741ba7e'
2023-12-18 19:43:43.010 DEBUG (MainThread) [bellows.uart] Sending: b'8520dd7e'
2023-12-18 19:43:43.014 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 7889, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=40), 148, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:43.015 DEBUG (MainThread) [bellows.zigbee.application] Invalid state on future for message tag 148 - probably duplicate response: invalid state
2023-12-18 19:43:43.015 DEBUG (MainThread) [bellows.uart] Data frame: b'5d99b1a96b2a158361904b25a95493099c4e278278a867cabe7e'
2023-12-18 19:43:43.015 DEBUG (MainThread) [bellows.uart] Sending: b'8610be7e'
2023-12-18 19:43:43.020 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 14385, EmberApsFrame(profileId=260, clusterId=768, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=41), 149, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:43.021 DEBUG (MainThread) [bellows.zigbee.application] Invalid state on future for message tag 149 - probably duplicate response: invalid state
2023-12-18 19:43:43.022 DEBUG (MainThread) [bellows.uart] Data frame: b'6d99b1a96b2a1580d0904b25a95493099c4e27817ba86799957e'
2023-12-18 19:43:43.022 DEBUG (MainThread) [bellows.uart] Sending: b'87009f7e'
2023-12-18 19:43:43.027 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 35122, EmberApsFrame(profileId=260, clusterId=768, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=42), 150, <EmberStatus.DELIVERY_FAILED: 102>, b'']
2023-12-18 19:43:43.028 DEBUG (MainThread) [bellows.zigbee.application] Invalid state on future for message tag 150 - probably duplicate response: invalid state
2023-12-18 19:43:43.028 DEBUG (MainThread) [bellows.uart] Data frame: b'7d99b1a96b2a1580d0904b23aa5493099c4e27807aa8679ba37e'
2023-12-18 19:43:43.028 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
[... cut ...]
2023-12-18 19:43:43.060 DEBUG (MainThread) [bellows.zigbee.application] Received _reset_controller_application frame with (<NcpResetCode.ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT: 81>,)
2023-12-18 19:43:43.060 DEBUG (MainThread) [zigpy.application] Connection to the radio has been lost: <NcpResetCode.ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT: 81>
2023-12-18 19:43:43.060 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] Connection to the radio was lost: <NcpResetCode.ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT: 81>
```

What version of Home Assistant OS is this happening with? Does it still happen if you downgrade *just* the OS, nothing else?

MattWestb, 2023-12-20:
It was 2023.12.0 that was updating to 12.3 and latest addon with RCP firmware and 25 devices and Armbina 64 bit.
But the production system is getting it also and its running Firmware: 6.10.7.0 build 435 (multi rail with GP) with 79 devices and Raspbina 64 bit.

MattWestb, 2023-12-21:
Both systems is also getting this strange errors:
```
Logger: zigpy.zcl
Source: runner.py:188
First occurred: 06:00:14 (1 occurrences)
Last logged: 06:00:14

[0x39AC:1:0x0020] AssertionError The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 377, in request return await self._endpoint.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/endpoint.py", line 253, in request return await self.device.request( ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/device.py", line 298, in request with self._pending.new(sequence) as req: ^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/zigpy/util.py", line 297, in new raise ControllerException(f"duplicate {sequence} TSN") from AssertionError zigpy.exceptions.ControllerException: duplicate 74 TSN The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/general.py", line 504, in check_in_response await self.checkin_response(True, self.CHECKIN_FAST_POLL_TIMEOUT, tsn=tsn) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper with wrap_zigpy_exceptions(): File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__ self.gen.throw(typ, value, traceback) File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 75, in wrap_zigpy_exceptions raise HomeAssistantError(message) from exc homeassistant.exceptions.HomeAssistantError: Failed to send request: duplicate 74 TSN
```
But i think its only to sleepers that is doing checkins and normally its OTA requests but i think the TSN is the problem here.

brodock, 2024-01-02:
I've disabled OTA and it didn't improve on anything... I went back to the ITead Dongle-P and I was able to get the thermostat enrolled again.

I was considering whether to flash the Dongle-E to enable the MultiProtocol approach and if that would provide a different result or not.

MattWestb, 2024-01-02:
I is getting it on NCP (production ) and RCP (Test networks) system then having over 20 end devices but i think our **puddly** is on the track fixing  the problems.

brodock, 2024-07-01:
do we have any update on this?

closed 2025-05-31 by github-actions[bot]