zigpy/zigpy #1301: How to parse/access an incoming ZCL frame?

garrethcain, 2023-12-21:
Hopefully this is the right place, if not please direct me.

I'm trying to implement a super lightweight Zigbee implementation to send and receive messages from an MQTT broker and have chosen Zigpy and Zigpy-Znp.

I've reached the stage where I need to parse the message and publish it to my MQTT broker, I can see the following in the logs;

```
zigpy.zcl[33287] DEBUG [0x5EA2:1:0x0006] Received ZCL frame: b'\x11,\x01'
zigpy.zcl[33287] DEBUG [0x5EA2:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=44, command_id=1, *direction=<Direction.Server_to_Client: 0>)
zigpy.zcl[33287] DEBUG [0x5EA2:1:0x0006] Decoded ZCL frame: OnOff:on()
zigpy.zcl[33287] DEBUG [0x5EA2:1:0x0006] Received command 0x01 (TSN 44): on()
zigpy.zcl[33287] DEBUG [0x5EA2:1:0x0006] No explicit handler for cluster command 0x01: on()
```

Essentially I want access to the "Decoded ZCL frame" so I can pass it on to where it needs to go as it seem to have what I want. I have implemented my own `handle_message` method, which is receiving data.

I just want to know how to access the "decoded" ZCL frame or how to, as is implied in the log above, implement my own `explicit handler for the cluster command`.

Any help supper appreciated and many thanks in advance.

garrethcain, 2024-01-02:
Even just a link to a how-to would be appreciated.

puddly, 2024-01-02:
To access incoming ZCL frames, you would have to attach a listener to the cluster object on the device you're interested in receiving frames:

https://github.com/zigpy/zigpy/blob/e54dab0fb0755d421fe610825c8c72b9542fbfca/zigpy/zcl/__init__.py#L427

Something like this:

```python
class MyListener:
    def __init__(self, endpoint):
        self.endpoint = endpoint
        self.device = endpoint.device

    def cluster_command(self, tsn, command_id, cmd):
        print(f"Device {self.device} received command {cmd} on endpoint {self.endpoint}")

ep = dev.endpoints[1]
ep.on_off.add_listener(MyListener(ep))
```

Alternatively, you can create a listener for the device:

```Python
def my_callback(cmd_hdr, cmd):
    print("Received command {cmd_hdr} {cmd}")

with app._callback_for_response(src=dev, filters=[], callback=my_callback):
    await main()

# Callback listener detaches once you exit the context
```

garrethcain, 2024-01-02:
Thanks for the help @puddly 

Using your first example, would I add the listener when the `device_initialized` gets been called?

Eg.
```
def device_initialized(self, device, *, new=True):
    if device.some_var == "some_value":
        ep = dev.endpoints[1]
        ep.on_off.add_listener(MyListener(ep))
```

This would allow me to add different/bespoke listeners to specific devices, I'm hoping.

closed 2024-07-11 by github-actions[bot]