zigpy/zigpy #1305: OTA update fails with Wrong magic number for OTA Image and other errors

frli4797, 2023-12-29:
On Homeassistant as follows
Core 2023.12.4
Supervisor 2023.12.0
Operating System 11.2
Hardware: Homeassistant Yellow

Relevant config:
`# ZHA Configuration
zha:
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Tr..dfri devices
      # ikea_update_url: http://fw.ota.homesmart.ikea.net/feed/version_info.json
      # http://fw.test.ota.homesmart.ikea.net/feed/version_info.json
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      inovelli_provider: true                    # Auto update INOVELLI devices
      sonoff_provider: true
      thirdreality_provider: true
      otau_directory: /config/ota                # Utilize .ota files to update everything else
`

When triggering an OTA update using ZHA Toolkit 
`service: zha_toolkit.ota_notify
data:
  ieee: select.mooon_ab_dimmer_switch_zb3_0_startuponoffselect
  download: true
  tries: 3`

I get a LOT of errors and warnings in my logfile. (Limited myself to a few rows. I get a few hundred of these when triggering an OTA update.)
`38322:2023-12-29 16:22:52.981 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f89bc0e10>> with args (ImageKey(manufacturer_id=4489, image_type=59),): ValueError('Wrong magic number for OTA Image: 1931616379')
38323:2023-12-29 16:22:52.982 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4364, image_type=5), 'https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=5&Version=1.2.5.16', LedvanceImage(manufacturer_id=4364, image_type=5, version=16909584, image_size=140550, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=5&Version=1.2.5.16')
38324:2023-12-29 16:22:53.102 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f89bc0e10>> with args (ImageKey(manufacturer_id=4364, image_type=5),): ValueError('Wrong magic number for OTA Image: 1931616379')
38325:2023-12-29 16:22:53.103 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4364, image_type=7), 'https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=7&Version=1.2.5.16', LedvanceImage(manufacturer_id=4364, image_type=7, version=16909584, image_size=123884, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=7&Version=1.2.5.16')
38326:2023-12-29 16:22:53.241 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f89bc0e10>> with args (ImageKey(manufacturer_id=4364, image_type=7),): ValueError('Wrong magic number for OTA Image: 1931616379')
38327:2023-12-29 16:22:53.242 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4489, image_type=210), 'https://api.update.ledva
nce.com/v1/zigbee/firmwares/download?Company=4489&Product=210&Version=3.32.54.96', LedvanceImage(manufacturer_id=4489, image_type=210, version=52442720, image_size=188416
, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4489&Product=210&Version=3.32.54.96')
38335:2023-12-29 16:22:53.405 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f89b
c0e10>> with args (ImageKey(manufacturer_id=4489, image_type=210),): ValueError('Wrong magic number for OTA Image: 1931616379')
38336:2023-12-29 16:22:53.406 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4489, image_type=111), 'https://api.update.ledva
nce.com/v1/zigbee/firmwares/download?Company=4489&Product=111&Version=2.5.101.80', LedvanceImage(manufacturer_id=4489, image_type=111, version=33908048, image_size=190758
, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4489&Product=111&Version=2.5.101.80')
38206:2023-12-29 16:22:44.836 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=8706), None, IKEAImage(image_ty
pe=8706, binary_url='https://fw.ota.homesmart.ikea.com/files/tradfri-bulb-ws-1000lm_release_prod_v587814449_2b13625e-17a8-40d9-bad3-47a9a96ab002.ota', sha3_256_sum='9318a
9cfbb6aef7bc50586d9b16a3a39b470fa6c88e4fd05f5f77834f9ceeceb')
38207:2023-12-29 16:22:44.837 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=4365), None, IKEAImage(image_ty
pe=4365, binary_url='https://fw.ota.homesmart.ikea.com/files/zingo-lds-plugin-unit_release_prod_v65538_b29c3768-a102-4414-a84d-405c99654e74.ota', sha3_256_sum='2512a74ba2
0f445494e75f9f05c4debf051cff14496af2abad7a2074d635b6d3')
38208:2023-12-29 16:22:44.837 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=4364), None, IKEAImage(image_ty
pe=4364, binary_url='https://fw.ota.homesmart.ikea.com/files/zingo-lds-starkvind_release_prod_v69633_2044addf-0a35-4845-b851-74df04ab3a76.ota', sha3_256_sum='80c8b788ff09
4b3943d8b74bebefba8b73df2c526602bca0da126e777dca3f73')
38209:2023-12-29 16:22:44.837 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=4549), None, IKEAImage(image_ty
pe=4549, binary_url='https://fw.ota.homesmart.ikea.com/files/tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota', sha3_256_sum='499
f08c42fdcd5019c8d48b3a17c2f3406cdc97b4bfb41db064a1487e695596c')
38210:2023-12-29 16:22:44.837 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=16644), None, IKEAImage(image_t
ype=16644, binary_url='https://fw.ota.homesmart.ikea.com/files/zingo-sigma-driver-silverglans-ww_release_prod_v65569_73466dc4-1320-4242-add6-717432538a77.ota', sha3_256_s
um='f546464079b5e3c03862acfa03dcba693e03155681cd07c67c3f0b12558d7987')`

puddly, 2023-12-29:
Can you post the full error? Your lines are cut off at 170 characters.

frli4797, 2023-12-30:
Ah, sorry.  Attaching full log file as well. There's a lot of noise in the file though, but I have a hard time sorting out on what is causing which problem. 

```2023-12-29 17:41:29.519 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=16898), None, IKEAImage(image_type=16898, binary_url='https://fw.ota.homesmart.ikea.com/files/tradfri-driver-hp_release_prod_v587757105_b42c57bf-cb9c-478c-8327-5812a3286a64.ota', sha3_256_sum='c60d3ced6fc1a5c0d32f8044f28f19c4e324c2b105a39d1666871036ced68222')
2023-12-29 17:41:29.574 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=4557), None, IKEAImage(image_type=4557, binary_url='https://fw.ota.homesmart.ikea.com/files/rodret-dimmer-soc_release_prod_v16777287_9812b73c-b02e-4678-b737-d21251a34fd2.ota', sha3_256_sum='896edfb0a9d8314fb49d44fb11dc91fb5bb55e2ee1f793d53189cb13f884e13c')
2023-12-29 17:41:29.659 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=4552), None, IKEAImage(image_type=4552, binary_url='https://fw.ota.homesmart.ikea.com/files/tradfri-motion-sensor2_release_prod_v604241925_8afa2f7c-19c3-4ddf-a96c-233714179022.ota', sha3_256_sum='1b5fbea79c5b41864352a938a90ad25d9a0118054bf1cdc0314ef9636a60143a')
2023-12-29 17:41:29.716 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4476, image_type=8450), None, IKEAImage(image_type=8450, binary_url='https://fw.ota.homesmart.ikea.com/files/zingo-lds-bulb-hwpwm-ww_release_prod_v69638_173a15e1-8fd2-415c-812d-3de3724ce512.ota', sha3_256_sum='498a9a05dfb2c03c1dbd5edf4943f8abc2b2ec96322845cefc4fc563098e48ef')
```

```
2023-12-29 17:41:42.311 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f71adf010>> with args (ImageKey(manufacturer_id=4364, image_type=6),): ValueError('Wrong magic number for OTA Image: 1931616379')
2023-12-29 17:41:42.312 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4364, image_type=20), 'https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=20&Version=1.2.5.16', LedvanceImage(manufacturer_id=4364, image_type=20, version=16909584, image_size=132928, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4364&Product=20&Version=1.2.5.16')
2023-12-29 17:41:42.431 WARNING (MainThread) [zigpy.util] Error calling listener <bound method Basic.get_image of <zigpy.ota.provider.ThirdReality object at 0x7f71adf010>> with args (ImageKey(manufacturer_id=4364, image_type=20),): ValueError('Wrong magic number for OTA Image: 1931616379')
2023-12-29 17:41:42.432 ERROR (MainThread) [custom_components.zha_toolkit.ota] Try getting ImageKey(manufacturer_id=4489, image_type=33), 'https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4489&Product=33&Version=0.16.36.40', LedvanceImage(manufacturer_id=4489, image_type=33, version=1057832, image_size=170776, url='https://api.update.ledvance.com/v1/zigbee/firmwares/download?Company=4489&Product=33&Version=0.16.36.40')
```
[home-assistant.log](https://github.com/zigpy/zigpy/files/13796720/home-assistant.log)

puddly, 2023-12-30:
This is a problem with ZHA toolkit. It's logging under the `ERROR` log level during normal operation and also spamming the Ledvance API with so many requests that you are getting rate limited:

```python
2023-12-29 17:42:06.982 WARNING (MainThread) [zigpy.ota.provider] Ledvance: Couldn't download 'https://api.update.ledvance.com/v1/zigbee/firmwares': 429/Too Many Requests
```

CC @mdeweerd

closed 2023-12-30 by puddly

frli4797, 2023-12-30:
@mdeweerd , FYI. I'll re-report on ZHA Toolkit for traceability.