zigpy/zigpy #1313: [Feature Request] Support of Aeotec Pico (Duo) Switch

CHTHSCH, 2024-01-01:
I would appreciate, if these two new products would be supported by the Zigbee Integration.

[ZGA002 - Aeotec Pico Switch](https://aeotec.com/products/Aeotec-Smartthings-Pico-Switch/)
[ZGA003 - Aeotec Pico Duo Switch](https://aeotec.com/products/Aeotec-Smartthings-Pico-Switch/)

tube0013, 2024-01-01:
https://aeotec.freshdesk.com/support/solutions/articles/6000266747-setup-pico-switch-with-home-assistant

CHTHSCH, 2024-01-01:
Thanks

In that case the list here [https://zigbee.blakadder.com/index.html](url) is outdated. Sorry for posting an irrelevant request.

puddly, 2024-01-01:
ZHA supports any standard-compliant device out of the box so that list will never be complete: it's built from user-reported "it works for me" submissions.

closed 2024-01-01 by puddly