zigpy/zigpy #1315: Failed to send request: Failed to enqueue message after 3 attempts: <EmberStatus.NO_BUFFERS: 24>

marmolak, 2024-01-05:
I'm using Home Assistant and after upgrade to 2024.1.0, I'm no longer able to use zigbee stack.

If I want to switch device on and off, I just receive:
```
[140564133696832] Failed to send request: Failed to enqueue message after 3 attempts: <EmberStatus.NO_BUFFERS: 24>
```

And later:
```
Logger: zigpy.topology
Source: /usr/local/lib/python3.11/site-packages/zigpy/topology.py:84
First occurred: 22:32:51 (1 occurrences)
Last logged: 22:32:51

Topology scan failed
```

I'm using hacked lidl gateway with last known firmware (I can share more details if needed).

My ezsp config is:
```
ezsp_config:
        CONFIG_SOURCE_ROUTE_TABLE_SIZE: 130
        CONFIG_NEIGHBOR_TABLE_SIZE: 16
        CONFIG_MAX_END_DEVICE_CHILDREN: 6
        CONFIG_APS_UNICAST_MESSAGE_COUNT: 32
        CONFIG_BINDING_TABLE_SIZE: 32
        CONFIG_ADDRESS_TABLE_SIZE: 130
        CONFIG_ROUTE_TABLE_SIZE: 32
        CONFIG_BROADCAST_TABLE_SIZE: 32
        CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE: 4
        CONFIG_MTORR_FLOW_CONTROL: 1
```

MattWestb, 2024-01-05:
firs take all your "extra" ZHA configs the they is not needed and can making problems and restart HA and then post full debug log of the HA start so can see how the NCP is initiated.

The only useful setting you having **is**  `CONFIG_MAX_END_DEVICE_CHILDREN: 0` (but with zero) so you is not getting end device doping then the system is offline.

marmolak, 2024-01-05:
Hello. I removed all CONFIG_ (which I set because then my zigbee net is stalble enough), restarted HA, enabled debug logs and then reloaded ZHA.

Where I can upload debug log? It should contain sensitive data so I would like to not share it on public channels and also I don't want to do cleanup because I can remove some crucial informations.

MattWestb, 2024-01-05:
I can pinging @puddly and hi have one private email for logs and hi is on the hunt of the problem.

stguz, 2024-01-21:
I have the same Problem since Update to 2024.1.1-5. Back to 2023.12.3 solves the Problems. No Idee why.

nerdosity, 2024-01-31:
Same for me.
I can make them working again if I restart the coordinator or home assistant.
But ZHA will eventually die in a couple of hours

abramovtv, 2024-06-17:
> But ZHA will eventually die in a couple of hours
The same for me.

closed 2025-05-31 by github-actions[bot]