zigpy/zigpy #1322: Device requires a corresponding out cluster when communicating with the coordinator

matejx, 2024-01-13:
I have a device in the network, trying to read time (ep 1, cluster 0xa, attr 0)  from the coordinator. This fails in device.py:packet_received() line 385 with the message "Ignoring message on unknown cluster 10 for endpoint 1". 

I'm not super familliar with Zigbee protocol, but this seems wrong. There's even a comment in the code suggesting this shouldn't be necessary. Reading other devices' attributes (I have mostly sonoff) works fine and does not require this.

Adding the required output cluster in my code fixes this, i.e.
mydev.add_output_cluster(10)
but I consider this a workaround. Nothing on the actual device is changed.

Hedda, 2024-01-17:
Are you using the ZHA (Zigbee Home Automation) integration component in Home Assistant? And what exactly is the device in question? Do you brand/model and device signature if it is a commercial product you purchased or is it a DIY device you made?  

If you use the ZHA integration in Home Assistant then you will likely need a custom ZHA Device Handler (also known as a "quirk" for ZHA), and if so then the normal way to raise an issue is usually to start by creating a new device support request in that repository (even if you plan on writing custom quirks for ZHA yourself). See:

https://www.home-assistant.io/integrations/zha/#how-to-add-support-for-new-and-unsupported-devices

and 

https://github.com/zigpy/zha-device-handlers/

If you are not using the ZHA (Zigbee Home Automation) integration component in Home Assistant then the zigpy developers will probably need a lot more information on your exact software setup, environment and hardware using?:

* Device details?
* Operating system?
* Zigbee Coordinator adapter and firmware used?
* Debug logs.
* etc.

matejx, 2024-01-17:
I understand this is currently expected behavior, I just thought I'd give the comment

`# TODO: this isn't actually necessary, we can parse most packets by cluster ID.`

more weight by providing a real world example where this implementation prevents functionality.

The device I'm using is eByte E18-MS1-PCB, a serial to ZigBee module. It unfortunately does not support adding device local clusters. I'm currently trying to get an explanation from eByte engineers on why this is so.

closed 2024-07-22 by github-actions[bot]