zigpy/zigpy #1328: How to use remote_provider with home assistant and zigpy

Fredrik81, 2024-01-23:
Is it possible to use remote_providers with home assistant in current version of zigpy.
I see no documentation on the topic (if my searches did not file me).

The code I'm testing and I receive the following error:
voluptuous.error.MultipleInvalid: expected a list for dictionary value @ data['ota']['remote_providers]

```
zha:
  zigpy_config:
    ota:
      ikea_provider: true                        # Auto update Trådfri devices
      inovelli_provider: true                    # Auto update INOVELLI devices
      ledvance_provider: true                    # Auto update LEDVANCE/OSRAM devices
      salus_provider: true                       # Auto update SALUS/Computime devices
      sonoff_provider: true                      # Auto update Sonoff
      thirdreality_provider: true                # Auto update 3REALITY devices
#      otau_directory: /config/zipgy_ota          # Utilize .ota files to update everything
      remote_providers:
        url: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json"
```

MattWestb, 2024-01-23:
If you is not having ZHA on the master you need putting it in the config for the client (on the master) or it cant loading all component OK.
Your problem is the OTA configure that is not standard and making problems not the ZHA itself.
Delete the
 ```
remote_providers:
        url: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json"
```
and im very sure you problems is solved then its **not supported** setting for ZHA.

Fredrik81, 2024-01-23:
> If you is not having ZHA on the master you need putting it in the config for the client (on the master) or it cant loading all component OK. Your problem is the OTA configure that is not standard [...]

I'm not asking why is my ZHA reporting error my question is:
```Is it possible to use remote_providers with home assistant in current version of zigpy```

puddly, 2024-01-23:
Remote providers are currently for development purposes. The metadata format is zigpy-specific and won't accept just any JSON. See https://github.com/zigpy/zigpy-cli#generate-ota-index-files

Hedda, 2024-01-26:
> Is it possible to use remote_providers with home assistant in current version of zigpy?
> I see no documentation on the topic (if my searches did not file me).

FYI, the Documentation about that zigpy OTA index files schema format for manufacturers is in the zigpy wiki in this repository:

https://github.com/zigpy/zigpy/wiki/OTA-Information-for-Manufacturers

See readme zigpy-cli file about generating compatible OTA index files:

https://github.com/zigpy/zigpy-cli#generate-ota-index-files

https://github.com/zigpy/zigpy-cli?tab=readme-ov-file#ota

As mentioned, the existing index.json file in Koenkk's zigbee-OTA does not follow metadata format that is specific to zigpy, see:

https://github.com/Koenkk/zigbee-OTA/issues/263

and

https://github.com/zigpy/zigpy/issues/535

PS: Off-topic but FYI, if you are and end-users that just want a workaround for downloading those then check out zha-toolkit :

https://github.com/mdeweerd/zha-toolkit?tab=readme-ov-file#ota_notify---downloadtrigger-device-fw-update

PPS: Other than that see the documentation in zigpy wiki for manual OTA updates or when a provider is already available:

https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates