zigpy/zigpy #1347: IKEA OTA failing with error 1

MattWestb, 2024-02-29:
One system have updating the new motion sensor (RCP) but one in the production system refusing doing it with this error in the log:
```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 12:41:37 (1 occurrences)
Last logged: 12:41:37

[547857706816] Update was not successful - result: <Status.FAILURE: 1>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 161, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 465, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/zha/update.py", line 177, in async_install
    raise HomeAssistantError(f"Update was not successful - result: {results}")
homeassistant.exceptions.HomeAssistantError: Update was not successful - result: <Status.FAILURE: 1>
```
Then sniffing the device is requesting the new images OK foroffset 0 and getting one ack in the app layer but not response with the payload so its the device is re requesting untill its giving up.

The updated deviceinfo:
```
Device info
VALLHORN Wireless Motion Sensor
by IKEA of Sweden
Firmware: 0x01000064
Zigbee info
IEEE: 8c:6f:b9:ff:fe:1f:19:86
Nwk: 0x84d5
Device Type: EndDevice
LQI: 208
RSSI: -48
Last seen: 2024-02-29T13:13:11
Power source: Battery or Unknown
```
PS: The OTA is only in Dirigera feed not in TF.

puddly, 2024-02-29:
Is this with Home Assistant 2024.3.0b0?

MattWestb, 2024-02-29:
No its on the production system thta was running HA 2024.2.0 and was having that problem and was updating all system to 2023.2.5 and its still there.
Its have one Billy EZSP 6.10.X.

I thinking letting it on the production system until HA 2024.3 is released and see if its doing the same. The problem i cant downgrading it then its updated and very likely locked bootloader like all new devices and i dont like suing one hammer for opening it (have not found and taking it apart without destroying it). If next major is working all is OK. If moving it to one of the RCP its very likely can working and we dont have any live system for looking on the problem.

Do you like have one sniff of it ? I can do it filtering one the mac layer so getting all and not other bad things.
If yes i mailing it to you plus network key in the mail.

TheJulianJES, 2024-02-29:
There are OTA improvements (including an increased timeout) in HA Core 2024.3.0 betas.

puddly, 2024-02-29:
Yeah, the most notable one I think for stability is that the max data size is actually respected. Before, when the device asked for a 64 byte chunk of the OTA file, it got a 64 byte chunk. We limit it to 40 bytes in the beta.

MattWestb, 2024-02-29:
Its requesting 64 byte
```
Frame 3745: 121 bytes on wire (968 bits), 121 bytes captured (968 bits) on interface 
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x073b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x995e
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x01)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...0 .... = Disable Default Response: False
    Sequence Number: 90
    Command: Upgrade End Request (0x06)
    Payload
        Status: Ota Abort (0x95)
        Manufacturer Code: Ikea of Sweden (0x117c)
        Image Type: Manufacturer Specific (0x1938)
        File Version: 0x01000064
            0000 0001 .... .... .... .... .... .... = Application Release: 1
            .... .... 0000 0000 .... .... .... .... = Application Build: 0
            .... .... .... .... 0000 0000 .... .... = Stack Release: 0
            .... .... .... .... .... .... 0110 0100 = Stack Build: 100
```
And data to and from the device in the Zigbee network layer:
[ikeaotaA.txt](https://github.com/zigpy/zigpy/files/14453664/ikeaotaA.txt)

MattWestb, 2024-03-03:
Ping @puddly  !!!

Have mailing one sniff to you with failing OTA.

agoode, 2024-03-07:
I am having the same issue with HA 2024.3.0.

MattWestb, 2024-03-07:
With HA 2014.3.0 it have starting updating an looks doing it OK so liiks being fixed in the last rework done in the OTA code

mese16, 2024-03-07:
Same  issue with HA 2024.3.0.

puddly, 2024-03-07:
@mese16 Please don't cross-post "same issue" multiple times across repositories. If you aren't adding any new information or new debug logs of the entire update process, your comment isn't really contributing anything.

markdinh, 2024-03-07:
Updated to HA 2024.3.0 on HAOS.   Moved motion sensor within 5" of my coordinator and put batteries back in.  Still fails firmware update from 0x01000057 to 0x01000064.  Just figured I'd add one more data point.  If it helps to send logs, appreciate a little hint on how to do so.  I'm a HA newbie and it seems like I can only filter logbook by "firmware" entity and can't seem to figure out where the logging is for the failed firmware update attempts.